espressif/esp-idf #12191: ESP-IDF v4.3.6 not showing up in release/v4.3 branch

MMI, 2023-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The esp-idf v4.3.6 release commits are not available on the `release/v4.3` branch via pull...  oddly, recloning per the release instructions _does_ get the release correctly.

Additionally, [this page](https://github.com/espressif/esp-idf/commit/5dc5a442a291f0e6180244a55095a4a4ca1776b5) (for the 4.3.6 commit) has a warning: `This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository.
`

[This link](https://github.com/espressif/esp-idf/tree/v4.3.6) shows the same warning.

[image]

zikalino, 2023-09-05:
@MMI the tag was updated, but there's been some delay with the pipeline that updates the branch. I will keep you updated when the issue is resolved.

zikalino, 2023-09-05:
@MMI now release/v4.3 branch was updated. closing the ticket

closed 2023-09-05 by zikalino