espressif/esp-idf #12193: wifi mesh node do not find parent found (or root)

ClaesIvarsson, 2023-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

With fixed root up and running I expect any node to find and connect to the mesh network directly (10-20s) on power on.

### What is the actual behavior?

Node searches for root/parent but fail to find any and ends up throwing MESH_EVENT_NO_PARENT_FOUND after . After that, no new searches are performed and the node ends up dangling outside of the mesh network.

After power cycle and a fresh start may or may not resolve the issue.
Problem occurs approximately 50% of the times at startup.

I have also tried to delay application related tasks to not interfere at startup without any better result.

### Steps to reproduce.

1. Startup of the root
2. wait until stable
3. power on mesh node 
4. wait for node to startup ~10-15s

Init code for the wifi/mesh is identical to code in the [ip_internal_example](https://github.com/espressif/esp-idf/blob/6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7/examples/mesh/ip_internal_network/main/mesh_main.c#L386): 

### Debug Logs.

```plain
Log when failure occurs:
I (27) boot: ESP-IDF v4.4.3-1-gcc5640aef3 2nd stage bootloader
I (27) boot: compile time 08:15:55
I (27) boot: chip revision: 3
I (29) boot.esp32: SPI Speed      : 80MHz
I (32) boot.esp32: SPI Mode       : DIO
I (34) boot.esp32: SPI Flash Size : 16MB
I (36) boot: Enabling RNG early entropy source...
I (39) boot: Partition Table:
I (41) boot: ## Label            Usage          Type ST Offset   Length
I (44) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (48) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (52) boot:  2 coredump         Unknown data     01 03 00010000 00010000
I (56) boot:  3 reserved         Unknown data     01 fe 00020000 00020000
I (59) boot:  4 hwadmin          Unknown data     01 d0 00040000 00001000
[... 355 lines cut ...]
I (9585) [mesh_netif, 469]: It was a wifi station removing stuff
I (9588) [mesh_netif, 84]: Receiving task started
I (9637) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (9641) [node_main, 88]: <MESH_EVENT_PARENT_CONNECTED>
I (9642) [node_main, 110]: MESH_EVENT_TODS_STATE toDS=1
I (9645) [node_main, 97]: <MESH_EVENT_ROOT_ADDRESS>root address:c4dee21e053d
```

### More Information.

Using a mesh network system and have a custom software based on the ip_internal_network example (with a lot of application additions).
Root node is fixed!

ESP32-WROVER 16MB flash / 4MB ram
running on both cores
wifi module assigned to core 1

I also tried to modify the original example project with same changes we have in our project (sdkconfig):
* enable SPIRAM
* dual core
* wifi on core 1
* other tuning of SPIRAM usage
None of the changes in the example project caused similar behavior. The example project connect to my projects root 100% of the startups.

I have not tested more recent version of esp-idf since it breaks my project.

ClaesIvarsson, 2023-09-05:
Now tested with es-idf 4.4.5. Same result as for 4.4.3

ClaesIvarsson, 2023-09-12:
> 

 1. after reset of node it fails to find network 50% of the times
 2. ip_internal_network example combined with sdkconfig from my project connects well
 3. yes, all nodes except root (root gets esp_mesh_set_type(MESH_ROOT)
 4. I do not call esp_mesh_set_self_organized as documentation says it is enabled by default

zhangyanjiaoesp, 2023-09-12:
@ClaesIvarsson  Can you provide a demo for us to reproduce this issue?

ClaesIvarsson, 2023-09-12:
> @ClaesIvarsson Can you provide a demo for us to reproduce this issue?

Yes, I can demonstrate the problem, Maybe Friday at 10 CET is ok ? Please let me know and I will provide an invitation.

zhangyanjiaoesp, 2023-09-13:
> > @ClaesIvarsson Can you provide a demo for us to reproduce this issue?
> 
> Yes, I can demonstrate the problem, Maybe Friday at 10 CET is ok ? Please let me know and I will provide an invitation.

Do you mean that you want to demonstrate your problem online? Actually, we hope you can provide a reproduce code so that we can test and debug it locally.

ClaesIvarsson, 2023-09-14:
Well, 

> Do you mean that you want to demonstrate your problem online? Actually, we hope you can provide a reproduce code so that we can test and debug it locally.

Maybe a misunderstanding from my side!

I am not sure I am allowed to share the existing code, will have to check how in such case.
To strip it down to something shareable would be rather large effort that I am not able to fit in my schedule right now. I will check how to proceed with my manager.

ClaesIvarsson, 2023-09-15:
Is there anyway to get more debug information out from the mesh module ? Is there any information the provided logs that gives any hints on the problem at hand ?
Any explanation why the scanning of network doesn't restart after failure of finding a network ?
It looks like there is a full scan of WiFi networks at startup but that only happens once ... is this true ?

ClaesIvarsson, 2023-10-19:
I have spent a lot of time trying to find a reason for this issue to happen.
My thesis has been that there is some kind of memory corruption, but I have not been able to find a pattern to follow.

I can see that some builds are more prune to fail than others. I have used git bisect function to try to find a commit that cause the issue. But my conclusion is that this happens on almost versions of my software since I introduced the NAT feature based on the ip_internal_network. If present, mainly depends on the number of tests performed.
I created a script that restart the node and waits for the event MESH_EVENT_PARENT_CONNECTED (with a timeout of 90s) and repeat this for N time (I ran my tests with 20 repetitions).

yanmartins, 2023-10-19:
> I have spent a lot of time trying to find a reason for this issue to happen. My thesis has been that there is some kind of memory corruption, but I have not been able to find a pattern to follow.
>
> I [...]

I noticed the same problem when using NAT with this same example (ip_internal_network). Yesterday I opened an issue, maybe it's the same problem: #12431

ClaesIvarsson, 2023-10-19:
> I noticed this problem when using NAT with this same example (ip_internal_network). Yesterday I opened an issue, maybe it's the same problem: #12431

We tried the example with minor adjustments used as node in the mesh network (not root) and didn't see this behavior at that point, but maybe we didn't test it enough times. I will try to do this test again with more repetitions.

ClaesIvarsson, 2023-10-20:
I believe I have found the root cause for this problem!

The mesh password variable stored in the cfg structure was not null terminated when my password was copied. It rely on the fact that the cfg.mesh_ap.password buffer is initialized with zero value.
After adding a command to set the complete cfg structure to zero, I got rid of this problem.
I think it would be good if all mesh network examples are updated with this.
```
    mesh_cfg_t cfg; /*  = MESH_INIT_CONFIG_DEFAULT(); */
    memset(&cfg, 0, sizeof(mesh_cfg_t);
    cfg.crypto_funcs = &g_wifi_default_mesh_crypto_funcs;
```
or at least make sure that 
buffers in the structure without length information is also terminated properly

```
cfg.router.password[sizeof(CONFIG_MESH_ROUTER_PASSWD)] = '\0';
cfg.mesh_ap.password[sizeof(CONFIG_MESH_AP_PASSWD)] = '\0';
```

zhangyanjiaoesp, 2023-10-23:
> I believe I have found the root cause for this problem!
>
> The mesh password variable stored in the cfg structure was not null terminated when my password was copied. It rely on the fact that the [...]

@ClaesIvarsson Thanks for your update, we will consider to update it.

zhangyanjiaoesp, 2023-10-24:
@ClaesIvarsson 
Why the IDF official example can work well ?  Is there anything special about your password？

ClaesIvarsson, 2023-10-24:
> @ClaesIvarsson Why the IDF official example can work well ? Is there anything special about your password？

I think it is luck or coincidence. The cfg variable in located on the stack which is uninitialized. My task is located in external ram, maybe this is the major difference.

MaxwellAlan, 2024-03-05:
@ClaesIvarsson @zhangyanjiaoesp 

Thanks for the reporting and debuging.

We will close this issue, plz feel free to reopen is still have this issue.

Thanks.

closed 2024-03-05 by MaxwellAlan