espressif/esp-idf #12195: cJSON_Delete() and memory leak

Mbded, 2023-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Helo

I have a IDF 4.4.3. I want to build JSON request each will be send via web socket. I noticed memory leak thanks for function,  _heap_caps_get_free_size(MALLOC_CAP_8BIT));_ 
Generaly my code works is simple, when I press some button I also send to server notification. This notification message is build as below. 

First usage:
```
{
                ESP_LOGI(TAG, "Size heap ---%d",  heap_caps_get_free_size(MALLOC_CAP_8BIT)); // function call after each button press. The return value is also each one decreased after each call, so memory leak
                auto state = SwitchStateRequest(); // create  object which has similar to builder. This one build  below JSON message
                auto msg = state.GetJSONMsg();
                ESP_LOGI(TAG, "Message -------------------------- %s", msg.c_str());
}
```
And finally here. Below is my way to build JSON message.
```
std::string SwitchStateRequest::GetJSONMsg()
{
    auto jsonMsg = cJSON_CreateObject();
    cJSON_AddStringToObject(jsonMsg, "mac", "someMac");
    cJSON_AddItemToObject(jsonMsg, "report", CreateReportMsg());
    std::string msg = std::string(cJSON_Print(jsonMsg));
    cJSON_Delete(jsonMsg); 
    return msg; 
}

cJSON *SwitchStateRequest::CreateReportMsg()
{
    auto reportMsg = cJSON_CreateObject();
    cJSON_AddStringToObject(reportMsg, "type", "switch");
    cJSON_AddNumberToObject(reportMsg, "timestamp", 19840420);
    cJSON_AddStringToObject(reportMsg, "uuid", "idSwitch");
    cJSON_AddNumberToObject(reportMsg, "value", 1);
    return reportMsg;
}
```

I can not find where is problem. For sure it is in build JSON message. I checked carefully other part my code everything shows on this part. If only I adding this part code, memory leak at once occurred. To this moment I not have a another idea what can I do with it.  I can add that memory leak is always around 170 to 180 bytes

mahavirj, 2023-09-06:
@Mbded 

> cJSON_Print(jsonMsg)

This also does an internal memory allocation and the returned buffer must be freed after its usage.

I think you can try out the library on host OS side (e.g., Linux) and see if its API usage problem. On the host side you can use tools like `valgrind` to check on the memory profile or leaks. Please see this [Makefile](https://github.com/DaveGamble/cJSON/blob/master/Makefile) for build instructions on the host OS.

Mbded, 2023-09-06:
Yeh. Thanks for help, and today I'm also itself find answer for my problem. 
About valgrind. Is it working for esp32 ?
I also want to add ,that for me function name 

> cJSON_Print(msg);

Is not clear in this case. Because if she allocate memory on heap it should be has short brief in info pop up, or should be different name as cJSON_MallocAndToText(). Print is for me just print. But ok now I will be know.

mahavirj, 2023-09-07:
[cJSON](https://github.com/DaveGamble/cJSON) is an upstream library and my recommendation was to try out its usage in an independent manner on the host OS, and once you finalize the implementation then you can integrate with ESP-IDF.

Regarding its API design, we can not comment about it, if you have any concerns please feel free to file an issue in the upstream repository.

Since, you have now found the problem, I will go ahead and close the issue. Please feel free to drop a comment if you still have any concerns from ESP-IDF side.

closed 2023-09-07 by mahavirj