espressif/esp-idf #12198: Touching a larger touch sensor makes touch_sensor_v2 hang

richardstefun, 2023-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The library should not hang at any time

### What is the actual behavior?

When more firmly touching the capacitive sensor, the library hangs and reports the same values

### Steps to reproduce.

1. Open the touch_pad_read example of touch_sensor_v2
2. Connect a larger capacitive sensor to any touch sensitive pin, in series with 510 ohm resistor (for example TOUCH4) 
3. Observe, how the values stop updating
...

### More Information.

Before touching:

```
T4: [104959] T5: [103897] T6: [110782] T7: [97745] T8: [99613] T9: [97447] T10: [99638] T11: [96719] T12: [89022] T13: [82840] T14: [103454]
T4: [104984] T5: [103927] T6: [110811] T7: [97745] T8: [99632] T9: [97467] T10: [99689] T11: [96741] T12: [89034] T13: [82857] T14: [103457] 
T4: [104971] T5: [103891] T6: [110777] T7: [97754] T8: [99614] T9: [97485] T10: [99628] T11: [96736] T12: [89034] T13: [82837] T14: [103445]
```

After touching (Observe how the first value starts to rise and then all values hang..):

```
T4: [866051] T5: [103892] T6: [110789] T7: [97722] T8: [99570] T9: [97483] T10: [1421758] T11: [96701] T12: [89013] T13: [82836] T14: [103387] 
T4: [1417005] T5: [103872] T6: [110800] T7: [97737] T8: [99593] T9: [97457] T10: [1768583] T11: [96724] T12: [89013] T13: [82836] T14: [103387] 
T4: [2236163] T5: [103873] T6: [110799] T7: [97743] T8: [99605] T9: [97462] T10: [1768583] T11: [96724] T12: [89040] T13: [82863] T14: [103425] 
T4: [2236163] T5: [103873] T6: [110799] T7: [97743] T8: [99605] T9: [97462] T10: [3297204] T11: [96690] T12: [89034] T13: [82838] T14: [103429] 
T4: [2236163] T5: [103873] T6: [110799] T7: [97743] T8: [99605] T9: [97462] T10: [3297204] T11: [96690] T12: [89034] T13: [82838] T14: [103429]
```

richardstefun, 2023-09-05:
I found a temporary workaround, by setting the voltage as such:

`touch_pad_set_voltage(TOUCH_HVOLT_2V4, TOUCH_PAD_LOW_VOLTAGE_THRESHOLD, TOUCH_HVOLT_ATTEN_1V5);`

Now it does not hang with the sensor size I use..

leeebo, 2023-09-06:
@richardstefun Hi, Please make sure all the enabled touch channels are not configured to other functions, and **Do not** connect the enabled touch IO to other hardware.

richardstefun, 2023-09-06:
@leeebo I use official example for the library, and the pins are not connected to other hardware

leeebo, 2023-09-06:
@richardstefun Which development board are you using?

richardstefun, 2023-09-06:
Unfortunately, it also hangs with the "touch_pad_interrupt" example

leeebo, 2023-09-07:
@richardstefun Both example works for me:

[image]

As the example enabled `T4-T14`, for esp32s3 that means all `GPIO4-GPIO14`  can **only be used for touch**, if one of the pins was accidentally connected to another device, the touch sensor will be trapped in this pin, it's causing abnormal readings in other channels also.

[image]

Please double check the hardware, or use the dev-kit board to test the code first.

richardstefun, 2023-09-07:
@leeebo 

I made this simple setup using a generic devboard, where the abnormality can be tested:

[image]

When you run the example and touch all three metal electrodes, the reading hangs..

What is more interesting, if I touch just one metal plate and GND, the reading hangs, until I stop touching the electrode. After few seconds, the readings come back to normal (Some internal capacitor needs to discharge? Just guessing..).
When I repeat this with two electrodes, the time it takes to restore the readings is much longer.
But the case where I touch three electrodes causes the readings to hang indefinitely..

leeebo, 2023-09-13:
@richardstefun The coin is equivalent to a large capacitor, which takes longer to charge and discharge, so it may have caused the overflow of some registers in touch.

Here are some solutions, please consider choosing 1 or more:
1. Use a thin copper (like copper on a PCB), or a small spring with finger size, to reduce the size of the inherent capacitance
2. DO NOT touch the metal directly, cover it with plastic or glass material then touch it (like the actual product)
3. Reduce the charge and discharge times per sample, the default is 500, you can adjust it to a smaller one, such as 100, please refer [main/tp_read_main.c#L67](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/touch_sensor/touch_sensor_v2/touch_pad_read/main/tp_read_main.c#L67)
4. Add a timeout monitor, and resume the touch when overflow happens, please refer [main/tp_interrupt_main.c#L199](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/touch_sensor/touch_sensor_v2/touch_pad_interrupt/main/tp_interrupt_main.c#L199)

richardstefun, 2023-09-13:
Thank you for the tips. I currently fixed it by covering the touch plates with a foil, but I'll implement more of your suggestions to be sure

quicksketch, 2024-07-17:
Thank you @richardstefun and @leeebo. I found this issue Googling, as I had a similar coin-sized touch sensor.

My sensor was fine touching lightly or for a second at a time, but one tight squeeze or a few seconds holding and it would max out the sensor input and stay stuck at the high sensor reading value.

Based on @leeebo's suggestions, my solution was to apply a coat of fingernail polish to my brass disc sensor. That lowered the conductivity between the sensor and the finger, and seems to now be highly predictable. I like that I can potentially apply more or less nail polish to adjust how high or low a sensor reading I want. Thank you for the suggestion!

closed 2025-04-07 by Alvin1Zhang