espressif/esp-idf #12199: Nimble notify stops working after connecting another device to computer

13Pixel, 2023-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-231-ga7b62bbcaf

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32Wroover-Kit 2, Custom board

### Power Supply used.

USB

### What is the expected behavior?

I use Nimble for BLE, and implemented data sending with notify  (ESP- server, PC- client). Everything works fine. When I connect any other device(tested with keyboard, mouse, headphones, ) to computer via Bluetooth the behavior changes. 

### What is the actual behavior?

When I connect another device to computer- I start to miss data, and ESP no longer sending notify messages. 

### Steps to reproduce.

Just Add another bluetooth device.

### Debug Logs.

```plain
E (10148) NimBLECharacteristic: Conn_handle (65535) 0.
W (10154) NimBLECharacteristic: Conn_handle (65535) is above the maximum value (3839)
I (10163) NimBLE: GATT procedure initiated: notify; 
I (10168) NimBLE: att_handle=43

E (10172) NimBLECharacteristic: Conn_handle (65535) 0.
W (10178) NimBLECharacteristic: Conn_handle (65535) is above the maximum value (3839)
I (10187) NimBLE: GATT procedure initiated: notify; 
I (10192) NimBLE: att_handle=43

E (10196) NimBLECharacteristic: Conn_handle (65535) 0.
W (10202) NimBLECharacteristic: Conn_handle (65535) is above the maximum value (3839)
I (10211) NimBLE: GATT procedure initiated: notify; 
I (10216) NimBLE: att_handle=43

I (10220) NimBLE: GATT procedure initiated: notify; 
I (10226) NimBLE: att_handle=43

I (10238) NimBLE: GATT procedure initiated: notify; 
I (10238) NimBLE: att_handle=43
```

### More Information.

I tried Windows computer, the result is the same.

ESPAbhinav, 2023-09-06:
Hi @13Pixel ,
We tried to reproduce the issue by invoking the `ble_gatts_custom_notify` from the main function of the server application to notify the PC (client). While notification sending was going on I connected a mobile device and mouse to a PC via Bluetooth of PC and the notification-sending procedure didn't stop and worked fine.
So can you please share the following things in order to get more about your issue:
- Exact steps you followed including (API calls and application used) during which faced the issue.
- Debug enabled logs. 
Following is the procedure for getting debug enable logs:

                            Component Config  --> Log Output --> Default Log Verbosity --> Debug

                            Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs

ESPAbhinav, 2023-09-11:
Thank you for your confirmation.

closed 2023-10-13 by rahult-github