espressif/esp-idf #12200: FOR ESP32-C3 GPIO

hzhh110, 2023-09-06:
### Is your feature request related to a problem?

Does the ESP32-C3 have a bias when operating the GPIO?I feel that when sending infrared, the deviation is very serious, resulting in AC control, and when sending radio frequency, the deviation will be more serious than ESP32.

### Describe the solution you'd like.

Does the ESP32-C3 have a bias when operating the GPIO?I feel that when sending infrared, the deviation is very serious, resulting in AC control, and when sending radio frequency, the deviation will be more serious than ESP32.

ginkgm, 2023-09-07:
Hi @hzhh110 ,

Do you mean bias in the level or the time?

hzhh110, 2023-09-08:
Hi @ginkgm
For example, before ESP32, ESP32-S3 operation GPIO port can control IR and RF data, but the same operation ESP32-C3, you will find that it is basically difficult to control, RF data is better, but the same deviation is much higher, but AC air conditioning, basically as long as wifi or Bluetooth is turned on, it can not be controlled directly.10 or 20 times, once successfully.
For example, before ESP32, ESP32-S3 operation GPIO port can control IR and RF data, but the same operation ESP32-C3, you will find that it is basically difficult to control, RF data is better, but the same deviation is much higher, but AC air conditioning, basically as long as wifi or Bluetooth is turned on, it can not be controlled directly.10 or 20 times, once successfully.

hzhh110, 2023-09-08:
Hi @ginkgm
For example, before ESP32, ESP32-S3 operation GPIO port can control IR and RF data, but the same operation ESP32-C3, you will find that it is basically difficult to control, RF data is better, but the same deviation is much higher, but AC air conditioning, basically as long as wifi or Bluetooth is turned on, it can not be controlled directly.10 or 20 times, once successfully.