espressif/esp-idf #12202: CMake Error at /home/onepower/esp/esp-idf/components/mqtt/CMakeLists.txt:1   (include):   include could not find requested file

Teesmo, 2023-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I encounter an error when I try to set-my target to esp32s2, using the esp-mesh-lite component, and including mesh_client.h
I expect idf.py to set the target to esp32s2

### What is the actual behavior?

Instead, it returns the error 
```

 CMake Error at /home/onepower/esp/esp-idf/components/mqtt/CMakeLists.txt:1
  (include):

    include could not find requested file
CMake Error at /home/onepower/esp/esp-idf/components/mqtt/CMakeLists.txt:1
  (include):

    include could not find requested file:

  

      /home/onepower/esp/esp-idf/components/mqtt/esp-mqtt/CMakeLists.txt

  Call Stack (most recent call first):

    /home/onepower/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/onepower/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)
```
I searched through /esp/esp-idf/components/mqtt/esp-mqtt, and could not locate the required file, CMakeLists.txt, which the error correctly picks out. 

### Steps to reproduce.

1. run install.sh for required target, from esp-idf folder
2. run export.sh
3. Go to example project (in my case, the project is esp-mesh-lite/examples/mesh_local_control)
4. Run idf.py set-target esp32s2 (The error occurs here)
...

### Build or installation Logs.

```plain
Updating lock file at /home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/dependencies.lock
Deleting 1 unused components
 espressif__tinyusb
Processing 7 dependencies:
[1/7] espressif/cmake_utilities (0.5.1)
[2/7] espressif/esp_modem (0.1.28)
[3/7] espressif/esp_modem_usb_dte (1.1.0~1)
[4/7] espressif/iot_bridge (0.7.1)
[5/7] espressif/usb_host_cdc_acm (2.0.0)
[6/7] idf (5.1.1)
[7/7] mesh_lite (0.1.2)
CMake Error at /home/onepower/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/onepower/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at /home/onepower/esp/esp-idf/components/mqtt/CMakeLists.txt:1
  (include):

    include could not find requested file:

  

      /home/onepower/esp/esp-idf/components/mqtt/esp-mqtt/CMakeLists.txt

  Call Stack (most recent call first):

    /home/onepower/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/onepower/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/onepower/esp/esp-idf/tools/cmake/build.cmake:574 (__component_get_requirements)
  /home/onepower/esp/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/build/log/idf_py_stderr_output_2656785 and /home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/build/log/idf_py_stdout_output_2656785
```

igrr, 2023-09-08:
>  I searched through /esp/esp-idf/components/mqtt/esp-mqtt, and could not locate the required file, CMakeLists.txt, which the error correctly picks out.

Can you try running `git submodule update --init` in `/home/onepower/esp/esp-idf`? CMakeLists.txt file of the component comes from a submodule. If you don't see it, it is probably because the submodule is not initialized.

Teesmo, 2023-09-18:
I tried it and it worked, thanks

closed 2023-09-18 by Teesmo