espressif/esp-idf #12203: UART sleep wakeup isn't working

AAl-Hadi, 2023-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3 Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The MCU should wakeup by uart data recieved

### What is the actual behavior?

The UART wakeup works on Arduino sketch but NOT on IDF

### Steps to reproduce.

This is the fw that doesn't work:
My UART0 is at RX: 36
`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_sleep.h"
#include "esp_log.h"
#include "driver/uart.h"
#include "driver/gpio.h"

#define RX_PIN 36 //UART_PIN_NO_CHANGE 

void app_main() {
    // Configure UART
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };

    // init uart 0
    uart_param_config(UART_NUM_0, &uart_config);
    uart_set_pin(UART_NUM_0, UART_PIN_NO_CHANGE, RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    uart_driver_install(UART_NUM_0, 256, 0, 0, NULL, 0);

    esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
    esp_sleep_enable_timer_wakeup(30 * 1000 * 1000);
    uart_set_wakeup_threshold(UART_NUM_0, 3);
    esp_sleep_enable_uart_wakeup(UART_NUM_0);

    while(1){
        // Enter light sleep mode
        ESP_LOGI("TAG", "Before sleep");
        vTaskDelay(pdMS_TO_TICKS(1000));
        esp_light_sleep_start();
        
        ESP_LOGI("TAG", "After wakeup");
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
`

The Arduino sketch that works:
`#include <esp_sleep.h>
#include <driver/uart.h>
 
int variable = 0;
 
void setup() {
    Serial.begin(115200);
 
    while(!Serial){delay(500);}
 
    delay(1000);
 
    Serial.println();
    Serial.println("LIGHT SLEEP ENABLED");
    delay(100);

    uart_set_wakeup_threshold(UART_NUM_0, 3);
    esp_sleep_enable_uart_wakeup(UART_NUM_0);
     
    esp_light_sleep_start();
 
    Serial.begin(115200);
    while(!Serial){delay(500);}
    // Clean Serial from wake up message
    Serial.flush();
    Serial.readString();
 
    Serial.println();
    Serial.println("LIGHT SLEEP WAKE UP");
}
 
void loop() {
    while (Serial.available()>0){
        Serial.write(Serial.read());
    }
}`

AAl-Hadi, 2023-09-11:
Any response please?

songruo, 2023-09-12:
Hi @AAl-Hadi, please check our UART wakeup example code (https://github.com/espressif/esp-idf/blob/master/examples/system/light_sleep/main/uart_wakeup.c). I think you are missing configuration for the IO at sleep.

AAl-Hadi, 2023-09-17:
Hello
Thank you @songruo . Yes, the example works. I'll try to test the differences between the example and mine.

closed 2023-09-17 by AAl-Hadi