espressif/esp-idf #12204: Task name IDLE not smame as in the documentaion

Botond-Colop, 2023-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,

I found a little difference between the documentation and the actual code.

According to the IDF Programming guide, two IDLEs task with different numbers should be started.
[image]
Otherwise, the task information cannot be requested from the operating system. (as you can see on this picture with IDLE)
[image]
The task list was queried with this function: _vTaskList(taskList)_
As we can see, the two IDLE tasks have the same name.
I am using IDF Version 4.4.2

What is also not optimal is that a task has a space _Tmr Svc_ in its name instead of being separated by an underscore **Tmr_Scv**.

Thank you in advance.

Best regards,
Botond

ESP-Marius, 2023-09-07:
Yeah, seems like this got changed back in a3c90bf59add3 during a freertos kernel upgrade.

We'll take a look at it!

ammaree, 2023-09-07:
Can I second the comment on ‘Tmr Svc” becoming “TmrSvc” or similar.

We use the task name as part of our Syslog support and a space in the name breaks the RFC standard!!

Took the issue up with Richard Barry many years ago, he agreed to resolve but probably forgot….

Botond-Colop, 2023-09-08:
The fact that the Tmr Svc has a space in the name is only a small problem, but the two "IDLE" tasks have the same name, in my opinion it is a bug.
With the same name "IDLE" you are not able to get some separately Task info to the two tasks. You can always query information about the last started task.

igrr, 2023-09-08:
FWIW there is [xTaskGetIdleTaskHandleForCPU](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html?highlight=xtaskgetidletaskhandle#_CPPv428xTaskGetIdleTaskHandleForCPU10BaseType_t) which should allow you to get each of the idle task handles.

closed 2023-09-27 by espressif-bot