espressif/esp-idf #12205: ESP32c6 returning WiFi primary channel for all packets as 0 in promiscuous mode

fliptrail, 2023-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2317-g3247253671

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32c6 official

### Power Supply used.

USB

### What is the expected behavior?

In promiscuous mode, the field `channel` in `rx_ctrl` in `wifi_promiscuous_pkt_t`, should indicate the correct WiFi channel in which packet is received. 
This should work regardless of whether `HE_SUPPORT` is on or off. 
This should work for all ESP SoCs.

### What is the actual behavior?

In promiscuous mode, the field `channel` in `rx_ctrl` in `wifi_promiscuous_pkt_t`, is set as 0 for all packets received. 
This bug is occurring if `HE_SUPPORT` is on or off. 

However, this only occurs on esp32c6. The same code when flashed on esp32c3 and esp32 shows the correct recipient channel. 

### Steps to reproduce.

Register a callback which prints the value of channel field of `rx_ctrl` after parsing the buffer as `wifi_promiscuous_pkt_t`.

### More Information.

This bug is occurring if `HE_SUPPORT` is on or off. This only occurs on esp32c6. The same code when flashed on esp32c3 and esp32 shows the correct recipient channel.

xuxiao111, 2023-09-14:
@fliptrail Thanks for the reply. We are aware of the issue and the fix is already in progress.

AxelLin, 2023-10-19:
@xuxiao111 Is the fix available? (This is marked as "Resolution: Done" for a month).

xuxiao111, 2023-10-19:
Hi @AxelLin @fliptrail, in the latest version, it has been fixed.

AxelLin, 2023-10-19:
@xuxiao111 
It's difficult to trace the status, can you share the commit id for the fix?
Also please close the issue if it's fixed for all impacted branches (should be v5.1 and master).

xuxiao111, 2023-10-19:
Hi @AxelLin, we are using this commit: b82a55d41bbf604a8a82fc43d95549a9a0bdfda4 to fix this issue.

xuxiao111, 2023-11-15:
Hi @fliptrail @AxelLin,  I'm closing this issue and if you still have issues, please feel free to open another GitHub issue to track it.

closed 2023-11-20 by Indastri