espressif/esp-idf #12206: assert failed: block_trim_free tlsf.c:496 (block_is_free(block) && "block must be free") #4845

srikanthpalvai, 2023-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.2-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 S3 USB OTG 

### Power Supply used.

USB

### What is the expected behavior?

I am sending http request using socket and used lwip 
facing memory assert 

### What is the actual behavior?

It's should not through assertion 

### Steps to reproduce.

send socket send 

### Debug Logs.

```plain
assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)

Backtrace: 0x4037623a:0x3fcb7880 0x4037d9dd:0x3fcb78a0 0x40387625:0x3fcb78c0 0x40383608:0x3fcb79e0 0x40382ff2:0x3fcb7a00 0x40376459:0x3fcb7a20 0x403764b9:0x3fcb7a40 0x403764ee:0x3fcb7a60 0x40376ee0:0x3fcb7a80 0x4037707d:0x3fcb7b00 0x42026a21:0x3fcb7b20 0x42026ab9:0x3fcb7b40 0x42026b41:0x3fcb7b60 0x4202bbd0:0x3fcb7b80 0x4202c211:0x3fcb7bb0 0x420c9cb6:0x3fcb7c20 0x40384b7c:0x3fcb7c40 0x40386a11:0x3fcb7c60 0x403869e8:0x3fcb7c80 0x403814ce:0x3fcb7cb0
0x4037623a: panic_abort at /home/lab/esp/esp-idf/components/esp_system/panic.c:423

0x4037d9dd: esp_system_abort at /home/lab/esp/esp-idf/components/esp_system/esp_system.c:153

0x40387625: __assert_func at /home/lab/esp/esp-idf/components/newlib/assert.c:78

0x40383608: block_locate_free at /home/lab/esp/esp-idf/components/heap/tlsf/tlsf.c:566
 (inlined by) tlsf_malloc at /home/lab/esp/esp-idf/components/heap/tlsf/tlsf.c:1004

0x40382ff2: multi_heap_malloc_impl at /home/lab/esp/esp-idf/components/heap/multi_heap.c:207

0x40376459: heap_caps_malloc_base at /home/lab/esp/esp-idf/components/heap/heap_caps.c:167

0x403764b9: heap_caps_malloc at /home/lab/esp/esp-idf/components/heap/heap_caps.c:187

0x403764ee: heap_caps_malloc_default at /home/lab/esp/esp-idf/components/heap/heap_caps.c:212

0x40376ee0: trace_malloc at /home/lab/esp/esp-idf/components/heap/include/heap_trace.inc:97

0x4037707d: __wrap_malloc at /home/lab/esp/esp-idf/components/heap/include/heap_trace.inc:159

[... 270 lines cut ...]
Rebooting...
```

### More Information.

I (29) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (29) boot: compile time 13:26:53
I (29) boot: chip revision: v0.1
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00177000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=3d254h (250452) map
I (142) esp_image: segment 1: paddr=0004d27c vaddr=3fc99600 size=02d9ch ( 11676) load
I (145) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=cd5b4h (841140) map
I (299) esp_image: segment 3: paddr=0011d5dc vaddr=3fc9c39c size=016c8h (  5832) load
I (300) esp_image: segment 4: paddr=0011ecac vaddr=40374000 size=155e0h ( 87520) load
I (332) boot: Loaded app from partition at offset 0x10000
I (333) boot: Disabling RNG early entropy source...

srikanthpalvai, 2023-09-07:
My main.cpp contains 

usb_host_power_init();
//Initialize NVS TODO first boot is crashing, Yet to check this issue
esp_err_t nfret = nvs_flash_init();
if (nfret == ESP_ERR_NVS_NO_FREE_PAGES || nfret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
ESP_ERROR_CHECK(nvs_flash_erase());
nfret = nvs_flash_init();
}
ESP_ERROR_CHECK(nfret);

ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
ESP_LOGI(TAG, "After nvs [APP] Free memory: %lu bytes Internal %lu", esp_get_free_heap_size(), esp_get_free_internal_heap_size());

wifi_init_sta();
vTaskDelay(500);
usbInstall(); //here i will open the cdc_acm_host_open_vendor_specific()
vTaskDelay(500);

xTaskCreatePinnedToCore(MyTask, "MyTask", 16384, NULL,10, &myTaskHandle, 1);

srikanthpalvai, 2023-09-07:
# ESP NETIF Adapter
#
CONFIG_ESP_NETIF_IP_LOST_TIMER_INTERVAL=120
CONFIG_ESP_NETIF_TCPIP_LWIP=y
# CONFIG_ESP_NETIF_LOOPBACK is not set
# CONFIG_ESP_NETIF_L2_TAP is not set
# CONFIG_ESP_NETIF_BRIDGE_EN is not set
# end of ESP NETIF Adapter

#
# LWIP
#
CONFIG_LWIP_LOCAL_HOSTNAME="espressif"
# CONFIG_LWIP_NETIF_API is not set
# CONFIG_LWIP_TCPIP_CORE_LOCKING is not set
# CONFIG_LWIP_CHECK_THREAD_SAFETY is not set
CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES=y
# CONFIG_LWIP_L2_TO_L3_COPY is not set
CONFIG_LWIP_IRAM_OPTIMIZATION=y
CONFIG_LWIP_TIMERS_ONDEMAND=y
CONFIG_LWIP_MAX_SOCKETS=10
# CONFIG_LWIP_USE_ONLY_LWIP_SELECT is not set
# CONFIG_LWIP_SO_LINGER is not set
CONFIG_LWIP_SO_REUSE=y
CONFIG_LWIP_SO_REUSE_RXTOALL=y
CONFIG_LWIP_SO_RCVBUF=y
# CONFIG_LWIP_NETBUF_RECVINFO is not set
CONFIG_LWIP_IP4_FRAG=y
CONFIG_LWIP_IP6_FRAG=y
# CONFIG_LWIP_IP4_REASSEMBLY is not set
# CONFIG_LWIP_IP6_REASSEMBLY is not set
# CONFIG_LWIP_IP_FORWARD is not set
# CONFIG_LWIP_STATS is not set
CONFIG_LWIP_ESP_GRATUITOUS_ARP=y
CONFIG_LWIP_GARP_TMR_INTERVAL=60
CONFIG_LWIP_ESP_MLDV6_REPORT=y
CONFIG_LWIP_MLDV6_TMR_INTERVAL=40
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=32
CONFIG_LWIP_DHCP_DOES_ARP_CHECK=y
# CONFIG_LWIP_DHCP_DISABLE_CLIENT_ID is not set
CONFIG_LWIP_DHCP_DISABLE_VENDOR_CLASS_ID=y
# CONFIG_LWIP_DHCP_RESTORE_LAST_IP is not set
CONFIG_LWIP_DHCP_OPTIONS_LEN=255
CONFIG_LWIP_NUM_NETIF_CLIENT_DATA=0
CONFIG_LWIP_DHCP_COARSE_TIMER_SECS=1

#
# DHCP server
#
CONFIG_LWIP_DHCPS=y
CONFIG_LWIP_DHCPS_LEASE_UNIT=60
CONFIG_LWIP_DHCPS_MAX_STATION_NUM=8
# end of DHCP server

# CONFIG_LWIP_AUTOIP is not set
CONFIG_LWIP_IPV6=y
# CONFIG_LWIP_IPV6_AUTOCONFIG is not set
CONFIG_LWIP_IPV6_NUM_ADDRESSES=3
# CONFIG_LWIP_IPV6_FORWARD is not set
# CONFIG_LWIP_NETIF_STATUS_CALLBACK is not set
CONFIG_LWIP_NETIF_LOOPBACK=y
CONFIG_LWIP_LOOPBACK_MAX_PBUFS=8

#
# TCP
#
CONFIG_LWIP_MAX_ACTIVE_TCP=16
CONFIG_LWIP_MAX_LISTENING_TCP=16
CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION=y
CONFIG_LWIP_TCP_MAXRTX=12
CONFIG_LWIP_TCP_SYNMAXRTX=12
CONFIG_LWIP_TCP_MSS=1440
CONFIG_LWIP_TCP_TMR_INTERVAL=250
CONFIG_LWIP_TCP_MSL=60000
CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT=20000
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=12288
CONFIG_LWIP_TCP_WND_DEFAULT=12288
CONFIG_LWIP_TCP_RECVMBOX_SIZE=6
CONFIG_LWIP_TCP_QUEUE_OOSEQ=y
# CONFIG_LWIP_TCP_SACK_OUT is not set
CONFIG_LWIP_TCP_OVERSIZE_MSS=y
# CONFIG_LWIP_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_LWIP_TCP_OVERSIZE_DISABLE is not set
CONFIG_LWIP_TCP_RTO_TIME=1500
# end of TCP

#
# UDP
#
CONFIG_LWIP_MAX_UDP_PCBS=16
CONFIG_LWIP_UDP_RECVMBOX_SIZE=6
# end of UDP

#
# Checksums
#
# CONFIG_LWIP_CHECKSUM_CHECK_IP is not set
# CONFIG_LWIP_CHECKSUM_CHECK_UDP is not set
CONFIG_LWIP_CHECKSUM_CHECK_ICMP=y
# end of Checksums

CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=12288
CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_LWIP_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_LWIP_PPP_SUPPORT is not set
CONFIG_LWIP_IPV6_MEMP_NUM_ND6_QUEUE=3
CONFIG_LWIP_IPV6_ND6_NUM_NEIGHBORS=5
# CONFIG_LWIP_SLIP_SUPPORT is not set

#
# ICMP
#
CONFIG_LWIP_ICMP=y
# CONFIG_LWIP_MULTICAST_PING is not set
# CONFIG_LWIP_BROADCAST_PING is not set
# end of ICMP

#
# LWIP RAW API
#
CONFIG_LWIP_MAX_RAW_PCBS=16
# end of LWIP RAW API

#
# SNTP
#
CONFIG_LWIP_SNTP_MAX_SERVERS=1
# CONFIG_LWIP_DHCP_GET_NTP_SRV is not set
CONFIG_LWIP_SNTP_UPDATE_DELAY=3600000
# end of SNTP

CONFIG_LWIP_BRIDGEIF_MAX_PORTS=7
CONFIG_LWIP_ESP_LWIP_ASSERT=y

#
# Hooks
#
# CONFIG_LWIP_HOOK_TCP_ISN_NONE is not set
CONFIG_LWIP_HOOK_TCP_ISN_DEFAULT=y
# CONFIG_LWIP_HOOK_TCP_ISN_CUSTOM is not set
CONFIG_LWIP_HOOK_IP6_ROUTE_NONE=y
# CONFIG_LWIP_HOOK_IP6_ROUTE_DEFAULT is not set
# CONFIG_LWIP_HOOK_IP6_ROUTE_CUSTOM is not set
CONFIG_LWIP_HOOK_ND6_GET_GW_NONE=y
# CONFIG_LWIP_HOOK_ND6_GET_GW_DEFAULT is not set
# CONFIG_LWIP_HOOK_ND6_GET_GW_CUSTOM is not set
CONFIG_LWIP_HOOK_NETCONN_EXT_RESOLVE_NONE=y
# CONFIG_LWIP_HOOK_NETCONN_EXT_RESOLVE_DEFAULT is not set
# CONFIG_LWIP_HOOK_NETCONN_EXT_RESOLVE_CUSTOM is not set
CONFIG_LWIP_HOOK_IP6_INPUT_NONE=y
# CONFIG_LWIP_HOOK_IP6_INPUT_DEFAULT is not set
# CONFIG_LWIP_HOOK_IP6_INPUT_CUSTOM is not set
# end of Hooks

CONFIG_LWIP_DEBUG=y
CONFIG_LWIP_DEBUG_ESP_LOG=y
# CONFIG_LWIP_NETIF_DEBUG is not set
# CONFIG_LWIP_PBUF_DEBUG is not set
# CONFIG_LWIP_ETHARP_DEBUG is not set
# CONFIG_LWIP_API_LIB_DEBUG is not set
# CONFIG_LWIP_SOCKETS_DEBUG is not set
# CONFIG_LWIP_IP_DEBUG is not set
# CONFIG_LWIP_ICMP_DEBUG is not set
# CONFIG_LWIP_DHCP_STATE_DEBUG is not set
# CONFIG_LWIP_DHCP_DEBUG is not set
# CONFIG_LWIP_IP6_DEBUG is not set
# CONFIG_LWIP_ICMP6_DEBUG is not set
# CONFIG_LWIP_TCP_DEBUG is not set
# CONFIG_LWIP_SNTP_DEBUG is not set
# CONFIG_LWIP_DNS_DEBUG is not set
# CONFIG_LWIP_BRIDGEIF_DEBUG is not set
# CONFIG_LWIP_BRIDGEIF_FDB_DEBUG is not set
# CONFIG_LWIP_BRIDGEIF_FW_DEBUG is not set
# end of LWIP

SoucheSouche, 2023-10-02:
Hi @srikanthpalvai, as for https://github.com/espressif/esp-idf/issues/12207, it seems that this is another case of heap corruption. There might be some out of bound writing on the buffer you allocate in your application leading to the corruption of the metadata of the heap component (more precisely the tlsf data structures) that end up triggering this kind of assert.

SoucheSouche, 2023-10-12:
Closing issue since it does not report an ESP-IDF bug. The heap corruption encountered is introduced in the application code itself.

closed 2023-10-12 by SoucheSouche