espressif/esp-idf #12207: assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)

srikanthpalvai, 2023-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 S3 USB OTG

### Power Supply used.

USB

### What is the expected behavior?

lwip socket issue

### What is the actual behavior?

it should not throw assert 

### Steps to reproduce.

[sdkconfig.log](https://github.com/espressif/esp-idf/files/12552010/sdkconfig.log)

### Debug Logs.

```plain
Backtrace: 0x40376342:0x3fca9ba0 0x4037db0d:0x3fca9bc0 0x4038808d:0x3fca9be0 0x40383738:0x3fca9d00 0x40383122:0x3fca9d20 0x40376561:0x3fca9d40 0x403765c1:0x3fca9d60 0x403765f6:0x3fca9d80 0x40376fe8:0x3fca9da0 0x40377185:0x3fca9e20 0x4037a151:0x3fca9e40 0x42063947:0x3fca9e60 0x4206c193:0x3fca9e80 0x42069479:0x3fca9ea0 0x42008522:0x3fca9ec0 0x42008575:0x3fca9ef0 0x403815fe:0x3fca9f10
0x40376342: panic_abort at /home/lab/esp/esp-idf/components/esp_system/panic.c:423

0x4037db0d: esp_system_abort at /home/lab/esp/esp-idf/components/esp_system/esp_system.c:153

0x4038808d: __assert_func at /home/lab/esp/esp-idf/components/newlib/assert.c:78

0x40383738: block_locate_free at /home/lab/esp/esp-idf/components/heap/tlsf/tlsf.c:566
 (inlined by) tlsf_malloc at /home/lab/esp/esp-idf/components/heap/tlsf/tlsf.c:1004

0x40383122: multi_heap_malloc_impl at /home/lab/esp/esp-idf/components/heap/multi_heap.c:207

0x40376561: heap_caps_malloc_base at /home/lab/esp/esp-idf/components/heap/heap_caps.c:167

0x403765c1: heap_caps_malloc at /home/lab/esp/esp-idf/components/heap/heap_caps.c:187

0x403765f6: heap_caps_malloc_default at /home/lab/esp/esp-idf/components/heap/heap_caps.c:212

0x40376fe8: trace_malloc at /home/lab/esp/esp-idf/components/heap/include/heap_trace.inc:97

0x40377185: __wrap_malloc at /home/lab/esp/esp-idf/components/heap/include/heap_trace.inc:159

0x4037a151: wifi_malloc at /home/lab/esp/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:68

0x42063947: ieee80211_timer_process at ??:?
[... 217 lines cut ...]
.coredump.tasks.data 0x3fcb5014 0x160 RW 
.coredump.tasks.data 0x3fcb8530 0x2e0 RW 
.coredump.tasks.data 0x3fcc2278 0x160 RW 
.coredump.tasks.data 0x3fcc5510 0x1a20 RW 
.coredump.tasks.data 0x3fca7654 0x160 RW 
.coredump.tasks.data 0x3fca73c0 0x280 RW 
.coredump.tasks.data 0x3fcb1064 0x160 RW 
.coredump.tasks.data 0x3fcb0dd0 0x280 RW 
.coredump.tasks.data 0x3fcb2e7c 0x160 RW 
.coredump.tasks.data 0x3fcb2bb0 0x2c0 RW 
.coredump.tasks.data 0x3fca7cbc 0x160 RW 
.coredump.tasks.data 0x3fca7a30 0x280 RW 
.coredump.tasks.data 0x3fcb95dc 0x160 RW 
.coredump.tasks.data 0x3fcb9310 0x2c0 RW 
.coredump.tasks.data 0x3fcb4088 0x160 RW 
.coredump.tasks.data 0x3fcb3db0 0x2d0 RW 
.coredump.tasks.data 0x3fcbc124 0x160 RW 
.coredump.tasks.data 0x3fcbbe60 0x2b0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='74a81bf4'
Rebooting...
```

ESP-Marius, 2023-09-11:
@srikanthpalvai this kind of errors are usually caused by heap memory getting corrupted (overwritten by something in your app).

You can take a look at the [heap memory debugging guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html) to see a few ways you can debug this. I suggest trying to run with a higher level of [Heap Poisoning](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#configuration) first.

srikanthpalvai, 2023-09-13:
Thank you @ESP-Marius 
I will try and let you know the results

srikanthpalvai, 2023-09-15:
void convertByteArrayToHexString (char *target, int len)
{
    ESP_LOGD(TAG,  "convertByteArrayToHexString Start");
    char *val = NULL;
    printf("check 1\n");
    val =  (char *)malloc (len * 2);
    printf("check 2\n");
    if (val != NULL)
    {
        printf("check 3\n");
        int i, j;
        printf("check 4\n");
        for (i = 0, j = 0; i < len; ++i, j += 2)
        {
            printf("check 5\n");
            sprintf (val + j, "%02x", target[i] & 0xff);
            printf("check 6\n");
        }

        printf("check 7\n");
        memset (target, 0, strlen (target));
        printf("check 8\n");
        memcpy (target, val, len * 2);
        printf("check 9\n");
        **free (val);**
        printf("check 10\n");
    }
    else
    {
        ESP_LOGD(TAG,  "No memory available.");
        //log_debug("No memory available.");
    }
    ESP_LOGD(TAG,  "convertByteArrayToHexString End");
}

After deep debugging, this POC creating free issue 

check 5
check 6
check 7
check 8
check 9
CORRUPT HEAP: Bad tail at 0x3fcb0e66. Expected 0xbaad5678 got 0xbaad5600

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x4037630e:0x3fccbb30 0x4037dbc9:0x3fccbb50 0x40387941:0x3fccbb70 0x403832f5:0x3fccbc90 0x4037671a:0x3fccbcb0 0x4037719f:0x3fccbcd0 0x40377221:0x3fccbd20 0x4202028e:0x3fccbd40 0x4201cd96:0x3fccbd80 0x4201e08a:0x3fccbed0 0x4201ec29:0x3fccc720 0x4201be6b:0x3fccce70 0x4201c27b:0x3fcccf80 0x4201b0d5:0x3fcccfd0 0x420184cb:0x3fccd3f0 0x4200d802:0x3fccd4c0 0x403816ba:0x3fccd5e0
0x4037630e: panic_abort at /home/lab/esp/esp-idf/components/esp_system/panic.c:423

0x4037dbc9: esp_system_abort at /home/lab/esp/esp-idf/components/esp_system/esp_system.c:153

0x40387941: __assert_func at /home/lab/esp/esp-idf/components/newlib/assert.c:78

0x403832f5: multi_heap_free at /home/lab/esp/esp-idf/components/heap/multi_heap_poisoning.c:276 (discriminator 1)

0x4037671a: heap_caps_free at /home/lab/esp/esp-idf/components/heap/heap_caps.c:374

0x4037719f: trace_free at /home/lab/esp/esp-idf/components/heap/include/heap_trace.inc:119

0x40377221: __wrap_free at /home/lab/esp/esp-idf/components/heap/include/heap_trace.inc:164

0x4202028e: convertByteArrayToHexString at ??:?

Could you please suggest how to get rid of it

ESP-Marius, 2023-09-16:
> memcpy (target, val, len * 2);

Isnt the length of `target` `len` here? so you would be way outside the allocated area if you write `len * 2`.

I suggest going to e.g. our forum `www.esp32.com` for generic questions like this, the github issue tracker is mainly meant for reporting ESP-IDF related bugs.

SoucheSouche, 2023-10-12:
Closing issue since it does not report an ESP-IDF bug. The heap corruption encountered is introduced in the application code itself.

closed 2023-10-12 by SoucheSouche