espressif/esp-idf #12208: install.sh gets stuck right after installing openocd-esp32@v0.12.0-esp32-20230419

tpitman, 2023-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Copy these install instructions:

`git clone --recursive https://github.com/espressif/esp-idf.git`
`cd esp-idf; git checkout v5.1.1; git submodule update --init --recursive;`
`./install.sh`
`cd ..`

 from this page:

https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html

Paste into a terminal

Hit enter

I expect it to install the tools

### What is the actual behavior?

After this line of output in the terminal:

`Extracting /Users/tpitman/.espressif/dist/openocd-esp32-macos-arm64-0.12.0-esp32-20230419.tar.gz to /Users/tpitman/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419`

The terminal just hangs.

I hit Ctrl-C to stop it after many minutes, went into the esp-idf folder and ran ./install.sh and this was the output:

`Detecting the Python interpreter`
`Checking "python3" ...`
`Python 3.11.4`
`"python3" has been detected`
`Checking Python compatibility`
`Installing ESP-IDF tools`
`Current system platform: macos-arm64`
`Selected targets are: esp32c2, esp32c3, esp32, esp32s3, esp32h2, esp32c6, esp32s2`
`Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs`
`Skipping xtensa-esp-elf-gdb@12.1_20221002 (already installed)`
`Skipping riscv32-esp-elf-gdb@12.1_20221002 (already installed)`
`Skipping xtensa-esp32-elf@esp-12.2.0_20230208 (already installed)`
`Skipping xtensa-esp32s2-elf@esp-12.2.0_20230208 (already installed)`
`Skipping xtensa-esp32s3-elf@esp-12.2.0_20230208 (already installed)`
`Skipping riscv32-esp-elf@esp-12.2.0_20230208 (already installed)`
`Skipping esp32ulp-elf@2.35_20220830 (already installed)`
`Skipping openocd-esp32@v0.12.0-esp32-20230419 (already installed)`

At that point it gets stuck again. I have waited for 10 minutes.

### Steps to reproduce.

See the expected behavior above.

### Build or installation Logs.

```plain
NA
```

### More Information.

NA

dobairoland, 2023-09-08:
Hi @tpitman. Could you please try to run the following command to narrow down and confirm my suspicion? 

```sh
$IDF_PATH/tools/idf_tools.py install esp-rom-elfs
```

dobairoland, 2023-09-08:
(please also share the output)

Do you have only one ESP-IDF copy on your computer? If there are multiple one, then is IDF_PATH set to the right one?

tpitman, 2023-09-08:
I have started over and used the shallow repo method. I am running into other issues at this point, but this one doesn't matter anymore...

closed 2023-09-08 by tpitman