espressif/esp-idf #12213: export.sh hangs on Checking other ESP-IDF version.

tpitman, 2023-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected the export to get the environment ready for me to work with esp-matter.

I am following this page of instructions:

[https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html)

### What is the actual behavior?

The script shows the following output and then stays there forever. I let it go for 15 minutes.

### Steps to reproduce.

1. Use VS Code to install the esp-idf environment.
2. Follow the above mentioned URL step for "For MacOS host" to do the shallow install
3. At the step to cd into the esp-idf folder and execute source/export.sh, notice it stops at the Checking other ESP-IDF version. step

dobairoland, 2023-09-10:
This is likely the same issue you've experienced at https://github.com/espressif/esp-idf/issues/12208.

> The script shows the following output and then stays there forever. I let it go for 15 minutes.

You forgot to share the output.

dobairoland, 2023-09-11:
Are you even trying this natively on macOS? [This issue](https://github.com/espressif/esp-matter/issues/623) suggests that you are using a virtual machine.

dobairoland, 2023-09-11:
And [this report](https://github.com/project-chip/connectedhomeip/issues/29118) of yours suggest that you have issues with the connection speed. Maybe this is not stuck just very slow?

tpitman, 2023-09-11:
I am trying several different things to get something to work.

The latest thing I am trying is to use the VSCode extension for everything. It is a bit confusing to have tutorials that are outside of VSCode and then come to find there are commands built into the plugin that do the same thing.

I am learning as I go.

closed 2023-12-20 by dobairoland