espressif/esp-idf #12215: handling of Process ID Controllers (PID)

Talha-Babar, 2023-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf 5

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP Wroom 43

### Power Supply used.

USB

### What is the expected behavior?

 I'm focusing on the handling of Process ID Controllers (PID). Now, the PID is essentially a value that indicates the mode in which the system is operating.

Here's the situation broken down:

1) On boot-up, the ESP32 begins its operation in what is known as "kernel mode" with a PID set to 0. Think of the kernel mode as the core operating level of the system.

2) My objective is to transition the system to "user mode" by changing the PID to 4. User mode, as the name suggests, is a more restricted environment, typically used to run application tasks, keeping them separate from the core system tasks.

3) After completing certain tasks in user mode, I'd then like to revert the system back to the kernel mode by resetting the PID to 0.

### What is the actual behavior?

It get restarted as i reaches to this function:
REG_WRITEE(PIDCTRL_NMI_MASK_DISABLE_REG, 1);

### Steps to reproduce.

Code:

#include <stdio.h>
#include <stdbool.h>
#include <unistd.h>
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include <stdint.h>
#include "dport_access.h"

#define PID_CTRL_BASE_ADDR 0x3FF1F000

#define PIDCTRL_INTERRUPT_ENABLE_REG (PID_CTRL_BASE_ADDR + 0x000)
#define PIDCTRL_PID_DELAY_REG (PID_CTRL_BASE_ADDR + 0x020)
#define PIDCTRL_NMI_DELAY_REG (PID_CTRL_BASE_ADDR + 0x024)
#define PIDCTRL_PID_NEW_REG (PID_CTRL_BASE_ADDR + 0x048)
#define PIDCTRL_PID_CONFIRM_REG (PID_CTRL_BASE_ADDR + 0x04C)
#define PIDCTRL_NMI_MASK_ENABLE_REG (PID_CTRL_BASE_ADDR + 0x054)
#define PIDCTRL_NMI_MASK_DISABLE_REG (PID_CTRL_BASE_ADDR + 0x058)
[... 181 lines cut ...]
I (285) cpu_start: Chip rev:         v3.0
I (290) heap_init: Initializing. RAM available for dynamic allocation:
I (297) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (303) heap_init: At 3FFB2770 len 0002D890 (182 KiB): DRAM
I (309) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (316) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (322) heap_init: At 4008B9EC len 00014614 (81 KiB): IRAM
I (330) spi_flash: detected chip: generic
I (333) spi_flash: flash io: dio
W (337) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (351) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (361) APP_MAIN: Starting PID switch demo...
I (371) APP_MAIN: Switching to PID 4...
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6944
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?
```

Talha-Babar, 2023-09-11:
I have try to debug it more and found that when it cpu reaches to REG_WRITEE(PIDCTRL_PID_CONFIRM_REG, 1) in function void IRAM_ATTR step_6_to_8(). The MCU get restart.

mahavirj, 2023-10-09:
Please have a look at the [esp-privilege-separation](https://github.com/espressif/esp-privilege-separation) framework which should help you for the privilege based code execution. This framework is supported on ESP32-C3 and ESP32-S3. Regarding the PID controller in ESP32, unfortunately we do not support this peripheral. There were certain limitations associated with this peripheral and they have been addressed through newer world controller and permission control peripherals in ESP32-C3/ESP32-S3 SoCs. 

Hope this helps!

closed 2023-10-09 by mahavirj