espressif/esp-idf #12216: ESPNOW ersu send frame

MacWyznawca, 2023-09-10:
### Is your feature request related to a problem?

What it is and what the setting changes: **ESPNOW using ersu send frame**?

The documentation and the internet are silent on this subject.

### Describe the solution you'd like.

More info about ersu send frame.

Xiehanxin, 2023-09-11:
hi, @MacWyznawca , ERSU is a transmission mode related to 802.11 ax . another mode in 802.11 ax is SU. this setting means sending ESP-NOW frame is ERSU mode.  ERSU is always usde in long range transmission, and its frame has lower rate compared with SU mode

MacWyznawca, 2023-09-11:
Thanks, @Xiehanxin. So using this frame with ESP32-S3 and C3 connections only is useless?

Xiehanxin, 2023-09-11:
@MacWyznawca yes, it is the feature for ESP32-C6

MacWyznawca, 2023-09-11:
@Xiehanxin Can you add this explanation to the documentation? Because I haven't found anywhere.

zhangyanjiaoesp, 2024-06-03:
@MacWyznawca will update the doc ASAP.

closed 2024-10-28 by espressif-bot