espressif/esp-idf #12217: Host-Based Privacy Include Issue on Chips Other Than Standard ESP32

jdeanGL, 2023-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When using NimBLE on the ESP32-C3, an option for NimBLE host-based privacy should appear in menuconfig. Selecting this option enables flags in ble_hs_pvcy.h allowing privacy configuration.

### What is the actual behavior?

The option for host-based privacy no longer appears when setting an ESP32-C3 (or other non ESP32 standard chip) due to the following line in kconfig.in* in NimBLE host files:

`depends on BT_NIMBLE_ENABLED && IDF_TARGET_ESP32`

* Note: kconfig.in located invesp-idf\components\bt\host\nimble

### Steps to reproduce.

1. Attempt utilization `ble_hs_pvcy_rpa_config(NIMBLE_HOST_ENABLE_RPA);` in a project including host/ble_hs_pvcy.h" while setting an ESP32-C3 as the target. 
2. Fullclean the project.
3. Attempt building the project.

* Note: This cannot be remedied by a setting in menuconfig as the setting is no longer present for non ESP32 standard chips.

### Build or installation Logs.

```plain
PS C:\Users\jjd5t\Documents\GitHub\LITSLINK_App_Test_Firmware> idf.py build flash monitor
Executing action: all (aliases: build)
Running cmake in directory C:\Users\jjd5t\Documents\GitHub\LITSLINK_App_Test_Firmware\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=c:\esp\tools\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\jjd5t\Documents\GitHub\LITSLINK_App_Test_Firmware"...
-- Found Git: C:/esp/tools/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/esp/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/esp/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file C:/Users/jjd5t/Documents/GitHub/LITSLINK_App_Test_Firmware/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
[... 74 lines cut ...]
-- Build files have been written to: C:/Users/jjd5t/Documents/GitHub/LITSLINK_App_Test_Firmware/build/bootloader
[947/956] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
ccache C:\esp\tools\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/jjd5t/Documents/GitHub/LITSLINK_App_Test_Firmware/build/config -IC:/Users/jjd5t/Documents/GitHub/LITSLINK_App_Test_Firmware/main -IC:/Users/jjd5t/esp/esp-idf/components/newlib/platform_include -IC:/Users/jjd5t/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/jjd5t/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -IC:/Users/jjd5t/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/jjd5t/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/jjd5t/esp/esp-idf/components/freertos/esp_additions/arch/riscv/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_hw_support/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/jjd5t/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -IC:/Users/jjd5t/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -IC:/Users/jjd5t/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -IC:/Users/jjd5t/esp/esp-idf/components/heap/include -IC:/Users/jjd5t/esp/esp-idf/components/log/include -IC:/Users/jjd5t/esp/esp-idf/components/soc/include -IC:/Users/jjd5t/esp/esp-idf/components/soc/esp32c3 -IC:/Users/jjd5t/esp/esp-idf/components/soc/esp32c3/include -IC:/Users/jjd5t/esp/esp-idf/components/hal/esp32c3/include -IC:/Users/jjd5t/esp/esp-idf/components/hal/include -IC:/Users/jjd5t/esp/esp-idf/components/hal/platform_port/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_rom/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_rom/include/esp32c3 -IC:/Users/jjd5t/esp/esp-idf/components/esp_rom/esp32c3 -IC:/Users/jjd5t/esp/esp-idf/components/esp_common/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_system/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_system/port/soc -IC:/Users/jjd5t/esp/esp-idf/components/esp_system/port/include/riscv -IC:/Users/jjd5t/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/jjd5t/esp/esp-idf/components/riscv/include -IC:/Users/jjd5t/esp/esp-idf/components/lwip/include -IC:/Users/jjd5t/esp/esp-idf/components/lwip/include/apps -IC:/Users/jjd5t/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/jjd5t/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/jjd5t/esp/esp-idf/components/lwip/port/include -IC:/Users/jjd5t/esp/esp-idf/components/lwip/port/freertos/include -IC:/Users/jjd5t/esp/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/jjd5t/esp/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/jjd5t/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/jjd5t/esp/esp-idf/components/efuse/include -IC:/Users/jjd5t/esp/esp-idf/components/efuse/esp32c3/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/deprecated -IC:/Users/jjd5t/esp/esp-idf/components/driver/analog_comparator/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/dac/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/gpio/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/gptimer/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/i2c/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/i2s/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/ledc/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/mcpwm/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/parlio/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/pcnt/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/rmt/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/sdio_slave/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/sdmmc/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/sigma_delta/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/spi/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/temperature_sensor/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/touch_sensor/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/twai/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/uart/include -IC:/Users/jjd5t/esp/esp-idf/components/driver/usb_serial_jtag/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_pm/include -IC:/Users/jjd5t/esp/esp-idf/components/mbedtls/port/include -IC:/Users/jjd5t/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/jjd5t/esp/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/jjd5t/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_app_format/include -IC:/Users/jjd5t/esp/esp-idf/components/bootloader_support/include -IC:/Users/jjd5t/esp/esp-idf/components/bootloader_support/bootloader_flash/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_partition/include -IC:/Users/jjd5t/esp/esp-idf/components/app_update/include -IC:/Users/jjd5t/esp/esp-idf/components/esp_mm/include -IC:/Users/jjd5t/esp/esp-id [... cut ...]
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\jjd5t\Documents\GitHub\LITSLINK_App_Test_Firmware\build\log\idf_py_stderr_output_29900 and C:\Users\jjd5t\Documents\GitHub\LITSLINK_App_Test_Firmware\build\log\idf_py_stdout_output_29900
```

### More Information.

Adjusting the line `depends on BT_NIMBLE_ENABLED && IDF_TARGET_ESP32` in kconfig.in* to the following `depends on BT_NIMBLE_ENABLED && !IDF_TARGET_ESP32` allows for building and proper use of host-based privacy.

rahult-github, 2023-09-11:
Host based privacy is to be enabled by default for ESP32. For rest chips, we have controller based privacy enabled by default. So the change was done to make this correction.

jdeanGL, 2023-09-11:
I see, thank you for the explanation and sorry for the mistake!

jdeanGL, 2023-09-11:
Identified 'issue' not an issue and is intentional.

closed 2023-09-11 by jdeanGL