espressif/esp-idf #12218: ADC oneshot calibration max 3114 mV

Yardie-, 2023-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom ESP32-S3 board

### Power Supply used.

USB

### What is the expected behavior?

When a gpio is pulled high externally it should read after calibration 3300 mV.
Especially if the raw result is 4095

### What is the actual behavior?

version 5.1 produced
adc_gpio_read: ADC0 Channel[2] Raw Data: 4095
adc_gpio_read: ADC0 Channel[2] Calibrated Data: 3114
which was bad but 
since the upgrade to 5.1.1 
(14593) adc_gpio_read: ADC0 Channel[2] Raw Data: 4095
(14593) adc_gpio_read: ADC0 Channel[2] Calibrated Data: 3076

### Steps to reproduce.

please see https://esp32.com/viewtopic.php?f=13&t=35154
1. tie pin high 
2. read ADC with adc_oneshot_read(....
3. do adc_cali_raw_to_voltage(
result in millivolts as described above.

### Debug Logs.

```plain
adc_gpio_read: ADC0 Channel[2] Raw Data: 4095
adc_gpio_read: ADC0 Channel[2] Calibrated Data: 3114
which was bad but 
since the upgrade to 5.1.1 
(14593) adc_gpio_read: ADC0 Channel[2] Raw Data: 4095
(14593) adc_gpio_read: ADC0 Channel[2] Calibrated Data: 3076
```

caidamao, 2023-09-21:
ESP32 系列的ADC就是这样的，量程并不是0-3300mv，而是与adc配置有关。精度还比较飘

Yardie-, 2023-09-21:
I can see no reference to ADC configuration in the documentation.
It would be normal to expect a function called
adc_cali_raw_to_voltage
with Parameters
        handle – [in] ADC calibration handle
        raw – [in] ADC raw data
        voltage – [out] Calibrated ADC voltage (in mV)
would actually return a voltage close to that read by a multimeter.