espressif/esp-idf #12221: What are the best Wi-Fi settings for ESP-NOW in terms of connection stability and range?

MacWyznawca, 2023-09-11:
### Is your feature request related to a problem?

For ESP-NOW, when you use WIFI_PROTOCOL_LR, should you additionally enable WIFI_PROTOCOL_11B and WIFI_PROTOCOL_11G?

Is it worthwhile to additionally force the baud rate with esp_now_set_peer_rate_config()?

Under what conditions can the baud rate be changed? Does it require reinitialization of ESP-NOW?

lybgo, 2023-10-16:
I also want to know how to configure espnow to have the longest communication distance

zhangyanjiaoesp, 2023-10-17:
@MacWyznawca 

> For ESP-NOW, when you use WIFI_PROTOCOL_LR, should you additionally enable WIFI_PROTOCOL_11B and WIFI_PROTOCOL_11G?

The default rate for ESP-NOW is 1Mbps. The data rate for LR is 1/2 Mbps and 1/4 Mbps, if you want to use 802.11 b/g data rate, you should additionally enable WIFI_PROTOCOL_11B and WIFI_PROTOCOL_11G, otherwise don't need.

> Is it worthwhile to additionally force the baud rate with esp_now_set_peer_rate_config()?

 If you want to use other fixed rate (not 1Mbps)to send , then you can call `esp_now_set_peer_rate_config()`to set rate for each peer. If you don't use Wi-Fi 6, you can call `esp_wifi_config_espnow_rate()` to set rate for all ESP-NOW peers.

> Under what conditions can the baud rate be changed? Does it require reinitialization of ESP-NOW?

Whether or not to change the rate depends on your application. `esp_now_set_peer_rate_config()` should be called after wifi init and before espnow init https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/components/esp_wifi/include/esp_now.h#L274

lybgo, 2023-10-17:
I call esp_now_set_peer_rate_config() function  error, help to see what happened.
```c
static void example_wifi_init(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(ESPNOW_WIFI_MODE) );
    ESP_ERROR_CHECK( esp_wifi_set_ps(WIFI_PS_NONE));
    ESP_ERROR_CHECK( esp_wifi_start());
    ESP_ERROR_CHECK( esp_wifi_set_channel(CONFIG_ESPNOW_CHANNEL, WIFI_SECOND_CHAN_NONE));
    //ESP_ERROR_CHECK( esp_wifi_set_max_tx_power(80));//设置功率，反而距离和丢包率更短更大
//#if CONFIG_ESPNOW_ENABLE_LONG_RANGE
    //ESP_ERROR_CHECK( esp_wifi_set_protocol(ESPNOW_WIFI_IF, WIFI_PROTOCOL_11B|WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N|WIFI_PROTOCOL_LR) );
    ESP_LOGI(TAG, "set protocol LR");
    ESP_ERROR_CHECK( esp_wifi_set_protocol(ESPNOW_WIFI_IF, WIFI_PROTOCOL_LR) );
//#endif
    ESP_LOGI(TAG, "set protocol LR rate");
    ESP_ERROR_CHECK( esp_wifi_config_espnow_rate(ESPNOW_WIFI_IF, WIFI_PHY_RATE_LORA_250K));//减少速率，增加距离
    ESP_LOGI(TAG, "wifi init ok");
}
``` 
-------------------------------------------
errorInfo:
```
 (453) wifi_init: udp mbox: 6
I (453) wifi_init: tcp mbox: 6
I (453) wifi_init: tcp tx win: 5744
I (463) wifi_init: tcp rx win: 5744
I (463) wifi_init: tcp mss: 1440
I (473) wifi_init: WiFi IRAM OP enabled
I (473) wifi_init: WiFi RX IRAM OP enabled
I (483) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (533) wifi:mode : sta (60:55:f9:71:d4:30)
I (533) wifi:enable tsf
I (533) espnow_example: set protocol LR
I (533) wifi:ifx:0, phymode(new:0x3, nvs:0x3)
I (543) wifi:enable tsf
I (543) espnow_example: set protocol LR rate
W (543) wifi:invalid rate, need change phy mode to LR
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42007f90
0x42007f90: example_wifi_init at E:/esp32Dev/espIDF/espnow/main/espnow_example_main.c:61 (discriminator 1)

file: "./main/espnow_example_main.c" line 61
func: example_wifi_init
expression: esp_wifi_config_espnow_rate(ESPNOW_WIFI_IF, WIFI_PHY_RATE_LORA_250K)

abort() was called at PC 0x40386b31 on core 0
```

zhangyanjiaoesp, 2023-10-17:
@lybgo which IDF branch do you use ? If it's master branch, this issue has been fixed in cc01cc13aaf0de985504697d2f2d1df3fce1b037

lybgo, 2023-10-17:
> @lybgo which IDF branch do you use ? If it's master branch, this issue has been fixed in [cc01cc1](https://github.com/espressif/esp-idf/commit/cc01cc13aaf0de985504697d2f2d1df3fce1b037). And the fix has been backported to all branches.

Okay, thanks. I'll update it

MaxwellAlan, 2023-11-17:
Hi @lybgo does your issue still exist ?

Alvin1Zhang, 2024-03-22:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-03-22 by Alvin1Zhang

lnsane-G, 2024-09-16:
> 通信

我也想知道

lnsane-G, 2024-09-16:
> 我还想知道如何配置 espnow 以获得最长的通信距离

我也想知道