espressif/esp-idf #12222: Wrong flash size detected for `mxic` flash chip (128 kbytes v 8192 kbytes) with v5.1.1

RobMeades, 2023-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

u-blox NINA-W15

### Power Supply used.

USB

### What is the expected behavior?

Detect the 8 megabyte `mxic` flash chip used within the u-blox NINA-W15 module (ESP32) as being of size 8192 kbytes, e.g. as below, which is what version 5.0.1 did.

```
spi_flash: detected chip: mxic
spi_flash: flash io: dio
spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
cpu_start: Starting scheduler on PRO CPU.
cpu_start: Starting scheduler on APP CPU.
```

### What is the actual behavior?

With version 5.1.1, ESP-IDF now thinks that the 8 megabyte `mxic` flash chip only has 128 kbytes of flash:

```
spi_flash: detected chip: mxic
spi_flash: flash io: dio
spi_flash: Detected size(128k) smaller than the size in the binary image header(2048k). Probe failed.
```

Browsing changes, it seems plausible that this commit might have caused the problem:

https://github.com/espressif/esp-idf/commit/e94d95103e53679300942728108eb93fbaa81240

### Steps to reproduce.

Attempt to run a build made with ESP-IDF v5.1.1, as opposed to ESP-IDF v5.0.1, on a u-blox NINA-W15 (ESP32 with 8 Megabyte `mxic` chip) module.

### Debug Logs.

```plain
See above.
```

RobMeades, 2023-09-11:
Hmm, no, maybe not [that commit](https://github.com/espressif/esp-idf/commit/e94d95103e53679300942728108eb93fbaa81240), that would have been in v5.0.1.  Will try versions between v5.0.1 and v5.1.1 to find out where it got introduced.

RobMeades, 2023-09-11:
Binary-chopping indicates that v5.0.3 does NOT have the issue but v5.1 DOES have the issue.

mythbuster5, 2023-09-12:
I didn't remember any changes about this part from 5.0 to 5.1. Could you please get the mxic size by `esptool.py flash_id`.

RobMeades, 2023-09-12:
Hi, and thanks for the swift response.  `esptool.py` reports:

```
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e8:31:cd:da:6a:00
Uploading stub...
Running stub...
Stub running...
Manufacturer: c2
Device: 2537
Detected flash size: 8MB
```

In terms of `sdkconfig`, find it attached (renamed as `.txt`): lots of stuff in it but it is "boilerplate" as far as we are concerned, we haven't touched it in a long time.

[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/12584471/sdkconfig.defaults.txt)

RobMeades, 2023-09-26:
<cough />

RobMeades, 2023-10-14:
Any suggestions as to how to move forward with this?

michaelw-inno, 2024-03-08:
I have the same problem with the NINA-W106 10B-00 with 8MB flash. @RobMeades as you said IDF < 5.1 works, IDF >= 5.1 does not work. I investigated the issue today and I found out that there is a patch [4c66d7e](https://github.com/espressif/esp-idf/commit/4c66d7e6df1c916a30f0246fd96e3e465eb7b752) which implements a `size_detect` function for MXIC chips. This function correctly detects the size of the flash used in the u-blox chips. It can be cherry-picked to 5.1 to fix the issue as a workaround. (I only tried 5.1.2, but I assume it also works for other patch versions)

@espressif is there any plan to port this patch to the newer IDF versions?

closed 2024-03-13 by espressif-bot

RobMeades, 2024-03-13:
Woohoo!  Thank you @espressif!

RobMeades, 2024-03-13:
...and @michaelw-inno of course.