espressif/esp-idf #12225: USB HID Gamepad receiving data from the host

monurbstnc, 2023-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello I implemented a USB HID gamepad and I want to receive data from the host pc. How can I do that ? Is there any example for esp32
Here is my configuration and report descriptor.

-> Configuration Descriptor
static const uint8_t hid_configuration_descriptor[] = {
    // Configuration number, interface count, string index, total length, attribute, power in mA
    TUD_CONFIG_DESCRIPTOR(1, 1, 0, TUSB_DESC_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),
    TUD_HID_INOUT_DESCRIPTOR(0, 4, HID_ITF_PROTOCOL_NONE, sizeof(hid_report_descriptor), EPNUM_HID, 0x80 | EPNUM_HID, CFG_TUD_HID_EP_BUFSIZE, 10)
};

-> Report Descriptor
#define TUD_HID_REPORT_DESC_GAMEPAD(...) \
  HID_USAGE_PAGE ( HID_USAGE_PAGE_DESKTOP     )                 ,\
  HID_USAGE      ( HID_USAGE_DESKTOP_GAMEPAD  )                 ,\
  HID_COLLECTION ( HID_COLLECTION_APPLICATION )                 ,\
    /* Report ID if any */\
    __VA_ARGS__ \
    /* 8 bit X, Y, Z, Rz, Rx, Ry (min -127, max 127 ) */ \
    HID_USAGE_PAGE     ( HID_USAGE_PAGE_DESKTOP                 ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_X                    ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_Y                    ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_Z                    ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_RZ                   ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_RX                   ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_RY                   ) ,\
    HID_LOGICAL_MIN    ( 0x81                                   ) ,\
    HID_LOGICAL_MAX    ( 0x7f                                   ) ,\
    HID_REPORT_COUNT   ( 6                                      ) ,\
    HID_REPORT_SIZE    ( 8                                      ) ,\
    HID_INPUT          ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE ) ,\
    /* 8 bit DPad/Hat Button Map  */ \
    HID_USAGE_PAGE     ( HID_USAGE_PAGE_DESKTOP                 ) ,\
    HID_USAGE          ( HID_USAGE_DESKTOP_HAT_SWITCH           ) ,\
    HID_LOGICAL_MIN    ( 1                                      ) ,\
    HID_LOGICAL_MAX    ( 8                                      ) ,\
    HID_PHYSICAL_MIN   ( 0                                      ) ,\
    HID_PHYSICAL_MAX_N ( 315, 2                                 ) ,\
    HID_REPORT_COUNT   ( 1                                      ) ,\
    HID_REPORT_SIZE    ( 8                                      ) ,\
    HID_INPUT          ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE ) ,\
    /* 32 bit Button Map */ \
    HID_USAGE_PAGE     ( HID_USAGE_PAGE_BUTTON                  ) ,\
    HID_USAGE_MIN      ( 1                                      ) ,\
    HID_USAGE_MAX      ( 32                                     ) ,\
    HID_LOGICAL_MIN    ( 0                                      ) ,\
    HID_LOGICAL_MAX    ( 1                                      ) ,\
    HID_REPORT_COUNT   ( 32                                     ) ,\
    HID_REPORT_SIZE    ( 1                                      ) ,\
    HID_INPUT          ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE ) ,\
  HID_COLLECTION_END \

roma-jam, 2023-09-21:
Hi @monurbstnc, 

There is a possibility to send reports to the USB HID Device from the Host. 

HID specification provide several ways how to do that, depending on the task you want to complete. 
You can use SetFeature or SetReport for the Device and send a data to the Device. The data filed can be vary, but you can use it to send data you want to send. 
These Set* command should use the correct ReportID and has a correct data structure, that is described in Report Descriptor. 

Once, I had explained how we can send reports between devices, you can check it out https://github.com/espressif/esp-idf/issues/9992#issuecomment-1481200090
Probably, you can find answers there (how to deal with data inside transfer packets and how to add another ReportID general or any, which is support the OUT Reports (from Host to Device)). 

Before I can tell more, may I ask to clarify what type of Host do you have (PC or ESP32-based Host)? 

If there is anything I can help with, please feel free to ask.
Thanks.

monurbstnc, 2023-09-21:
Thanks for your reply,
I have a host PC and I want to read haptics data from PC.

roma-jam, 2023-09-21:
Hey @monurbstnc, 

Just to point out: depending on the host OS it could be that you need to make a driver on your own. 
But in your case that is possible to add another ReportID to ReportDescriptior with haptic Data description, assign new number  which would represent ReportID and use this ReportID number in SetFeature(ReportID, Data) to set specific feature to the device (like shaking). 

There is a SetReport() callback implementation in tusb_device example that you can easily check out: https://github.com/espressif/esp-idf/blob/3b748a6cb76c2db7c6368a0dea32a88bc58bc44d/examples/peripherals/usb/device/tusb_hid/main/tusb_hid_example_main.c#L84

Logically, SetFeature() would fit better, but use SetReport() for this task is also possible. 

If you have any further questions, feel free to ask or open an issue again in case there would be anything else.

closed 2023-09-21 by roma-jam

monurbstnc, 2023-09-24:
Thank you so much.

monurbstnc, 2023-11-01:
Hello, I made some changes but when I  plug the device to the windows. It says driver error. I observe that when I use below report descriptor I got this error .

const uint8_t hid_report_descriptor[] = {
   TUD_HID_REPORT_DESC_GENERIC_INOUT(64,HID_REPORT_ID(3)),
   TUD_HID_REPORT_DESC_GAMEPAD()
};

When I use without TUD_HID_REPORT_DESC_GENERIC_INOUT. It works OK but there is not any output endpoint. Still confused how to use tinyUSB and how to implement haptics 

Complete Code :

/*
 * SPDX-FileCopyrightText: 2022-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include <stdlib.h>
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "tinyusb.h"
#include "class/hid/hid_device.h"
#include "driver/gpio.h"
#include "soc/usb_reg.h"

#define APP_BUTTON (GPIO_NUM_0) // Use BOOT signal by default
static const char *TAG = "example";

/************* TinyUSB descriptors ****************/

#define TUSB_DESC_TOTAL_LEN      (TUD_CONFIG_DESC_LEN + CFG_TUD_HID * TUD_HID_DESC_LEN)

/**
 * @brief HID report descriptor
 *
 * In this example we implement Keyboard + Mouse HID device,
 * so we must define both report descriptors
 */
const uint8_t hid_report_descriptor[] = {
   TUD_HID_REPORT_DESC_GENERIC_INOUT(64,HID_REPORT_ID(3)),
   TUD_HID_REPORT_DESC_GAMEPAD()
};

/**
 * @brief String descriptor
 */
const char* hid_string_descriptor[5] = {
    // array of pointer to string descriptors
    (char[]){0x09, 0x04},  // 0: is supported language is English (0x0409)
    "Example",             // 1: Manufacturer
    "Example Device",      // 2: Product
    "1",              // 3: Serials, should use chip ID
    "Example HID interface",  // 4: HID
};

/**
 * @brief Configuration descriptor
 *
 * This is a simple configuration descriptor that defines 1 configuration and 1 HID interface
 */
static const uint8_t hid_configuration_descriptor[] = {
    // Configuration number, interface count, string index, total length, attribute, power in mA
    TUD_CONFIG_DESCRIPTOR(1, 1, 0, TUSB_DESC_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),

    // Interface number, string index, boot protocol, report descriptor len, EP In address, size & polling interval
    TUD_HID_DESCRIPTOR(0, 4, false, sizeof(hid_report_descriptor), 0x81, 16, 10),
};

/********* TinyUSB HID callbacks ***************/

// Invoked when received GET HID REPORT DESCRIPTOR request
// Application return pointer to descriptor, whose contents must exist long enough for transfer to complete
uint8_t const *tud_hid_descriptor_report_cb(uint8_t instance)
{
    // We use only one interface and one HID report descriptor, so we can ignore parameter 'instance'
    return hid_report_descriptor;
}

// Invoked when received GET_REPORT control request
// Application must fill buffer report's content and return its length.
// Return zero will cause the stack to STALL request
uint16_t tud_hid_get_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t* buffer, uint16_t reqlen)
{
  (void) instance;
  (void) report_id;
  (void) report_type;
  (void) buffer;
  (void) reqlen;

  return 0;
}

// Invoked when received SET_REPORT control request or
// received data on OUT endpoint ( Report ID = 0, Type = 0 )
void tud_hid_set_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize)
{
    uint8_t frame[64] = {0};
    while (1) {
        printf("Seq: %u", frame[0]++);
            while(!tud_hid_n_ready(0));
            tud_hid_n_report(0, 0, frame, 64);
    }
}

/********* Application ***************/

static void app_send_hid_demo(void)
{
    
    ESP_LOGI(TAG, "Sending Gamepad report");
    tud_hid_gamepad_report(4,-10,33,-44,22,11,85,2,2); 
}

void app_main(void)
{
 

    ESP_LOGI(TAG, "USB initialization");
    const tinyusb_config_t tusb_cfg = {
        .device_descriptor = NULL,
        .string_descriptor = hid_string_descriptor,
        .string_descriptor_count = sizeof(hid_string_descriptor) / sizeof(hid_string_descriptor[0]),
        .external_phy = false,
        .configuration_descriptor = hid_configuration_descriptor,
    };

    ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));
    ESP_LOGI(TAG, "USB initialization DONE");

    while (1) {
        if (tud_mounted()) {
            app_send_hid_demo();
            
        }
        vTaskDelay(pdMS_TO_TICKS(100));
    }
}