espressif/esp-idf #12226: When Flahsing ESP32, I got an Error msg "CMake Error at run_serial_tool.cmake:50"

AJ-JunghoAn, 2023-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP_IDF 4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I was trying to debugging with ESP Guide and searching a similar issue however it won't work for me.
Help me out friends and I really appreciate.

### What is the actual behavior?

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:50 (message):
  C:/Espressif/python_env/idf4.3_py3.11_env/Scripts/python.exe
  C:/Espressif/frameworks/esp-idf-v4.3.5/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

### Steps to reproduce.

1. Excute ESP-IDF 4.3 PowerShell
2. going to project folder -> C:\Espressif\frameworks\esp-idf-v4.3.5> cd .\udp_serial_server\
3. Flash it -> C:\Espressif\frameworks\esp-idf-v4.3.5> cd .\udp_serial_server\
4. After build successfully, got the error msg during flash. 

### Build or installation Logs.

```plain
PS C:\Espressif\frameworks\esp-idf-v4.3.5> cd .\udp_serial_server\
PS C:\Espressif\frameworks\esp-idf-v4.3.5\udp_serial_server> idf.py -p COM3 -b 115200 flash monitor
Executing action: flash
Running cmake in directory c:\espressif\frameworks\esp-idf-v4.3.5\udp_serial_server\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\espressif\frameworks\esp-idf-v4.3.5\udp_serial_server"...
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
[... 183 lines cut ...]
Successfully created esp32 image.
Generated C:/Espressif/frameworks/esp-idf-v4.3.5/udp_serial_server/build/udp_serial_server.bin
[1004/1005] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-...sp-idf-v4.3.5/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 udp_serial_server.bin
esptool.py v3.3.3-dev
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:50 (message):
  C:/Espressif/python_env/idf4.3_py3.11_env/Scripts/python.exe
  C:/Espressif/frameworks/esp-idf-v4.3.5/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.3.5\components\esptool_py && C:\Espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="C:/Espressif/frameworks/esp-idf-v4.3.5" -D SERIAL_TOOL="C:/Espressif/python_env/idf4.3_py3.11_env/Scripts/python.exe C:/Espressif/frameworks/esp-idf-v4.3.5/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Espressif/frameworks/esp-idf-v4.3.5/udp_serial_server/build" -P C:/Espressif/frameworks/esp-idf-v4.3.5/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
PS C:\Espressif\frameworks\esp-idf-v4.3.5\udp_serial_server>
```

### More Information.

PC Window 10

radimkarnis, 2023-09-12:
Hello @AJ-JunghoAn,

the real issue here is this:

```
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Everything works well up to the point where flashing begins. Your devkit is not automatically switching into the download mode. You can read about the issue [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader).

You can switch the chip into the download mode manually by pressing and holding the `BOOT` button and then pressing the `RST` button. Or try to hold the `BOOT` button when flashing begins (if your devkit has these buttons).

Please try that and let me know if that helped.

AJ-JunghoAn, 2023-09-14:
@radimkarnis Thanks for replying.

By the way, my own board which doesn't have these buttons but it has automatic bootloader (pull-up). I have reserarched and somebodody talked that I need to control RTS and DTR in esptool.py. 

[image]

Would you explain this solution to get into boot-mode?

Thank you.
Jungho An

dobairoland, 2023-09-18:
> Would you explain this solution to get into boot-mode?

Everything is explained at https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html. Please read it.

closed 2023-09-18 by dobairoland