espressif/esp-idf #12228: Migration from 4.4.3 to 5.1.1

damian-kurek-wizzdev, 2023-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It compiles

### What is the actual behavior?

Nimble BLE Is missing some files 
/port/include/esp_nimble_mem.h:18:10: fatal error: bt_osi_mem.h: No such file or directory
   18 | #include "bt_osi_mem.h"
      |          ^~~~~~~~~~~~~~

### Steps to reproduce.

idf.py build

### Build or installation Logs.

```plain
/port/include/esp_nimble_mem.h:18:10: fatal error: bt_osi_mem.h: No such file or directory
   18 | #include "bt_osi_mem.h"
      |          ^~~~~~~~~~~~~~
```

### More Information.

I'm porting the existing project from idf version 4.4.3 to 5.1.1 and run into some problems. I described one with bt.

rahult-github, 2023-09-12:
Hi @damian-kurek-wizzdev , 

The 5.1.1 compiles cleanly.  The file mentioned as missing is actually present at --> ./components/bt/porting/mem/bt_osi_mem.h 

Please do a clean pull / clone of IDF. 

If you are manually making changes, then please ensure all needed files are pulled in correctly. You can start with CMakelists.txt file as seems some path may not be added in your setup resulting in above issue.

damian-kurek-wizzdev, 2023-09-12:
I checked it with fresh esp-idf. 
I can see the file in components/bt....
I added bt to COMPONENT_REQUIRES.
I tried to clean the build directory. 
Starting with fresh CMAKE will be very hard. I can check if an example for nimble ble compiles

rahult-github, 2023-09-12:
> I can check if an example for nimble ble compiles

You can try to compile existing nimble examples in fresh esp-idf clone and confirm if they are working.
Then you can try the same examples on your port. If that doesn't compile, it means some environment issue in the setup.

damian-kurek-wizzdev, 2023-09-12:
[logs.txt](https://github.com/espressif/esp-idf/files/12585955/logs.txt)
Clean esp-idf 5.1.1 random example + adding a header that worked with 4.4.3 to it.

damian-kurek-wizzdev, 2023-09-12:
OK, I figured it out. It landed in private including the dirs of the bt component. I will just patch it for my project.

rahult-github, 2023-09-12:
Hi @damian-kurek-wizzdev  ,

You are trying to compile ble_multi_conn_cent application for esp32 .  This application is not valid for esp32.   

You can try compiling bleprph example for esp32. Attached is log for clean compilation on 5.1 branch for same. 
[success_log.txt](https://github.com/espressif/esp-idf/files/12586134/success_log.txt)

I looked at your logs. I think some changes are missing in your CMakelists.txt . Can you please share your components/bt/CMakelists.txt file ?

damian-kurek-wizzdev, 2023-09-12:
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/12586466/CMakeLists.txt)

rahult-github, 2023-09-12:
Hi @damian-kurek-wizzdev  , 

  I observe you have updated the porting/mem/ include directory in CMakelistst.txt ( which includes the header file being pointed in the warning) .  Since you are modifying the IDF , you need to ensure that files are correctly present and modifications done correctly, to avoid such issues. This i think is more of a environment/ setup issue rather than release 5.1.1 issue.

damian-kurek-wizzdev, 2023-09-12:
1 more question regarding migration. Do you know if it is planned to support esp AWS IoT to 5.1 release of idf? I looked into it for 2-3 hours and did not make it work yet.
https://github.com/espressif/esp-aws-iot

rahult-github, 2023-09-13:
>   Do you know if it is planned to support esp AWS IoT to 5.1 esp-idf release of idf

I checked with my colleagues, and currently this is being worked upon.  The estimate for having the support is planned to be by mid next month.

damian-kurek-wizzdev, 2023-09-13:
OK, Thanks for info @rahult-github

closed 2023-09-13 by damian-kurek-wizzdev

dhavalgujar, 2023-10-02:
Hi @damian-kurek-wizzdev,
This issue was addressed in one of the recent commits ([69f9995](https://github.com/espressif/esp-aws-iot/commit/69f9995b1b1d001f4c1e9624695227c1063ab513)) in the master branch of esp-aws-iot.  

Hope this helps.