espressif/esp-idf #12229: How to subscribe Group address in BLE Mesh In esp32 without mobile app

ShahinHaque97-CT, 2023-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

New issue as the https://github.com/espressif/esp-idf/issues/9677 is closed. 

I am having trouble subscribing a model to a group. 

I have an ESP32C3 as a provisioner/client , nRF52832 as server. 
I have got them running and temperature is being sent from the server to the client no issue.

However, I would now like to have the server publish to a group address.

However the models are not found, 

AT-BMNS_TEST is just the LOG name
`
I (17102) BMNS_COMMON: ********************** Composition Data Start **********************
I (17112) BMNS_COMMON: * CID 0x0059, PID 0x0000, VID 0x0000, CRPL 0x000a, Features 0x0007 *
I (17122) BMNS_COMMON: * Loc 0x0001, NumS 0x03, NumV 0x00 *
I (17132) BMNS_COMMON: * SIG Model ID 0x0000 *
I (17132) BMNS_COMMON: * SIG Model ID 0x1100 *
I (17142) BMNS_COMMON: * SIG Model ID 0x1101 *
I (17142) BMNS_COMMON: *********************** Composition Data End ***********************
I (17162) BMNS_PROV: ESP_BLE_MESH_PROVISIONER_STORE_NODE_COMP_DATA_COMP_EVT, err_code 0
W (17382) BMNS_CONFIG: Provision and config successfully VENDOR
I (32212) AT-BMNS_TEST: argument count 1
I (32212) AT-BMNS_TEST: primary address 1
I (32212) AT-BMNS_TEST: element_address 1
I (32222) AT-BMNS_TEST: sig_model_count 3
I (32222) AT-BMNS_TEST: vnd_model_count 1
I (32232) AT-BMNS_TEST: [0] sig_models->model_id 0x0000
I (32232) AT-BMNS_TEST: [1] sig_models->model_id 0x0001
I (32242) AT-BMNS_TEST: [2] sig_models->model_id 0x1102
E (32242) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x0000
E (32252) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x0001
E (32262) BLE_MESH: Subscribe, model not found, cid 0x0059, mod_id 0x1102

`

gnameg, 2023-09-22:
Hi, 
    It seems that there are no model id such as 0x0001 and 0x1102 according to the **Composition Data**.  If you want to subscribe a SIG model, you may not specify the CID. But, if you want to subscribe a vendor model, the **Composition Data** may be not correct. Here is the contents of **Composition Data**:
[image]

chegewara, 2024-03-27:
The logs are actually very confusing.
Some logs are saying there is 3 SIG models and 1 VND model, other logs are saying there is 0 VND models.
It looks like part of logs are from one device and another part of logs from different device.
At first i thought it may be from 2 elements, but it cant be true, because we cant have "config server model" on non-root element. Different models, but in both cases there seems to be SIG 0x0000 model:
```
I (17132) BMNS_COMMON: * SIG Model ID 0x0000 *
I (17132) BMNS_COMMON: * SIG Model ID 0x1100 *
I (17142) BMNS_COMMON: * SIG Model ID 0x1101 *
```
and here
```
I (32232) AT-BMNS_TEST: [0] sig_models->model_id 0x0000
I (32232) AT-BMNS_TEST: [1] sig_models->model_id 0x0001
I (32242) AT-BMNS_TEST: [2] sig_models->model_id 0x1102
```
And here we can see VND model and NumV is 0:
```
I (17112) BMNS_COMMON: * CID 0x0059, PID 0x0000, VID 0x0000, CRPL 0x000a, Features 0x0007 *
I (17122) BMNS_COMMON: * Loc 0x0001, NumS 0x03, NumV 0x00 *
```

It would be the best to see raw composition data.