espressif/esp-idf #12232: ESP32 Auto Light Sleep

EUA, 2023-09-12:
### Is your feature request related to a problem?

Use of forced Light Sleep leads connection drops on ESP32 as expected.
So please get us Auto Light Sleep.
ESP8266 has such a feature but ESP32 still missing...
THanks

sudeep-mohanty, 2023-09-12:
We do support automatic light sleep for ESP32. You can find details about how to use it [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html#power-management).

EUA, 2023-09-13:
Oops. Wrong project. I wanted to send into Arduino project instead of here. Thanks!

closed 2023-09-13 by EUA