espressif/esp-idf #12236: BLE Client (NimBLE) not receiving data from BLE Server (NimBLE)

Firmware-beginner, 2023-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 I have tested the server code on esp32s3 board and connected it to a mobile phone and data transfer takes place from both sides. I have flashed client code in one esp32s3 board and server code into another esp32s3 board, connection gets established between them and client sends data but it can't receive data. What could be the problem?

rahult-github, 2023-09-13:
Hi @Firmware-beginner ,

Can you please give more information :
1. What idf  version / branch is being used .
2. Which nimble example is being used ?
3. Can you share a debug enabled log for your run ?

Firmware-beginner, 2023-09-13:
1. ESP-IDF v5.1-4
2. espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_spp
/spp_client and spp_server example
3.
I (593) NimBLE: Connection established
I (593) NimBLE:

I (593) NimBLE: GATT procedure initiated: discover all services

I (703) NimBLE: GATT procedure initiated: discover all characteristics;
I (703) NimBLE: start_handle=1 end_handle=5

I (903) NimBLE: GATT procedure initiated: discover all characteristics;
I (903) NimBLE: start_handle=6 end_handle=9

I (1103) NimBLE: GATT procedure initiated: discover all characteristics;
I (1103) NimBLE: start_handle=10 end_handle=65535

I (1303) NimBLE: GATT procedure initiated: discover all descriptors;
I (1303) NimBLE: chr_val_handle=8 end_handle=9

I (1403) NimBLE: GATT procedure initiated: discover all descriptors;
I (1403) NimBLE: chr_val_handle=12 end_handle=65535

I (1603) NimBLE: Service discovery complete; status=0 conn_handle=1

I (1603) NimBLE: GAP procedure initiated: discovery;
I (1603) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1
I (1613) NimBLE: duration=forever
I (1613) NimBLE:

I (3313) NimBLE: GATT procedure initiated: write;
I (3313) NimBLE: att_handle=12 len=1

I (3313) NimBLE_SPP_BLE_CENT: Write in uart task success!
I (6643) NimBLE: GATT procedure initiated: write;
I (6643) NimBLE: att_handle=12 len=1

I (6643) NimBLE_SPP_BLE_CENT: Write in uart task success!
I (9263) NimBLE: GATT procedure initiated: write;
I (9263) NimBLE: att_handle=12 len=2

rahult-github, 2023-09-13:
I gave a quick try for the application at mentioned version and see it is working correctly. 

Shared above is console log of client side. Can you please do below and share log of both client and server side:

Component Config  --> Log Output --> Default Log Verbosity --> Debug
Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs

Firmware-beginner, 2023-09-13:
Log of client:

I (1505) NimBLE: Connection established
D (1515) NimBLE: handle=1 our_ota_addr_type=0 our_ota_addr=f4:12:fa:81:4c:5e
D (1525) NimBLE: our_id_addr_type=0 our_id_addr=f4:12:fa:81:4c:5e
D (1525) NimBLE: peer_ota_addr_type=0 peer_ota_addr=34:85:18:8e:b6:a6
D (1535) NimBLE: peer_id_addr_type=0 peer_id_addr=34:85:18:8e:b6:a6
D (1535) NimBLE: conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypted=0 authenticated=0 bonded=0
I (1545) NimBLE:

I (1555) NimBLE: GATT procedure initiated: discover all services

D (1555) NimBLE: host tx hci data; handle=1 length=11

D (1565) NimBLE: ble_hs_hci_acl_tx():
D (1565) NimBLE: 0x01
D (1575) NimBLE: 0x00
D (1575) NimBLE: 0x0b
D (1575) NimBLE: 0x00
D (1575) NimBLE: 0x07
D (1575) NimBLE: 0x00
D (1585) NimBLE: 0x04
D (1585) NimBLE: 0x00
D (1585) NimBLE: 0x10
D (1585) NimBLE: 0x01
D (1595) NimBLE: 0x00
D (1595) NimBLE: 0xff
D (1595) NimBLE: 0xff
D (1595) NimBLE: 0x00
D (1605) NimBLE: 0x28
D (1605) NimBLE:

D (1605) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2

D (1615) NimBLE: 0x16
D (1615) NimBLE: 0x20
D (1615) NimBLE: 0x02
D (1615) NimBLE: 0x01
D (1625) NimBLE: 0x00
D (1625) NimBLE:

D (1625) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e
D (1635) NimBLE: subevent: 0x14
D (1635) NimBLE:

D (1635) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e
D (1645) NimBLE: subevent: 0x4
D (1645) NimBLE:

D (1715) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=24 data=
D (1715) NimBLE: 0x14
D (1715) NimBLE: 0x00
D (1715) NimBLE: 0x04
D (1715) NimBLE: 0x00
D (1715) NimBLE: 0x11
D (1715) NimBLE: 0x06
D (1725) NimBLE: 0x01
D (1725) NimBLE: 0x00
D (1725) NimBLE: 0x05
D (1725) NimBLE: 0x00
D (1735) NimBLE: 0x00
D (1735) NimBLE: 0x18
D (1735) NimBLE: 0x06
D (1735) NimBLE: 0x00
D (1745) NimBLE: 0x09
D (1745) NimBLE: 0x00
D (1745) NimBLE: 0x01
D (1745) NimBLE: 0x18
D (1755) NimBLE: 0x0a
D (1755) NimBLE: 0x00
[... 969 lines cut ...]
Log of server:

I (220570) NimBLE: connection established; status=0
I (220570) NimBLE: handle=1 our_ota_addr_type=0 our_ota_addr=
I (220570) NimBLE: 34:85:18:8e:b6:a6
I (220570) NimBLE:  our_id_addr_type=0 our_id_addr=
I (220580) NimBLE: 34:85:18:8e:b6:a6
I (220580) NimBLE:  peer_ota_addr_type=0 peer_ota_addr=
I (220590) NimBLE: f4:12:fa:81:4c:5e
I (220590) NimBLE:  peer_id_addr_type=0 peer_id_addr=
I (220600) NimBLE: f4:12:fa:81:4c:5e
I (220600) NimBLE:  conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypte                            d=0 authenticated=0 bonded=0

I (220610) NimBLE:

I (220620) NimBLE: GAP procedure initiated: advertise;
I (220620) NimBLE: disc_mode=2
I (220630) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_it                            vl_min=0 adv_itvl_max=0
I (220640) NimBLE:

I (229570) NimBLE: Data received in write event,conn_handle = 1,attr_handle = c
I (234820) NimBLE: Data received in write event,conn_handle = 1,attr_handle = c
I (266370) NimBLE: Data received in write event,conn_handle = 1,attr_handle = c

On the server side I am looping(sending) back the same data which the client sends. So client should receive what it sends.

rahult-github, 2023-09-13:
> On the server side I am looping(sending) back the same data which the client sends. So client should receive what it sends.

The default application doesn't do this. So ,

1.  in Gatt, client queries and server responds. 
2. When you use default application, does server get data ?
3.  It is now mentioned that you modified server side code to attempt something.  Not sure what changes you have done. So please check if the changes you are attempting are valid or not ?

Firmware-beginner, 2023-09-13:
3. I have mentioned that I tested it by connecting mobile phone to esp board (server).
 Connection was established and I could send data from a mobile phone (application) and receive that data back on mobile.

rahult-github, 2023-09-13:
As requested , please share server side debug enabled logs and the changes done on server side

rahult-github, 2023-09-13:
Can you please share the server side changes. It will help to reproduce issue locally and debug.

Firmware-beginner, 2023-09-14:
Hi @rahult-github 

In Callback function for custom service (ble_svc_gatt_handler)
at case BLE_GATT_ACCESS_OP_WRITE_CHR:
trying to access received data and send it to connected client

/* Callback function for custom service */
static int  ble_svc_gatt_handler(uint16_t conn_handle, uint16_t attr_handle, struct ble_gatt_access_ctxt *ctxt, void *arg)
{
	int rc = 0;
	
    switch (ctxt->op) {
    case BLE_GATT_ACCESS_OP_READ_CHR:
        MODLOG_DFLT(INFO, "Callback for read");		
        break;

    case BLE_GATT_ACCESS_OP_WRITE_CHR:
        MODLOG_DFLT(INFO, "Data received in write event,conn_handle = %x,attr_handle = %x", conn_handle, attr_handle);
		
#if(1)	/* BLE Loopback code */	
		for (int i = 0; i <= CONFIG_BT_NIMBLE_MAX_CONNECTIONS; i++) {
			/* Check if client has subscribed to notifications */
			if (conn_handle_subs[i]) {
				struct os_mbuf *txom;
				txom = ble_hs_mbuf_from_flat(ctxt->om->om_data, ctxt->om->om_len); //MODIFIED
				rc = ble_gatts_notify_custom(i, ble_spp_svc_gatt_read_val_handle,
											 txom);
				if (rc == 0) {
					MODLOG_DFLT(INFO, "Notification sent successfully");
				} else {
					MODLOG_DFLT(INFO, "Error in sending notification rc = %d", rc);
				}
			}
		}
#endif
        break;

    default:
        MODLOG_DFLT(INFO, "\nDefault Callback");
        break;
    }
    return 0;
}

Firmware-beginner, 2023-09-15:
Thank you for helping,
It works 👍

closed 2023-09-15 by rahult-github