espressif/esp-idf #12239: Object destructors are skipped if test fails

herculanodavi, 2023-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When using the Unity Test Framework to run unit tests, the destructors of the objects in the scope are not called upon a test assert fail. This leads to a memory leak, and can both impact the validity of future tests and can lead to a device crash.

When it leads to a device crash, the pytest utility is not able to process the reset and waits for the result until timeout. 

I wrote a small reproduction of this problem [in this repo](https://github.com/herculanodavi/esp32-unity-destructor). There, I have a RAII class that reserves a global resource on construction and frees it on destruction, and a factory function that returns a RAII optional that has a value when the resource is free to take. Then, running these two tests leads to two failures:

```c++
TEST_CASE("resource leak", "[raii]")
{
  auto raii = GetRAII();
  TEST_FAIL_MESSAGE("leaking resource");
  /* constructor is not called here */
}

TEST_CASE("getting resource", "[raii]")
{
  auto raii = GetRAII();
  /* constructor was not called above so resource is unavailable */
  TEST_ASSERT_MESSAGE(raii.has_value(), "RAII initialization failed!!");
}
```

The RAII factory:
```c++
std::optional<RAII> GetRAII()
{
  // Fail if resource is reserved
  if (RAII::resourceState == RAII::ResourceState::RESERVED) {
    return std::nullopt;
  }

  // Reserve resource and return RAII if free
  RAII::resourceState = RAII::ResourceState::RESERVED;
  return RAII{};
}
```

I think I'm maybe not using the framework in a proper way, so I come to ask you: how is the proper way to do unit testing with C++ and ESP-IDF?

igrr, 2023-09-13:
Unfortunately this is just how Unity works — when test fails it uses `longjmp` C function to jump from within the test back into the test runner. `setjmp`/`longjmp` is a rather low-level mechanism which isn't aware of C++ constructors/destructors.

It might be possible to use Catch2 C++ test framework on the target, it won't have this issue. However at the moment we only have tests where Catch2 is used in host tests, I think.

> the pytest utility is not able to process the reset and waits for the result until timeout

This specific issue might be something we could fix in pytest-embedded.

herculanodavi, 2023-09-13:
Thanks for such a fast answer! 

Would you be able to point me to any code or documentation that uses Catch2 for the host tests?

igrr, 2023-09-13:
For example, this one, and other tests under host_tests in the same repository: https://github.com/espressif/esp-idf-cxx/blob/main/host_test/esp_timer/main/esp_timer_test.cpp

hfudev, 2023-09-14:
Hi @herculanodavi 

Regarding your request here:

> the pytest utility is not able to process the reset and waits for the result until timeout.

The feature is supported with test function [`run_all_single_board_cases`](https://docs.espressif.com/projects/pytest-embedded/en/latest/api.html#pytest_embedded_idf.unity_tester.IdfUnityDutMixin.run_all_single_board_cases)

It would collect all the test cases first, if any error or crash happened during a test case, the dut would be reset, and run the following test case continuously.

You may find an example [here](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test_apps/i2c_lcd/pytest_i2c_lcd.py#L18)

herculanodavi, 2023-09-15:
Thanks for your support. I'm closing this as we have enough information now.

closed 2023-09-15 by herculanodavi

igrr, 2023-09-16:
By the way, when this PR is merged, Catch2 will be available in IDF component registry. The included examples work both in host and on hardware.
https://github.com/espressif/idf-extra-components/pull/238