espressif/esp-idf #12243: E (127) esp_image: Segment 0 0x3fcd5820-0x3fcd6f2c invalid: overlaps bootloader stack E (135) boot: OTA app partition slot 1 is not bootable

LHYYSY, 2023-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-13

### Power Supply used.

USB

### What is the expected behavior?

run the new firmware after the ota upgrade is successfu

### What is the actual behavior?

I (35) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (35) boot: compile time Sep 14 2023 14:10:48
I (35) boot: chip revision: v0.3
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (43) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00010000
I (77) boot:  1 otadata          OTA data         01 00 00019000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0001b000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 001e0000
I (99) boot:  4 ota_1            OTA app          00 11 00200000 001e0000
I (106) boot:  5 fctry            WiFi data        01 02 003e0000 00010000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=00200020 vaddr=3fcd5820 size=0170ch (  5900) load
E (127) esp_image: Segment 0 0x3fcd5820-0x3fcd6f2c invalid: overlaps bootloader stack
E (135) boot: OTA app partition slot 1 is not bootable

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
I (118684) wifi:flush txq
I (118685) wifi:stop sw txq
I (118688) wifi:lmac stop hw txq
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
0x40048b82: ets_secure_boot_verify_bootloader_with_keys in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x170c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (35) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (35) boot: compile time Sep 14 2023 14:10:48
I (35) boot: chip revision: v0.3
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (43) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00010000
I (77) boot:  1 otadata          OTA data         01 00 00019000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0001b000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 001e0000
I (99) boot:  4 ota_1            OTA app          00 11 00200000 001e0000
I (106) boot:  5 fctry            WiFi data        01 02 003e0000 00010000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=00200020 vaddr=3fcd5820 size=0170ch (  5900) load
E (127) esp_image: Segment 0 0x3fcd5820-0x3fcd6f2c invalid: overlaps bootloader stack
E (135) boot: OTA app partition slot 1 is not bootable
I (141) esp_image: segment 0: paddr=00020020 vaddr=3c150020 size=491b8h (299448) map
I (197) esp_image: segment 1: paddr=000691e0 vaddr=3fc95200 size=0366ch ( 13932) load
I (200) esp_image: segment 2: paddr=0006c854 vaddr=40380000 size=037c4h ( 14276) load
I (206) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=147b94h (1342356) map
I (426) esp_image: segment 4: paddr=001b7bbc vaddr=403837c4 size=11968h ( 72040) load
I (446) boot: Loaded app from partition at offset 0x20000
I (446) boot: Disabling RNG early entropy source...
```

### More Information.

oat_file.csv:
nvs,      data, nvs,     ,        0x10000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
ota_0,    app,  ota_0,   ,        0x1E0000,
ota_1,    app,  ota_1,   ,        0x1E0000,
fctry,    data, nvs,     ,        0x10000

Firmware size :Project build complete. To flash, run this command:
D:\idf5.1.1\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe ..\..\idf5.1.1\Espressif\frameworks\esp-idf-v5.1.1\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m 0x0 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x19000 build\ota_data_initial.bin 0x20000 build\ESP32_C3_Framework.bin
or run 'idf.py -p (PORT) flash'

mahavirj, 2023-09-14:
@LHYYSY 

- Can you please share your sample application that recreates this problem? Maybe you could modify the IDF OTA example and share the patch on top of it.
- Do you have any changes of your own, e.g. changes in the LD script? If yes, please share them.
- Please update the `Step to reproduce` section from the issue template

LHYYSY, 2023-09-15:
@mahavirj hello，this problem occurs only if the bin file size of my ota is greater than 1.6MB. Can the ota_0 and ota_1 of the partition table be set beyond 0x1C0000?

mahavirj, 2023-09-15:
> this problem occurs only if the bin file size of my ota is greater than 1.6MB. Can the ota_0 and ota_1 of the partition table be set beyond 0x1C0000?

Yes, it should work.

LHYYSY, 2023-09-15:
> > this problem occurs only if the bin file size of my ota is greater than 1.6MB. Can the ota_0 and ota_1 of the partition table be set beyond 0x1C0000?
> 
> Yes, it should work.

It is not that the partition table is set too small, I am not clear, which is the direction leading to this problem

LHYYSY, 2023-09-15:
@mahavirj ,The size of the bin file is 1.68 MB(1768176 bytes). This problem occurs when the device is restarted after the bin file is successfully written.

mahavirj, 2023-09-15:
@LHYYSY Need more information per https://github.com/espressif/esp-idf/issues/12243#issuecomment-1719155466 to comment on this issue

LHYYSY, 2023-09-18:
@mahavirj I checked the firmware and found a failure log in the heade
[image]
r

closed 2023-09-18 by LHYYSY