espressif/esp-idf #12245: ESP32 internal MAC ethernet + IP101GRI conflicts with SDSPI

andrej102, 2023-09-14:
## Environment

Development Kit: custom bord (very similar to ESP32-Ethernet-Kit)
Module or chip used: ESP32-WROOM32D
IDF version : v4.4.4
Build System: CMake
Operating System: Windows
Using an IDE?: Eclipse
Power Supply: USB 5V-1A -> LDO 1A-3.3V

## Problem Description 

SD card connection via SDSPI is used. 

 - SD_DETECT   39
 - CS_SD    	 13
 - MOSI_SD       15
 - MISO_SD        2
 - SCLK_SD        14

When read and write operations are performed periodically (at random times and with random frequency), errors occur on the CMD17, CMD18, CMD24, CMD25 commands (this is what I was able to track, there may be others, but these are the most used when reading and writing large amounts of data), as a rule, these are errors 0x107, 0x108, 0x109.  

This behavior is observed only after starting ethernet, regardless of whether we transmit data over ethernet or not. Ethernet is organized in the same way as on the ESP32-Ethernet-Kit board.

It seems that there are problems with hardware, with the implementation of the board. But if do not run ethernet, or use Wi-Fi instead of ethernet, then there are no problems with the SD card, everything works without reading or writing errors.

## Experiments and results

There is no dependence on the type and volume of the SD card used

I tried changing the clock frequency of the card from 20 to 10 Mhz. Errors appear much less frequently, but they are there.

I also did some experiments to find out the influence of the components of the board, including the 50 Mhz Ethernet clocking signal. But did not reveal any dependencies.

And most importantly, if I switch from SPI to SDIO (1 bit mode, default 20 Mhz) interface to work with the card, errors disappear. I.e. when using the SDIO interface (1 bit mode) on the same ESP 32 pins, there are no errors when working with the card.

 - SD_CMD			        15
 - define SD_CLK			14
 - define SD_D0			2

## Question:

 Are there any hidden problems in ESP32 with simultaneous use of SDSPI and internal MAC ethernet?
How to properly connect and work with an SD card  in ESP32 if an internal MAC eternet is used?

kostaond, 2023-09-15:
Could you please try to set EMAC DMA burst length to e.g. `EMAC_LL_DMA_BURST_LENGTH_4BEAT` at https://github.com/espressif/esp-idf/blob/v4.4.4/components/hal/emac_hal.c#L301? 

Note that there is user configuration option for this in higher IDF versions.

FixJA, 2023-09-20:
Modifying it according to your configuration solved my problem, but I would like to ask why changing this option works fine

andrej102, 2023-09-22:
Thank you very much.

I have tested and can confirm that the proposed changes solve the described problem of simultaneous use of SDSPI and Ethernet.

In lines 300 and 301 (it is necessary to change it in both lines) of file /components/hal/emac_hal.c, I changed the value of EMAC_LL_DMA_BURST_LENGTH_32BETA to EMAC_LL_DMA_BURST_LENGTH_4BEAT

After the changes, there are no errors when reading or writing the SD card via the SD SPI interface (20 MHz frequency) with simultaneous use internal MAC ethernet. Everything works stably.

I like to understand all the details of the work of ESP32, but I don't understand the meaning of the changes made. Could you clarify a little the reason for the errors that appeared without these changes, and what exactly these changes do.

kostaond, 2023-09-22:
Great to hear it helped! The DMA is shared resource between EMAC and the SPI so its utilization needs to be balanced.

closed 2023-09-25 by andrej102