espressif/esp-idf #12252: ESP32-C6 lp core components not being included

dryet, 2023-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2756-g2bc1f2f574

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect the code to build withou issues. There should be a reference to the func fuction.

### What is the actual behavior?

Instead it builds with error conserning the reference to the func function. 

[image]

### Steps to reproduce.

1. Create new project using the lp_i2c example
2. Add new component to the project with any name
3. Include the header file of the component in the main.c file
4. Call func function in the mainfuction of lp processor
5. Build the project

### Build or installation Logs.

```plain
[2/5] Linking C executable lp_core_main.elf
FAILED: lp_core_main.elf 
cmd.exe /C "cd . && C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe -Os -march=rv32imac_zicsr_zifencei -mdiv -fdata-sections -ffunction-sections -march=rv32imac_zicsr_zifencei --specs=nano.specs --specs=nosys.specs   -T C:/Users/Petr/Espressif_workspaces/Master_workspace/lp_i2c_component_test/build/esp-idf/main/lp_core_main/lp_core_riscv.ld -nostartfiles -Wl,--no-warn-rwx-segments -Wl,--gc-sections -Wl,-Map=C:/Users/Petr/Espressif_workspaces/Master_workspace/lp_i2c_component_test/build/esp-idf/main/lp_core_main/lp_core_main.map C:/Espressif/frameworks/esp-idf/components/soc/esp32c6/ld/esp32c6.peripherals.ld CMakeFiles/lp_core_main.dir/C_/Users/Petr/Espressif_workspaces/Master_workspace/lp_i2c_component_test/main/lp_core/main.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/start.S.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/vector.S.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/shared/ulp_lp_core_memory_shared.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/shared/ulp_lp_core_lp_timer_shared.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/lp_core_startup.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/lp_core_utils.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/lp_core_i2c.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/hal/uart_hal_iram.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/hal/uart_hal.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/lp_core_uart.c.obj CMakeFiles/lp_core_main.dir/C_/Espressif/frameworks/esp-idf/components/ulp/lp_core/lp_core/lp_core_print.c.obj -o lp_core_main.elf   && cd ."
c:/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld.exe: CMakeFiles/lp_core_main.dir/C_/Users/Petr/Espressif_workspaces/Master_workspace/lp_i2c_component_test/main/lp_core/main.c.obj: in function `main':
main.c:(.text.startup.main+0x1a): undefined reference to `func'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
```

### More Information.

Main.c file:

/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdio.h>
#include <stdint.h>
#include <stdbool.h>
#include "ulp_lp_core_i2c.h"
#include "ulp_lp_core_utils.h"
#include "../bh1750_defs.h"

#include "../../components/sensor/include/sensor.h"

#define LP_I2C_TRANS_TIMEOUT_CYCLES 5000
#define LP_I2C_TRANS_WAIT_FOREVER   -1

#define LUX_THRESHOLD_LOW           5
#define LUX_THRESHOLD_HIGH          1000

static uint32_t sensor_on = 0;
static uint32_t res_update_done = 0;
volatile uint32_t lux = 0;

static void bh1750_read()
{
    uint8_t data_rd[2];
    esp_err_t ret = lp_core_i2c_master_read_from_device(LP_I2C_NUM_0, BH1750_I2C_ADDR, data_rd, sizeof(data_rd), LP_I2C_TRANS_TIMEOUT_CYCLES);
    if (ret != ESP_OK) {
        // Skip this round of calculation and return
        return;
    }

    /* Calculate light intensity value */
    uint16_t level = ((data_rd[0] << 8) | data_rd[1]);
    lux = (level * 10) / 12;

    /* Wakeup main CPU if the Lux breaches the thresholds */
    if (lux < LUX_THRESHOLD_LOW || lux > LUX_THRESHOLD_HIGH) {
        ulp_lp_core_wakeup_main_processor();
    }
}

int main (void)
{
    uint8_t data_wr = 0;
    esp_err_t ret = ESP_OK;

    func();

    while (1) {
        if (!sensor_on) {
            /* Power ON the sensor */
            data_wr = BH1750_POWER_ON;
            ret = lp_core_i2c_master_write_to_device(LP_I2C_NUM_0, BH1750_I2C_ADDR, &data_wr, sizeof(data_wr), LP_I2C_TRANS_WAIT_FOREVER);
            if (ret != ESP_OK) {
                // Bail and try again
                continue;
            }
            sensor_on = 1;
        }

        if (!res_update_done) {
            data_wr = EXAMPLE_RES_MODE;
            ret = lp_core_i2c_master_write_to_device(LP_I2C_NUM_0, BH1750_I2C_ADDR, &data_wr, sizeof(data_wr), LP_I2C_TRANS_WAIT_FOREVER);
            if (ret != ESP_OK) {
                // Bail and try again
                continue;
            }
            res_update_done = 1;
        }

        /* Read BH1750 sensor data */
        bh1750_read();
    }

    return 0;
}

sudeep-mohanty, 2023-09-15:
Hi @dryet,
What kind of a function is `func()`? Is it being built for the LP core? To me it seems like the `func()` function is part of a component which isn't being built for the LP core but rather for the HP core (main core). Since the binary that is loaded on to the LP core is different, you may have to compile sources/components for the LP core specifically. You may have to update the [components/ulp/lp_core/CMakeLists.txt](https://github.com/espressif/esp-idf/blob/master/components/ulp/cmake/CMakeLists.txt#L81) to achieve this.

sudeep-mohanty, 2023-09-15:
No problem! My previous suggestion was under the assumption that the component is common to the HP core and LP core and hence the CMakeLists.txt would have been the place to specifically add source files. But I stand corrected and as you pointed out, it would be much easier to have your new component as part of your application to manage it. `main/lp_core` looks good to me.

dryet, 2023-09-15:
Yes it was common to both but I moved it to the lp_core folder. 

So I did the following: 
Created new component using the IDE.
Moved it to the lp_core folder.
Included the header file if main.c file.
But i still get the reference error. Is it because I have not edited the ulp part of the cmake list? 

#
# ULP support additions to component CMakeLists.txt.
#
# 1. The LP Core app name must be unique (if multiple components use LP Core).
set(ulp_app_name lp_core_${COMPONENT_NAME})
#
# 2. Specify all C files.
#    Files should be placed into a separate directory (in this case, lp_core/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_lp_core_sources "lp_core/main.c")

#
# 3. List all the component source files which include automatically
#    generated LP Core export file, ${ulp_app_name}.h:
set(ulp_exp_dep_srcs "lp_i2c_main.c")

#
# 4. Call function to build ULP binary and embed in project using the argument
#    values above.
ulp_embed_binary(${ulp_app_name} "${ulp_lp_core_sources}" "${ulp_exp_dep_srcs}")

Do I need to make any changes to this to include my component for the LP core binary? For testing purposes lets say my component name is "test" and I have a test.c file in main/lp_core/test/test.c path.

sudeep-mohanty, 2023-09-15:
@dryet If your application folder looks like this (notice `my_component`) -

```
├── CMakeLists.txt
├── README.md
├── main
│   ├── CMakeLists.txt
│   ├── bh1750_defs.h
│   ├── lp_core
│   │   ├── main.c
│   │   └── my_component
│   │       ├── include
│   │       │   └── test.h
│   │       └── test.c
│   └── lp_i2c_main.c
```

Then your `main/CMakeLists.txt` would look like this (notice changes to `INCLUDE_DIRS` and  `ulp_lp_core_sources`) - 

```
# Register the component
idf_component_register(SRCS "lp_i2c_main.c"
                       INCLUDE_DIRS "" "lp_core/my_component/include"
                       REQUIRES ulp)

#
# ULP support additions to component CMakeLists.txt.
#
# 1. The LP Core app name must be unique (if multiple components use LP Core).
set(ulp_app_name lp_core_${COMPONENT_NAME})
#
# 2. Specify all C files.
#    Files should be placed into a separate directory (in this case, lp_core/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_lp_core_sources "lp_core/main.c" "lp_core/my_component/test.c")

#
# 3. List all the component source files which include automatically
#    generated LP Core export file, ${ulp_app_name}.h:
set(ulp_exp_dep_srcs "lp_i2c_main.c")

#
# 4. Call function to build ULP binary and embed in project using the argument
#    values above.
ulp_embed_binary(${ulp_app_name} "${ulp_lp_core_sources}" "${ulp_exp_dep_srcs}")
```

And when you need to use this component, your `lp_core/main.c` would look like -

```
#include "test.h"

void main()
{
    func();
    ...
    ...
}

dryet, 2023-09-15:
@sudeep-mohanty Thank you so much this is exactly what I needed any my code works now. Your answer is very clear.

sudeep-mohanty, 2023-09-15:
Glad that it works for you now! Closing this issue now. Please feel free to reach out to us if you any issues. Thanks!

closed 2023-09-15 by sudeep-mohanty