espressif/esp-idf #12253: When the chip has been connected to a considerable number of Bluetooth devices, the chip restarts repeatedly

fzboffice, 2023-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-400-gc57d30e1df

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

Just nromal.

### What is the actual behavior?

The chip restarts repeatedly.

### Steps to reproduce.

Esp32 scan and connet ble device,and disconnect,and scan and connect another ble device...many times

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4011b0b8  PS      : 0x00060630  A0      : 0x8011b0e0  A1      : 0x3ffdbf60  
A2      : 0x00646973  A3      : 0x3ffdd400  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00060620  A7      : 0x00000001  A8      : 0x80153e16  A9      : 0x3ffdbf20  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffb7fa0  A15     : 0x3ffb7f4c  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00646973  LBEG    : 0x400917da  LEND    : 0x400917e5  LCOUNT  : 0x00000000  

Backtrace: 0x4011b0b5:0x3ffdbf60 0x4011b0dd:0x3ffdbf80 0x4011ef82:0x3ffdbfa0 0x4012d0f5:0x3ffdbfc0 0x401211bf:0x3ffdc030 0x40121171:0x3ffdc050 0x401212fd:0x3ffdc070 0x4011b203:0x3ffdc090
```

fzboffice, 2023-09-15:
If I change "CONFIG_GATTC_CACHE_NVS_FLASH=y" to "CONFIG_BT_GATTC_CACHE_NVS_FLASH is not set"in sdkconfig,
it is normal.

zhp0406, 2023-09-18:
@fzboffice 
Hi,thanks for report the issue,
Can you provide a complete log to help me analyze the issue?

fzboffice, 2023-09-18:
@zhp0406 Hi,here is the complete log.

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13752
load:0x40080400,len:3608
entry 0x400805f0
I (551) cpu_start: Multicore app
I (551) psram: This chip is ESP32-D0WD
I (552) spiram: Found 64MBit SPI RAM device
I (553) spiram: SPI RAM mode: flash 80m sram 80m
I (559) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (566) cpu_start: Pro cpu up.
I (570) cpu_start: Starting app cpu, entry point is 0x4008142c
I (563) cpu_start: App cpu up.
I (1076) spiram: SPI SRAM memory test OK
I (1088) cpu_start: Pro cpu start user code
I (1088) cpu_start: cpu freq: 240000000
I (1088) cpu_start: Application information:
I (1091) cpu_start: Project name:     witty_gateway
I (1096) cpu_start: App version:      release/2023021701-353-gd136297
I (1103) cpu_start: Compile time:     Sep 18 2023 14:17:25
I (1110) cpu_start: ELF file SHA256:  323c4e29ef3a7971...
I (1116) cpu_start: ESP-IDF:          v4.4.5-402-ge9d442d2b7
I (1122) cpu_start: Min chip rev:     v0.0
I (1127) cpu_start: Max chip rev:     v3.99 
I (1132) cpu_start: Chip rev:         v1.0
I (1137) heap_init: Initializing. RAM available for dynamic allocation:
I (1144) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1150) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1156) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1162) heap_init: At 3FFC80C8 len 00017F38 (95 KiB): DRAM
I (1169) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1175) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1181) heap_init: At 4009FC78 len 00000388 (0 KiB): IRAM
I (1188) spiram: Adding pool of 4034K of external SPI memory to heap allocator
I (1196) spi_flash: detected chip: gd
I (1200) spi_flash: flash io: qio
I (1204) esp_core_dump_flash: Init core dump to flash
I (1215) esp_core_dump_flash: Found partition 'coredump' @ 690000 65536 bytes
I (1240) esp_core_dump_flash: Core dump data checksum is correct
I (1240) esp_core_dump_flash: Found core dump 28452 bytes in flash @ 0x690000
I (1244) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1259) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations

I (2025) BTDM_INIT: BT controller compile version [7b24543]
I (2026) system_api: Base MAC address is not set
I (2026) system_api: read default base MAC address from EFUSE
I (2032) BTDM_INIT: Bluetooth MAC: a8:03:2a:d4:5a:3e
I (2041) phy_init: phy_version 4771,450c73b,Aug 16 2023,11:03:10
D (2258) BT_BTC: btc_transfer_context msg 0 0 0 0x0

D (2259) BT_BTC: btc_thread_handler msg 0 0 0 0x3f820988

D (2259) BT_BTC: btc_main_call_handler act 0

D (2262) BT_APPL: Enable HCI

D (2265) BT_OSI: alarm_cbs_lookfor_available 0 0x3f82182c

D (2271) BT_OSI: alarm_cbs_lookfor_available 1 0x3f821844

D (2277) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (2283) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (2289) BT_BTM: btm_acl_init

[... 78 lines cut ...]

D (2540) BT_GATT: =====> handle = [0x0009] uuid16 = [0x2803] perm=0x01

D (2546) BT_GATT: allocate attr 20 bytes

D (2550) BT_GATT: handle space full. handle_max = 10 next_handle = 10

I (2557) BT_GATT: GATTS_StartService
D (2561) BT_GATT: total buffer in db [2]
D (2565) BT_GATT: allocated i_sreg=0

D (2568) BT_GATT: s_hdl=1 e_hdl=9 type=0x2800 svc_inst=0 sdp_hdl=0x0

D (2575) BT_GATT: gatt_sr_is_new_srv_chg: Yes
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401245c1  PS      : 0x00060530  A0      : 0x8011a689  A1      : 0x3ffdef40  
A2      : 0x00000025  A3      : 0x3f823cf8  A4      : 0x00000003  A5      : 0x00000010  
A6      : 0x00000000  A7      : 0x3ffc7574  A8      : 0x0000002b  A9      : 0x00000001  
A10     : 0x3f823cf8  A11     : 0x00001400  A12     : 0x3ffb6608  A13     : 0x00001400  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000024  LBEG    : 0x40091fa6  LEND    : 0x40091fb1  LCOUNT  : 0x00000000  

Backtrace: 0x401245be:0x3ffdef40 0x4011a686:0x3ffdefd0 0x4010f103:0x3ffdf020 0x40111d46:0x3ffdf040 0x40113ea3:0x3ffdf080 0x40117729:0x3ffdf100 0x4010debd:0x3ffdf150 0x4010e1fb:0x3ffdf170 0x4012ab73:0x3ffdf190

fzboffice, 2023-09-18:
@zhp0406 Here is the backtrace

0x401245be: aes_cipher_msg_auth_code at esp-idf/components/bt/host/bluedroid/stack/smp/smp_cmac.c:284
0x4011a686: gatts_calculate_datebase_hash at esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_sr_hash.c:191
0x4010f103: gatt_update_for_database_change at esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_api.c:129
0x40111d46: GATTS_StartService at esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_api.c:530
0x40113ea3: gatt_profile_db_init at esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_attr.c:491 (discriminator 3)
0x40117729: gatt_init at esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_main.c:148
0x4010debd: btu_init_core at esp-idf/components/bt/host/bluedroid/stack/btu/btu_init.c:102
0x4010e1fb: btu_task_start_up at esp-idf/components/bt/host/bluedroid/stack/btu/btu_task.c:270
0x4012ab73: osi_thread_run at esp-idf/components/bt/common/osi/thread.c:165

zhp0406, 2023-09-18:
@fzboffice 
The log you provided is very helpful, and we will analyze the reasons as soon as possible. If you have any new findings during this time, please feel free to report them to me.thanks.

zhp0406, 2023-09-18:
@fzboffice 
How many chips connected  when restarts ? and could you provide the code to help me reproduce the issue?

zhp0406, 2023-09-18:
@fzboffice 
To monitor the heap usage, you can use **uint32_t esp_get_free_heap_size(void);** to print the remaining heap.

fzboffice, 2023-09-18:
@zhp0406 The chip which I use has 8M psram,so heap is enough.The chip is not connected to many devices at the same time, and it has been connected to at least 20 devices before.

zhp0406, 2023-09-18:
@fzboffice 
how can I reproduce this issue,
could you provide the code to help me reproduce the issue

zhp0406, 2023-09-18:
@fzboffice 
Thank you very much. I'll be waiting for your code . 
Additionally, I will also try to reproduce the issue through other means. 
Please let me know if you discover anything new.

zhp0406, 2023-10-11:
@fzboffice 
Hi,Is there any recent progress about reproducing the issue?

fzboffice, 2023-10-11:
@zhp0406 
Hi,I did two demos last week,but I can't reproduce any more.

zhp0406, 2023-10-11:
@fzboffice 
If you can reproduce the issue, it would be very helpful for us. I will continue to monitor the issue. If you have any new findings, please let me know. Thank you.

weiyuhannnn, 2024-07-01:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates

closed 2024-07-01 by weiyuhannnn