espressif/esp-idf #12254: Mesh v1.1 examples not working

ruslanslobozhenuk, 2023-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 I compiled the rpr_client example from the master branch and received an error message:

I (698) BLE_INIT: BT controller compile version [3a1003f]
I (698) BLE_INIT: Bluetooth MAC: 34:b4:72:7f:f1:8e
I (698) phy_init: phy_version 1110,9c20f0a,Jul 27 2023,10:42:54
Bluetooth Mesh v1.1 commit: [ccd76b06e2]
E (758) BLE_MESH: Bluetooth Mesh v1.1 init failed
I (758) EXAMPLE: ESP_BLE_MESH_PROVISIONER_SET_DEV_UUID_MATCH_COMP_EVT, err_code 0
E (758) BLE_MESH: No provisioning context provided
E (768) BLE_MESH: Failed to create network
I (768) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_ENABLE_COMP_EVT, err_code -22
E (778) BLE_MESH: Invalid NetKeyIndex 0x0000
I (788) EXAMPLE: ESP_BLE_MESH_PROVISIONER_ADD_LOCAL_APP_KEY_COMP_EVT, err_code -19
I (798) EXAMPLE: BLE Mesh Provisioner initialized
I (798) main_task: Returned from app_main()

Since the mesh library is supplied without source code, it is unclear why it does not work.

forx157, 2023-09-18:
@ruslanslobozhenuk 
Thank you for your use.
The issue has now been fixed, and we're trying to merge it into the Github branch as soon as possible.
Please wait for the next update.

forx157, 2023-09-19:
@ruslanslobozhenuk 
hello, now the problem is solved, please use the latest version of the master branch.

ruslanslobozhenuk, 2023-09-21:
After the last commit, the mesh initialization is successful. When trying to make provision using NRF Mesh, the following happens (example wifi_coexist):

I (1387) main_task: Returned from app_main()
esp> I (7847) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer: PB-GATT
E (10098) BLE_MESH: Not connected
I (10111) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (10113) EXAMPLE: net_idx: 0x0000, unicast_addr: 0x0122
I (10113) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (11165) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13
W (11168) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13

At the same time, I can add an App key and I see from the logs that the board receives it, but there is no ACK for confirmation. It's also not possible to get Device composition data (((

ruslanslobozhenuk, 2023-09-21:
> @ruslanslobozhenuk Thank you for your use. The issue has now been fixed, and we're trying to merge it into the Github branch as soon as possible. Please wait for the next update.

After the last commit, the mesh initialization is successful. When trying to make provision using NRF Mesh, the following happens (example wifi_coexist):

I (1387) main_task: Returned from app_main()
esp> I (7847) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer: PB-GATT
E (10098) BLE_MESH: Not connected
I (10111) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (10113) EXAMPLE: net_idx: 0x0000, unicast_addr: 0x0122
I (10113) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (11165) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13
W (11168) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13

At the same time, I can add an App key and I see from the logs that the board receives it, but there is no ACK for confirmation. It's also not possible to get Device composition data (((

ruslanslobozhenuk, 2023-09-29:
After commit 'revert/ble_mesh_example_changes' provision new ble_mesh node not working! (((

forx157, 2023-11-13:
@ruslanslobozhenuk 
I apologize for the delayed response. I am currently using the master branch, iOS NRF Mesh v4.0.1, and have not encountered this issue. Could you please provide information about your testing environment?

ruslanslobozhenuk, 2023-11-13:
I am using Android NRF Mesh 3.3.2 from Google Play Market

chegewara, 2024-02-23:
Hi,
I am having the same issue with master branch (v5.3.x) and I am not sure it is problem with nrf mesh app.
There is another app in the store called `nrf mesh sniffer`. This app can catch all adv beacons and with provided keys to decode them.
With old esp ble mesh v1.0 app mesh sniffer can see secure network beacons as authenticated, which means its decoded with provided key. 
With esp ble mesh v1.1 app its not working, because no packets captured.

What i found is the difference in advertisement `flags & services` tab on nrf connect app.
Mesh v1.0 is advertising somehow 2 sets of flags, and v1.1 only 1 set according to this app.

Maybe that can helps to narrow and solve the problem.

Thanks

chegewara, 2024-02-23:
This is the difference i found between v1.0 and v1.1

mesh v1.0 device
[image]

mesh v1.1 device
[image]