espressif/esp-idf #12255: Download 2 binaries files with OTA on Nina W106

zapsalis, 2023-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone,

I am currently looking to update my Nina via Wifi using OTA.

However, the problem that confronts me is that when I build my project "idf.py build", 2 binaries are generated: The first "web-form.bin" including all the "C" code, and the second "storage .bin", including all html/css/js.

I am currently able to download a "web-form.bin" binary, and it downloads well in my card, and my product resets well on it, the update is therefore done correctly. However, how can I also download the “storage.bin” binary? Because if I modify my web page, I want to be able to update it via wifi too.

If you have ideas for how to update my product when I have 2 binaries, I'm interested.

Thank you so much
Sincerely,
Zapsalis

hmalpani, 2023-09-20:
Hello @zapsalis 
You can refer to [spiffs example](https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs) in ESP-IDF for serving your purpose. I recommend you download the storage.bin file using esp_http_client and simultaneously write it into the spiffs partition.

zapsalis, 2023-09-22:
Hi @hmalpani,

Thanks for your response, I tried this problem.

But finally, with a python script, I convert all my html css and js files to variables to only have one binary to update.

THANKS,
Zapsalis

closed 2023-09-22 by zapsalis