espressif/esp-idf #12257: Passing custom sdkconfig file path to compiler

mbastida123, 2023-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,  I'm going to explain our setup to better understand why I'm asking this question.

We have kind of a complex project that relies heavily on sdkconfig.defaults file to compile binaries suited for different functionalities/hardware of our products.
The sdkconfig is included in the .gitignore so that there are no git conflicts.
This setup works great when developing on each engineer's computer.

Now I would like to deploy a CI/CD pipeline to automatically generate new releases. 
But for time reasons I want to compile those binaries in parallel.
I already know that I can set a custom `build` folder on the compiler. This way if there are multiple compilations running parallely there won't be any conflict.
However, since the sdkconfig file is not placed in the `build` folder by default I need a way to change that.  Because if I leave it as is there could be conflicts when multiple threads are compiling code based on different `sdkconfig.defaults` files.

So the question is: Is there any way to change the sdkconfig file path?

igrr, 2023-09-15:
Hi @mbastida123,

There are two options:

1. Specify the sdkconfig file path when invoking idf.py:
   ```bash
   idf.py -B build_1 -DSDKCONFIG=build_1/sdkconfig -DSDKCONFIG_DEFAULTS=sdkconfig.defaults.1
   idf.py -B build_2 -DSDKCONFIG=build_2/sdkconfig -DSDKCONFIG_DEFAULTS=sdkconfig.defaults.2
   ```
2. Set it in project CMakeLists.txt file once, to always place sdkconfig into the build directory:
   https://github.com/espressif/esp-idf/blob/2bc1f2f574d791d6bffc1688d115a0f549e6364c/examples/build_system/cmake/multi_config/CMakeLists.txt#L6

(the rest of [multi_config example readme](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config) might also be helpful to explore for your use case.)

If you have time, I would also recommend taking a look at [idf-build-apps](https://docs.espressif.com/projects/idf-build-apps/en/latest/) — it's a tool which can simplify building multiple configurations of multiple apps.

mbastida123, 2023-09-18:
Thanks @igrr !

PS: I always forget that there are thinks that are not documented but exist in the code examples :)

mbastida123, 2024-09-13:
@igrr ?

closed 2024-09-13 by mbastida123