espressif/esp-idf #12260: SPI Master spi_device_queue_trans call in interrupt issue

pnt325, 2023-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

I'm try to read SPI data by use DMA, and it's trigger inside of interrupt callback from GPIO by call the function `spi_device_queue_trans` it's make stuck because inside this function call to [BaseType_t r = xQueueSend(handle->trans_queue, (void *)&trans_buf, ticks_to_wait);](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/gpspi/spi_master.c?plain=1#L911) that not allow to  call inside hardware interrupt.

Not sure that is the issue, or I'm try the wrong way?

Thanks
Phat

suda-morris, 2023-09-18:
I think that function is not allowed to run in an ISR context. If you want to use a GPIO to trigger an SPI transaction, you may:

1. Create a  task (task A)
2. in the GPIO interrupt callback function, send a semaphore (sem B)
3. in task A, wait for sem B, after getting the semaphore, then start the SPI transaction

pnt325, 2023-09-19:
Hi @suda-morris 

Thank for your response and reference solution.

But it's possible some function or other ways to use direct SPI API instead of notify to handle by task. 

Currently I have a task to handle data after DMA completed. If now I create other task to handle the SPI DMA read that take time and also need more RAM for more task?

Thanks,

wanckl, 2023-09-28:
@pnt325  Hi     

If you are not request which API to use, I remember polling API for SPI master is able to used in ISR context, e.g. `spi_device_polling_transmit` and `polling_start/end`, these API transmit data directly without queue.

pnt325, 2023-09-28:
Thank @wanckl 

But I think that take time tom stay on ISR will affected to performance of other task. I have as least 10 task and 2 of them is continuous data read and calculate in short period under 50ms. It's 3 phase AC energy monitor device.

closed 2023-09-28 by pnt325