espressif/esp-idf #12261: esp_netif_list_lock/esp_netif_list_unlock freeing mutex creates problems

arnoutdekimo, 2023-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4 / latest master

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-wrover-kit

### Power Supply used.

USB

### What is the expected behavior?

No race conditions leading to crashes

### What is the actual behavior?

Race conditions leading to crashes:

The issue is that the mutex protecting the esp_netif_list isn't thread-safe.

Two reasons:
1) In theory two threads could try and create an interface at the same time, each reaching esp_netif_list_lock at the same time. The first thread could context-switch just after having detected that the current mutex is null, and planning to allocate a new one. The other could then allocate one as well, and lock it, before context switching back. When the first one is brought back up, it will overwrite the first one, causing corruption. 
The chances of this happening are small though, but it is still bad design.
The correct way of doing this is initing the relevant mutexes once in some init function.

2) The fact that the mutex is again optionally destroyed on esp_netif_list_unlock is really bad. (See [here  ](https://github.com/espressif/esp-idf/blob/2bc1f2f574d791d6bffc1688d115a0f549e6364c/components/esp_netif/esp_netif_objects.c#L52C15-L52C15))
For instance, if we call esp_netif_get_handle_from_ifkey (e.g. by mdns_init) at the same time and/or just before an interface is created, it can lead to the mutex being allocated, freed, but also simultaenously used.

This has lead to constant crashes of spinlocks asserts that can be found in the debug logs.

Note that this guy https://www.esp32.com/viewtopic.php?t=20233 has clearly seen this issue before in 2021 already.

A correct design would have an init function, deinit function, and not alloc/dealloc mutexes at runtime..

### Steps to reproduce.

Problem 1) Create two threads, that simultaneously try to create interfaces. (Very hard to reproduce though)

Problem 2) Also a tricky race condition, but was able to reproduce this in a constant crashloop: 
 - Call mdns_init in one thread
 - In other thread, simultaneously create the **FIRST** interface

By slightly varying the delays between the thread execution sequences, the spinlock crash can be triggered. (But as this is a race condition, this is tricky. Anyway, just from reading the code the issue can be clearly spotted)

### Debug Logs.

```plain
`
assert failed: spinlock_acquire spinlock.h:123 ((result == SPINLOCK_FREE) == (lock->count == 0))

Backtrace: 0x4008231d:0x3ffd73d0 0x40095185:0x3ffd73f0 0x4009c0ed:0x3ffd7410 0x40099441:0x3ffd7530 0x400963f5:0x3ffd7570 0x40142f55:0x3ffd75b0 0x4014310d:0x3ffd75d0 0x40143c61:0x3ffd75f0 0x401460db:0x3ffd7620 0x400f84a2:0x3ffd7650 0x40099211:0x3ffd7820
0x4008231d: panic_abort at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/esp_system/panic.c:408

0x40095185: esp_system_abort at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/esp_system/esp_system.c:137

0x4009c0ed: __assert_func at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/newlib/assert.c:85

0x40099441: spinlock_acquire at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/esp_hw_support/include/soc/spinlock.h:123
 (inlined by) xPortEnterCriticalTimeout at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/freertos/port/xtensa/port.c:301

0x400963f5: vPortEnterCritical at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) xQueueSemaphoreTake at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/freertos/queue.c:1563

0x40142f55: esp_netif_list_lock at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/esp_netif/esp_netif_objects.c:51

0x4014310d: esp_netif_get_handle_from_ifkey at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/esp_netif/esp_netif_objects.c:175

0x40143c61: esp_netif_new at C:/ESP32/esp-idf-eclipse-444/esp-idf/esp-idf-v4.4.4/components/esp_netif/lwip/esp_netif_lwip.c:462
`
```

closed 2023-10-25 by espressif-bot

AxelLin, 2023-10-26:
Hi @david-cermak 

Is it possible to change the esp_netif_next() implementation to use esp_netif_tcpip_exec()
so it won't impact with existing users?
Otherwise you are actually asking exist users to repeat similar implementation in various places.

david-cermak, 2023-10-30:
Hi @AxelLin 

I'm afraid that's not possible. The problem is in the API design itself (similar to the macro `NETIF_FOREACH()` in lwip), which could not assure thread safety between iterations.
That's why I marked the `esp_netif_next()` *deprecated*, so the existing users would get notified and should think about their usecases to do one of the following:

1) just rename `esp_netif_next()` -> `esp_netif_next_unsafe()` if all interfaces are strictly under their control and are created once.
2) rework the iteration, for example using `esp_netif_find_if()` for searching interfaces in a safe way, if the application interfaces could be created/destroyed runtime from multiple tasks

AxelLin, 2023-11-06:
Then I need to wait until the fix is available in stable branches, and update application code accordingly.

david-cermak, 2023-11-30:
Yes, will fix on all release branches (down to `v4.3`)