espressif/esp-idf #12263: USB controlled (non-JTAG) ESP32S3 through idf.py does not start the RISCV ULP

aircable, 2023-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master 5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-PICO-1

### Power Supply used.

USB

### What is the expected behavior?

After ulp_riscv_load_binary(), ulp_set_wakeup_period() and ulp_riscv_config_and_run() I expect the ULP processor to start up.

### What is the actual behavior?

The chip is connected only through USB to the idf.py -p /dev/ttyACM0 monitor. No JTAG used.
I can flash the chip and the ULP processor does not start up.

When connected through ESP_PROG controller to the UART, the same code starts the ULP processor correctly.

### Steps to reproduce.

this piece of startup code does not start the ULP processor
```
    ulp_riscv_cfg_t cfg = {
        .wakeup_source = ULP_RISCV_WAKEUP_SOURCE_TIMER, 
    };

    if( ulp_riscv_load_binary( ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start )) != ESP_OK ){
        ESP_LOGE( TAG, "USP load failed");
    }
    /* The first argument is the period index, which is not used by the ULP-RISC-V timer
     * The second argument is the period in microseconds, which gives a wakeup time period
     */
    ulp_set_wakeup_period( 0, 20000 );

    /* Start the program */
    if(( err = ulp_riscv_config_and_run( &cfg )) != ESP_OK ){
        ESP_LOGE( TAG, "ULP start failed, %s", esp_err_to_name( err ));
    } else {
        ESP_LOGI( TAG, "ULP started");
    }
```

### Debug Logs.

```plain
No log, only preconfigured variables actually have the correct value.
```

### More Information.

Workaround:
I added
```
    ulp_riscv_reset();

```
right after ulp_riscv_config_and_run() and the ULP starts working.

sudeep-mohanty, 2023-09-19:
Hi @aircable, I am able to run the ULP RISC-V normally with the code snippet you shared. How exactly can you tell that the ULP is not running? Would it be possible to share your `sdkconfig`, and some part of your `app_main` and ulp initialization code?

sudeep-mohanty, 2023-10-12:
Issue does not point to a big in ESP-IDF. Closing it. Please open a new ticket in case of other issues.

closed 2023-10-12 by sudeep-mohanty