espressif/esp-idf #12264: Decouple partition CSV location from project directory

brianredbeard, 2023-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Presently the specification of a CSV file for the `partition_table` component gets serialized to an absolute path relative to the derived project location:

https://github.com/espressif/esp-idf/blob/2bc1f2f574d791d6bffc1688d115a0f549e6364c/components/partition_table/project_include.cmake#L14-L15

This results in a problematic composition of the path when there is more than a single level of component composition within an ESP-IDF project.

For example, I cam currently extending the MicroPython project.  Within their build system, it's expected that a user will execute `idf.py build` from the path `ports/esp32` within the project.  This will then set `ports/esp32/partitions.csv` as the required file path.

When consuming MicroPython as a submodule and referencing the CMake scripts from another project while utilizing the same sdkconfig definition (e.g. `ext/micropython/ports/esp32`) due to the change in the location of the `project` call the definition loses portability between projects.

brianredbeard, 2023-09-18:
One resolution would be to allow the user to set a property on the build.  A different idea would be to use the `find_file()` CMake function (https://cmake.org/cmake/help/latest/command/find_file.html) with the `ENV` var sub-option allowing a user to specify an additional search path as a part of the build environment.

brianredbeard, 2023-09-18:
E.g. If I populate my `sdkconfig` with the following:

```
CONFIG_PARTITION_TABLE_FILENAME="boards/sampleboard/sampleboard-partitions.csv"
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="boards/sampleboard/sampleboard-partitions.csv"
CONFIG_MICROPY_FROZEN_MANIFEST="boards/sampleboard/manifest.py"
```

And then attempt to run a build I am met with the following error:

```
Loading defaults file /Users/bharrington/Projects/hardware/badgePython/build/sdkconfig.combined...
CMake Warning at esp-idf/components/partition_table/project_include.cmake:18 (message):
  Partition table CSV file
  /Users/bharrington/Projects/hardware/badgePython/sampleboard-partitions.csv
  not found.  Change custom partition CSV path in menuconfig.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:380 (include)
  esp-idf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  esp-idf/tools/cmake/project.cmake:530 (idf_build_process)
  CMakeLists.txt:88 (project)
```

sudeep-mohanty, 2023-09-19:
@brianredbeard Thanks for reporting the problem to us. Seems like the solution could be to have an absolute path configuration for the `csv` file irrespective of the project path. We shall take a look!

closed 2024-01-30 by espressif-bot