espressif/esp-idf #12265: Segmentation fault while building hello_world example

radian997, 2023-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Project configuration window appears after `idf.py menuconfig` command.

### What is the actual behavior?

Segmentation fault while calling `idf.py menuconfig ` command.

### Steps to reproduce.

cd ~/esp/hello_world
idf.py set-target esp32
idf.py menuconfig

### Build or installation Logs.

```plain
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Python [28469]
Path:                  /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python
Identifier:            com.apple.python3
Version:               3.9.6 (3.9.6)
Build Info:            python3-141000000000000~696
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [28468]
Responsible:           Terminal [24289]
User ID:               501

Date/Time:             2023-09-19 00:00:40.1363 +0200
OS Version:            macOS 13.5.2 (22G91)
Report Version:        12
Anonymous UUID:        B9BA0CEC-B78E-EF7E-0E04-CE914246E2B6

Sleep/Wake UUID:       0449642F-AE18-4D13-AEB8-61ED949B9C4B

Time Awake Since Boot: 19000 seconds
Time Since Wake:       10874 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
[... 95 lines cut ...]
MALLOC guard page                   96K        5 
MALLOC_MEDIUM (reserved)         120.0M        1         reserved VM address space (unallocated)
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                         16K        1 
Stack                             16.0M        1 
VM_ALLOCATE                       7936K       31 
__AUTH                             309K       59 
__AUTH_CONST                      3714K      143 
__DATA                            1934K      143 
__DATA_CONST                      4238K      153 
__DATA_DIRTY                       361K       58 
__LINKEDIT                       802.6M        9 
__OBJC_RO                         66.4M        1 
__OBJC_RW                         2012K        1 
__TEXT                           113.5M      159 
dyld private memory                272K        1 
shared memory                       32K        2 
===========                     =======  ======= 
TOTAL                              1.6G      782 
TOTAL, minus reserved VM space     1.1G      782
```

### More Information.

Tried to build v5.0.4 and v5.1.1 - segfault in both versions

sudeep-mohanty, 2023-09-19:
@radian997 Can you confirm if you have setup your IDF environment as it is mentioned in the [getting started guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html#standard-toolchain-setup-for-linux-and-macos) and have resolved all [Apple M1 issues](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html#apple-m1-users) as per the guide?

radian997, 2023-09-19:
Hi @sudeep-mohanty, I was following installation instructions step by step and it happened.

sudeep-mohanty, 2023-09-19:
@radian997 From the logs you shared, it seems that you are using the system python instance to run the `idf.py` script. I suspect that your environment isn't properly setup. If you've setup your [environment variables](/Users/sudeepmohanty/.espressif/python_env/idf5.2_py3.11_env/bin/python) correctly then your `idf.py` script should execute from the python virtual environment created during the setup process and not use the system python.

An example of the environment - 

```
~/esp/esp-idf  $ which python
/Users/sudeepmohanty/.espressif/python_env/idf5.2_py3.11_env/bin/python

~/esp/esp-idf  $ python --version
Python 3.11.4
```

radian997, 2023-09-19:
I also tried with removing the `#!/usr/bin/env python` line from idf.py, and then:
`/Users/adrian/.espressif/python_env/idf5.0_py3.9_env/bin/python ../../esp-idf/tools/idf.py menuconfig`

but it also results with segfault

igrr, 2023-09-19:
@radian997 Was it working for you before the upgrade to macOS 13.5.2?

I'm on 13.5.1, have tried to reproduce this with OS-provided copy of Python (also 3.9.6) and it seems to be working fine.

I'm also not sure this is an IDF issue and not a Python interpreter issue, since:
```
...
0   ???                           	               0x0 ???
1   Python3                       	       0x1034f91d0 0x1034b0000 + 299472
...
    sp: 0x000000016cf2d120   pc: 0x0000000000000000 cpsr: 0x80001000
   far: 0x0000000000000000  esr: 0x82000006 (Instruction Abort) Translation fault
...
       0x1034b0000 -        0x10370bfff com.apple.python3 (3.9.6) <e104613b-0d1f-36cc-a3a0-46520949a2f4> /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/Python3
...
```
Basically, something in Python3 library is jumping to a function pointer which is NULL. There don't seem to be any 3rd party modules (like `curses`) in the callstack, so seems like it is an interpreter issue?

Aside from the OS version being different, I don't know why it happens on your computer but not mine, though.

radian997, 2023-09-19:
I started digging into the problem and investigated that it happens while `make -j10 menuconfig` is called. Exactly the cmake command from build.make file:
`/opt/homebrew/Cellar/cmake/3.27.5/bin/cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=/Users/adrian/Workspace/esp32/esp32-cpp/projects/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=/Users/adrian/Workspace/esp32/esp32-cpp/projects/hello_world/build/kconfigs_projbuild.in KCONFIG_CONFIG=/Users/adrian/Workspace/esp32/esp32-cpp/projects/hello_world/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= /Users/adrian/.espressif/python_env/idf5.0_py3.9_env/bin/python -m menuconfig /Users/adrian/Workspace/esp32/esp32-cpp/esp-idf/Kconfig`
I tried with cmake 3.23.1 and 3.27.5 but it always ends with segfault.

dobairoland, 2023-09-19:
Please try to install a different Python version (guide is [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html#installing-python-3), using homebrew or macports).

radian997, 2023-09-19:
@dobairoland Thank you, I installed python3 with homebrew and it started working. It's weird that it doesn't work with Python3 provided by command line tools.

closed 2023-09-19 by radian997