espressif/esp-idf #12266: Deselecting VFS in menuconfig results in incorrect console device used

jmattsson, 2023-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Beetle-ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

`Component config -> ESP System Settings -> Channel for console output` is set to `USB Serial / JTAG controller`.
`Component config -> ESP System Settings -> Channel for console secondary output` is set to `No secondary console`.
`Component config -> Virtual File System -> Provide basic I/O functions` is disabled.

With this configuration I would expect `ESP_LOGx()` and `printf()` output to be sent to _only_ the USB console. If VFS is enabled, this is what happens (once in `app_main()` at least).

Observing the UART while flashing & booting the included example, I see:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x5 (DOWNLOAD(USB/UART0/1))
Saved PC:0x40382630
waiting for download
.OHAI...
...........
.....p.....	...........	...........	...........	...........	...........
...........
.. @.......	...........	...........
...........
.....p.....	...........	...........	...........	...........	...........
...........
.. @.......	...........	...........
...............................................................................................
..oP1..............8......n..V..).m.............ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380786
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xfe4
load:0x403cc710,len:0x740
load:0x403ce710,len:0x2c6c
entry 0x403cc710
I (52) cpu_start: Unicore app
I (53) cpu_start: Pro cpu up.
I (61) cpu_start: Pro cpu start user code
I (61) cpu_start: cpu freq: 160000000 Hz
I (61) cpu_start: Application information:
I (61) cpu_start: Project name:     foo
I (65) cpu_start: App version:      3980a26-dirty
I (69) cpu_start: Compile time:      
I (72) cpu_start: ELF file SHA256:  9294228d4118761b...
I (77) cpu_start: ESP-IDF:          v5.1.1
I (81) cpu_start: Min chip rev:     v0.3
I (85) cpu_start: Max chip rev:     v0.99 
I (89) cpu_start: Chip rev:         v0.4
I (92) heap_init: Initializing. RAM available for dynamic allocation:
I (99) heap_init: At 3FC8BEC0 len 00050850 (322 KiB): DRAM
I (104) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (110) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (116) spi_flash: detected chip: generic
I (118) spi_flash: flash io: dio
I (122) sleep: Configure to isolate all GPIO pins in sleep state
I (127) sleep: Enable automatic switching of GPIO sleep configuration
I (134) app_start: Starting scheduler on CPU0
```
Ideally the early log messages should not go to the UART either, but I can live with that (I'm guessing I might be able to stop them with `UART_PRINT_CONTROL` fuses if I really need to).

### What is the actual behavior?

Instead, with VFS disabled the console output appears across *all* of:
  - USB console (good)
  - Main UART output (bad)
  - Main UART input (extra bad)

Observing the UART while flashing & booting the included sample, I see:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x5 (DOWNLOAD(USB/UART0/1))
Saved PC:0x40382630
waiting for download
.OHAI...
...........
.....p.....	...........	...........	...........	...........	...........
...........
.. @.......	...........	...........
...........
.....p.....	...........	...........	...........	...........	...........
...........
.. @.......	...........	...........
...............................................................................................
.et.#.....Py..........ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380786
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xfe4
load:0x403cc710,len:0x740
load:0x403ce710,len:0x2c6c
entry 0x403cc710
I (52) cpu_start: Unicore app
I (52) cpu_start: Pro cpu up.
I (60) cpu_start: Pro cpu start user code
I (60) cpu_start: cpu freq: 160000000 Hz
I (60) cpu_start: Application information:
I (61) cpu_start: Project name:     foo
I (64) cpu_start: App version:      3980a26-dirty
I (68) cpu_start: Compile time:      
I (72) cpu_start: ELF file SHA256:  8565cb4c5c92da17...
I (77) cpu_start: ESP-IDF:          v5.1.1
I (80) cpu_start: Min chip rev:     v0.3
I (84) cpu_start: Max chip rev:     v0.99 
I (88) cpu_start: Chip rev:         v0.4
I (92) heap_init: Initializing. RAM available for dynamic allocation:
I (98) heap_init: At 3FC8BD90 len 00050980 (322 KiB): DRAM
I (103) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (109) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (114) spi_flash: detected chip: generic
I (118) spi_flash: flash io: dio
I (121) sleep: Configure to isolate all GPIO pins in sleep state
I (126) sleep: Enable automatic switching of GPIO sleep configuration
I (133) app_start: Starting scheduler on CPU0
I (137) main_task: Started on CPU0
I (137) main_task: Calling app_main()
Visible on UART (bad)
I (137) main_task: Returned from app_main()
```
Note that despite only the USB serial/jtag console has been enabled in menuconfig, the `Visible on UART (bad)` appears here (in addition to the log message about app_main() exiting). This should not happen.

In our use case, the UART is used to talk to an external device, and having console output injected on the UART interferes with the communication protocol to said external device. As a workaround we have to enable VFS, despite not using the feature.

### Steps to reproduce.

1. Use a ESP32-C3 with direct USB console, e.g. "Beetle" dev board
2. Connect a serial receiver to the default UART pins of the C3
3. Select USB serial/jtag as the only console in menuconfig
4. Disable VFS in menuconfig
5. Build, flash and monitor via USB port (e.g. `/dev/ttyACM0`)
6. Observe that console output goes both to the USB console *and* the UART

A trivial example to reproduce is included below.
Toggle CONFIG_VFS_SUPPORT_IO to switch between the two different behaviours.

sdkconfig.defaults:
```
CONFIG_IDF_TARGET="esp32c3"
CONFIG_APP_REPRODUCIBLE_BUILD=y
CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_LOG_COLORS=n
CONFIG_NEWLIB_STDOUT_LINE_ENDING_LF=y
CONFIG_NEWLIB_STDIN_LINE_ENDING_LF=y
CONFIG_VFS_SUPPORT_IO=n
```

main.cc:
```
#include <cstdio>

extern "C" void app_main()
{
  printf("Visible on UART (bad)\n");
}
```

main/CMakeLists.txt:
```
idf_component_register(
  SRCS
    "main.cc"
)
```

CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(foo)
```

igrr, 2023-09-19:
@jmattsson Thanks for the issue report. I understand the problem, basically we don't really control logging direction if VFS is disabled, since these options (UART, USB_SERIAL_JTAG, USB CDC) are implemented in VFS drivers. When VFS support is disabled, the only method of output is to use some basic I/O functions in the chip ROM. These functions, as you have observed, simply write to all available outputs.

We can do only a cosmetic change by disabling all of the `CONFIG_ESP_CONSOLE_*` options when `CONFIG_VFS_SUPPORT_IO` is disabled, and allow only one (new) option: `CONFIG_ESP_CONSOLE_SIMPLE` (or some other name), explaining the behavior in the help text of this option.

Just one question, is there any specific reason you have to disable CONFIG_VFS_SUPPORT_IO?

jmattsson, 2023-09-19:
Hi @igrr, thanks for the quick response!
Yeah, having the impact reflected in menuconfig would certainly reduce the "WTF" effect. I looked through the docs but could find no mention about the VFS impacting the console selection, so having it right there in menuconfig would be a good start 👍.

The reason in our case is the same as usual - reducing firmware size. We were hoping to be able to run the whole firmware out of RAM, so I was evicting everything we weren't obviously needing. In our case it was still too tight to be viable, but that was how I ended up running into this one.

jmattsson, 2023-09-19:
Actually, a follow-up question if I may, @igrr - if the reason is that the console is reduced a ROM'd output function, does that mean that burning b'11 into the `UART_PRINT_CONTROL` fuses would keep it away from the UART? Or is this a different print function which doesn't pay any heed to that?

ESP-Marius, 2023-09-20:
@jmattsson In this case burning the efuse should also keep it away from UART during secondary bootloader/app yes.

I wonder if we could fix this though, seems like a simple change like this would give a more reasonable behavior that is more in-line with the docs:
```
diff --git a/components/bootloader_support/src/bootloader_console.c b/components/bootloader_support/src/bootloader_console.c
index b7d6d42a6e..50a19173af 100644
--- a/components/bootloader_support/src/bootloader_console.c
+++ b/components/bootloader_support/src/bootloader_console.c
@@ -104,5 +104,6 @@ void bootloader_console_init(void)
 void bootloader_console_init(void)
 {
     esp_rom_uart_switch_buffer(ESP_ROM_USB_SERIAL_DEVICE_NUM);
+    esp_rom_uart_set_as_console(ESP_ROM_USB_SERIAL_DEVICE_NUM);
 }
 #endif //CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
```

With this the rom console will be redirect early in the secondary bootloader resulting in no output on UART after the initial ROM output, if `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`. As a bonus this also works even when VFS is disabled.

I'll take a look and see if there is any reason why we shouldnt do this.

ESP-Marius, 2024-06-12:
Seems to work as expected on master after 9e62e32a5b4cd40bc1 was merged, so marking this as resolved.

closed 2024-06-12 by ESP-Marius