espressif/esp-idf #12267: TWAI driver with support for minimum inter-frame-delay (µs granularity)

mickeyl, 2023-09-19:
Many automotive systems (e.g. ISOTP as per [ISO 15765-2](https://de.wikipedia.org/wiki/ISO_15765-2)) require a minimum inter-frame-delay when transmitting consecutive CAN frames.

Since TWAI is using an outgoing queue, it's not possible to achieve this inter-frame-delay _outside_ the TWAI driver, in particular, when µs-granularity is a requirement.

Since work on a new TWAI driver is underway (see post by @wanckl in https://github.com/espressif/esp-idf/issues/10423#issuecomment-1724779541), let me use this opportunity to file an enhancement request for this.

I'll probably hack away, since I have an immediate need for this feature, but official support in the TWAI driver would be nicer.

mickeyl, 2023-09-21:
There's a bit of discussion about that going on @ https://www.esp32.com/viewtopic.php?f=12&t=35520

I think I know _where_ to approach that (in `twai_handle_tx_buffer_frame`), but I'm a bit out of my depth as to how to create a timer that creates the proper IRQ and event so that I can delay the call to `twai_hal_set_tx_buffer_and_transmit`.

mickeyl, 2023-09-29:
Please also see https://github.com/espressif/esp-idf/issues/12316 which indicates very unstable TWAI timing.

r90548039, 2024-08-06:
Is there any progress on this request? Is it possible to use interrupts for reception?

diplfranzhoepfinger, 2025-04-03:
> Is there any progress on this request? Is it possible to use interrupts for reception?

new TWAI Driver is here for C6, as it supports 2 CAN Interfaces.