espressif/esp-idf #12268: Sysview tracing example crash right after boot

antoDimaggio, 2023-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-1127-gb9e8e0ef4e

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

The terminal output should be as stated in the doc 
````
I (295) example: 0x3ffb6c10: run task
I (297) example: Task[0x3ffb6c10]: received event 1
I (300) example: Task[0x3ffb6c10]: received event 2
I (306) example: Task[0x3ffb6c10]: received event 5
I (311) example: Task[0x3ffb6c10]: received event 8
I (317) example: Task[0x3ffb6c10]: received event 10
````

### What is the actual behavior?

It crash right after booting

### Steps to reproduce.

Just follow the README.md steps without customization. Set your target before building with:
`idf.py set-target esp32s3`

### Debug Logs.

```plain
C:\Users\DH6440\repos\exmples\sysview_tracing>idf.py monitor
Executing action: monitor
Serial port COM7
Connecting....
Detecting chip type... ESP32-S3
Running idf_monitor in directory C:\Users\DH6440\repos\exmples\sysview_tracing
Executing "C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\DH6440\repos\v3\esp-idf\tools/idf_monitor.py -p COM7 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 C:\Users\DH6440\repos\exmples\sysview_tracing\build\sysview_tracing.elf -m 'C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\Users\DH6440\repos\v3\esp-idf\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM7 instead...
--- idf_monitor on \\.\COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400490d5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x163c
load:0x403c9700,len:0xb5c
load:0x403cc700,len:0x3024
entry 0x403c9934
I (24) boot: ESP-IDF v4.4.3-1127-gb9e8e0ef4e 2nd stage bootloader
I (25) boot: compile time 14:07:35
I (25) boot: chip revision: v0.1
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0907ch ( 36988) map
I (100) esp_image: segment 1: paddr=000190a4 vaddr=3fc926b0 size=01fcch (  8140) load
I (104) esp_image: segment 2: paddr=0001b078 vaddr=40374000 size=04fa0h ( 20384) load
I (115) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17f5ch ( 98140) map
I (136) esp_image: segment 4: paddr=00037f84 vaddr=40378fa0 size=09710h ( 38672) load
I (151) boot: Loaded app from partition at offset 0x10000
I (151) boot: Disabling RNG early entropy source...
I (162) cpu_start: Pro cpu up.
I (162) cpu_start: Single core mode
I (172) cpu_start: Pro cpu start user code
I (172) cpu_start: cpu freq: 160000000
I (172) cpu_start: Application information:
I (175) cpu_start: Project name:     sysview_tracing
I (181) cpu_start: App version:      1
I (185) cpu_start: Compile time:     Sep 19 2023 14:07:02
I (191) cpu_start: ELF file SHA256:  3d7be25212c02c9a...
I (197) cpu_start: ESP-IDF:          v4.4.3-1127-gb9e8e0ef4e
I (203) cpu_start: Min chip rev:     v0.0
I (208) cpu_start: Max chip rev:     v0.99
I (213) cpu_start: Chip rev:         v0.1
I (218) heap_init: Initializing. RAM available for dynamic allocation:
I (225) heap_init: At 3FC95730 len 0003A8D0 (234 KiB): D/IRAM
I (231) heap_init: At 3FCD4000 len 0000C000 (48 KiB): D/IRAM
I (238) heap_init: At 3FCE4000 len 00005710 (21 KiB): D/IRAM
I (244) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (251) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (257) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (264) spi_flash: detected chip: generic
I (268) spi_flash: flash io: dio
W (272) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (285) sleep: Configure to isolate all GPIO pins in sleep state
I (292) sleep: Enable automatic switching of GPIO sleep configuration
I (299) esp_apptrace: Initialized TRAX on CPU0
I (304) cpu_start: Starting scheduler on PRO CPU.
I (310) example: 0x3fcea284: run task
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420044d0  PS      : 0x00060a33  A0      : 0x8200461d  A1      : 0x3fce9ff0
0x420044d0: find_desc_for_source at C:/Users/DH6440/repos/v3/esp-idf/components/esp_hw_support/intr_alloc.c:161

A2      : 0x00000035  A3      : 0x00000000  A4      : 0x3fc95b4c  A5      : 0x00000000
A6      : 0x00000031  A7      : 0x3fce9f20  A8      : 0x0004cb70  A9      : 0x00000000
A10     : 0x3fcea490  A11     : 0x3fcea490  A12     : 0x3fce9f44  A13     : 0x3c028220
A14     : 0x0000000a  A15     : 0xff000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0004cb70  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd

Backtrace: 0x420044cd:0x3fce9ff0 0x4200461a:0x3fcea010 0x42004857:0x3fcea050 0x42006bcb:0x3fcea090 0x420054ca:0x3fcea0d0 0x4037d89d:0x3fcea110
0x420044cd: find_desc_for_source at C:/Users/DH6440/repos/v3/esp-idf/components/esp_hw_support/intr_alloc.c:160

0x4200461a: get_available_int at C:/Users/DH6440/repos/v3/esp-idf/components/esp_hw_support/intr_alloc.c:299

0x42004857: esp_intr_alloc_intrstatus at C:/Users/DH6440/repos/v3/esp-idf/components/esp_hw_support/intr_alloc.c:486

0x42006bcb: timer_isr_register at C:/Users/DH6440/repos/v3/esp-idf/components/driver/timer.c:267 (discriminator 2)

0x420054ca: example_task at C:/Users/DH6440/repos/exmples/sysview_tracing/main/sysview_tracing.c:140

0x4037d89d: vPortTaskWrapper at C:/Users/DH6440/repos/v3/esp-idf/components/freertos/port/xtensa/port.c:142

ELF file SHA256: 3d7be25212c02c9a

Rebooting...
```

### More Information.

Also other colleagues from my team tried to run the same example with same results.

luciantodor, 2023-09-19:
I have the same issue on my system, using the esp-idf v4.4. Initially, I thought the issue was related to the memory allocation and tried to enable or disable the external RAM, but this had no impact. We need to explore this example further. Hopefully, someone will have the time to give it a try. Thank you.

erhankur, 2024-12-06:
I am closing this issue as the support period for v4.4 has ended.  @luciantodor Feel free to reopen it if you switch to v5.x and encounter any issues.

closed 2024-12-06 by erhankur