espressif/esp-idf #12269: ESP-IDF 5.1.1 with VSCODE problem show examples

filzek, 2023-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release 5.1.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

ESP-IDF 5.1.1 with VSCODE problem show examples, vscode says: "ESP-IDF: Loading Examples"   forever and does nothing.

### What is the actual behavior?

work and show examples

### Steps to reproduce.

install latest VSCODE and install ESPRESSIF extension. Setup express 5.1.1 release, after finish try to ES-IDF examples show.

Doest nothing.

filzek, 2023-09-19:
Find a solution simple, the link when using the example doesnt work, need to click the directory below, this is a build error on the extension.

see https://www.youtube.com/watch?v=1GXayjjWGoo&ab_channel=EasyLearning 
to solve it.

please solve this error on the EXTENSION.

brianignacio5, 2023-09-20:
Did you configure the extension as described in the documentation ? Can you share the troubleshooting information ?

filzek, 2023-09-20:
yes, it is configured exact like, the problem happen always, but, if in the menu set esp-idf example click and than select the folder it works

brianignacio5, 2023-09-20:
I don't understand the problem here. Can you see the examples ? Can you described the steps to reproduce the error and the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#Troubleshooting) ?