espressif/esp-idf #12271: ESP32S3 SPIFFS flash access during RMT transmission causes panic

higaski, 2023-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-2934-g3b748a6cb7

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

I've an application in which I need to access a SPIFFS partition upon a HTTP request to serve some data while RMT transmissions are running on a high priority task. For some reason I can't get that combination to work properly. Every time I access the SPIFFS partition while the RMT transmission is ongoing I'm getting core0 panics

```sh
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4037a8f0  PS      : 0x00060034  A0      : 0x4037ba9c  A1      : 0x3fca04e0  
0x4037a8f0: shared_intr_isr at /home/vinci/esp/esp-idf/components/esp_hw_support/intr_alloc.c:450

0x4037ba9c: _xt_lowint1 at /home/vinci/esp/esp-idf/components/xtensa/xtensa_vectors.S:1236

A2      : 0x3c134c60  A3      : 0xb33fffff  A4      : 0x600c4000  A5      : 0x403860e6  
0x403860e6: _frxt_int_enter at /home/vinci/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:139

A6      : 0x00000000  A7      : 0xb33fffff  A8      : 0x00000bad  A9      : 0x3fca04b0  
A10     : 0x00000001  A11     : 0x00060021  A12     : 0x00000000  A13     : 0x00060023  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
0x400570e8: memset in ROM

0x400570f3: memset in ROM
```

I can't tell why though. I've enabled the CONFIG_RMT_ISR_IRAM_SAFE option and attached the IRAM attribute to my RMT callback. Thankfully I can reproduce the issue in about 200 lines of code. I'll update this post in a sec.

### What is the actual behavior?

ESP32 crashes with a core0 cache panic

### Steps to reproduce.

1. Take the example, set WiFi SSID and password with idf.py menuconfig
2. Flash the example on an ESP32S3 (e.g. ESP32-S3-DevKitC-1)
3. Wait for WiFi to connect
4. Get the IP address from LOG and put it in a browser
5. Voila, your ESP32 will panic and reboot

suda-morris, 2023-09-20:
@higaski Could it be that the `std::array<char, 4uz> const chunk` was placed in the flash. And when the RMT copy encoder doing the job, it needs to read that raw data, however, the cache is disabled at that time

higaski, 2023-09-20:
That would only be possible if the data would be static.

igrr, 2023-09-20:
I haven't tried this with std::array, but simple C `const` arrays above certain size also get placed into rodata segment quite often. Could you please check the placement in the .map file, or attach the .map file here?

suda-morris, 2023-09-21:
Hi @higaski 

The issue is discovered by this Kconfig: `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0`. The interrupt allocator internally allocated some buffer from the PSRAM, so causing the crash.

```diff
diff --git a/components/esp_hw_support/intr_alloc.c b/components/esp_hw_support/intr_alloc.c
index 02d72ccd5d1..99906ad9a1f 100644
--- a/components/esp_hw_support/intr_alloc.c
+++ b/components/esp_hw_support/intr_alloc.c
@@ -560,7 +560,7 @@ esp_err_t esp_intr_alloc_intrstatus(int source, int flags, uint32_t intrstatusre
     //Allocate that int!
     if (flags & ESP_INTR_FLAG_SHARED) {
         //Populate vector entry and add to linked list.
-        shared_vector_desc_t *sh_vec=malloc(sizeof(shared_vector_desc_t));
+        shared_vector_desc_t *sh_vec=heap_caps_malloc(sizeof(shared_vector_desc_t), MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
         if (sh_vec == NULL) {
             portEXIT_CRITICAL(&spinlock);
             free(ret);
@@ -583,7 +583,7 @@ esp_err_t esp_intr_alloc_intrstatus(int source, int flags, uint32_t intrstatusre
         vd->flags = VECDESC_FL_NONSHARED;
         if (handler) {
 #if CONFIG_APPTRACE_SV_ENABLE
-            non_shared_isr_arg_t *ns_isr_arg=malloc(sizeof(non_shared_isr_arg_t));
+            non_shared_isr_arg_t *ns_isr_arg=heap_caps_malloc(sizeof(non_shared_isr_arg_t), MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT)
             if (!ns_isr_arg) {
                 portEXIT_CRITICAL(&spinlock);
                 free(ret);

```

higaski, 2023-09-21:
That was a fast catch, thanks!

higaski, 2023-09-21:
> Hi @higaski
>
> The issue is discovered by this Kconfig: `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0`. The interrupt allocator internally allocated some buffer from the PSRAM, so causing the crash.
>
> ```diff [...]

Btw, I spot a missing `;` at the second change.

closed 2023-10-07 by espressif-bot