espressif/esp-idf #12273: error: implicit declaration of function 'htole16'

franz-ms-muc, 2023-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.1.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

No Error

### What is the actual behavior?

Error

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
https://gist.github.com/franz-ms-muc/b70d3d13a259acf3dbf5303eaaa4651a
```

franz-ms-muc, 2023-09-19:
LINUX: same Error.

igrr, 2023-09-20:
> -IC:/git/krone/ksc-2n/src/DLL -IC:/git/krone/ksc-2n/src/OS/os_wrapper/include

Very likely the issue may be caused by a conflicting header file in one of these include directories which you are adding globally. Probably wpa_supplicant is including some header file, and it gets found in these directories instead of the place where it should be found.

If that is the case, you can fix the issue by adding include directories just in your components, not globally for the whole application.

franz-ms-muc, 2023-09-26:
i will check this ASAP

franz-ms-muc, 2023-09-26:
no Errors anymore, just Warnings. must rename endian.c/h to os\_endian.c/h, then it works.

closed 2023-09-26 by franz-ms-muc