espressif/esp-idf #12274: Guru Meditation Error: Core  0 panic'ed (IllegalInstruction).

matislovas, 2023-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Olimex ESP32-POE-ISO Wroom 32E

### Power Supply used.

USB

### What is the expected behavior?

No core panic

### What is the actual behavior?

I am using ESP32 in gattc mode, and it sometimes panics, when BLE signal is on the edge of disconnect. I recently updated `esp-idf` version from 4.4.5 to 5.1.1, but this issue still persists.

### Steps to reproduce.

1. Setup BLE Gattc.
2. Put server device in near-out-of-range location.

### Debug Logs.

```plain
ASSERT_PARAM(0 0), in lld_evt.c at line 1487
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008fc94: f01d020c 00004136 f01d0000
0x4008fc94: btdm_sleep_check_duration at /home/matas/esp/esp-idf/components/bt/controller/esp32/bt.c:909

Core  0 register dump:
PC      : 0x4008fc9b  PS      : 0x00060c34  A0      : 0x80083baa  A1      : 0x3ffc1be0  
0x4008fc9b: r_assert at /home/matas/esp/esp-idf/components/bt/controller/esp32/bt.c:1849

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f410046  
A6      : 0x000005cf  A7      : 0xfffffffb  A8      : 0x8000814b  A9      : 0x3ffc1b50  
A10     : 0x00000000  A11     : 0x3ffc1b74  A12     : 0x3ffc1b1f  A13     : 0x00000037  
A14     : 0x00000000  A15     : 0x3ffc1b25  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffe  
0x400012c5: strcmp in ROM

0x400012d5: strcmp in ROM

Backtrace: 0x4008fc98:0x3ffc1be0 0x40083ba7:0x3ffc1c00 0x4008778f:0x3ffc1c20 0x4008416e:0x3ffc1c60 0x4008a0de:0x3ffc1c80 0x4008ad3f:0x3ffc1ca0 0x40083479:0x3ffc1cc0 0x400833e5:0x3ffc1ce0 0x400833c7:0x00000000 |<-CORRUPTED
0x4008fc98: r_assert at /home/matas/esp/esp-idf/components/bt/controller/esp32/bt.c:1848

0x40083ba7: r_assert_param at ??:?

0x4008778f: r_lld_evt_schedule at ??:?

0x4008416e: r_ea_finetimer_isr at ??:?

0x4008a0de: r_rwbt_isr at ??:?

0x4008ad3f: r_rwbtdm_isr_wrapper at intc.c:?

0x40083479: hli_c_handler at /home/matas/esp/esp-idf/components/bt/controller/esp32/hli_api.c:92

0x400833e5: _highint4_stack_switch at /home/matas/esp/esp-idf/components/bt/controller/esp32/hli_vectors.S:177

0x400833c7: xt_highint4 at /home/matas/esp/esp-idf/components/bt/controller/esp32/hli_vectors.S:161
```

matislovas, 2023-09-20:
I had similar [issue](https://github.com/espressif/esp-idf/issues/12182) already closed, but I'm seeing a bit different stack trace on this one.

zhp0406, 2023-09-20:
@matislovas 
yes,thanks for report the issue.
could you please provide  complete logs and elf file(or code better ) to help me reproduce the problem?

zhp0406, 2023-09-20:
@matislovas 
sure,You can send it to me via email and let me know the reproduction process. My email is [zhanghaipeng@espressif.com].

matislovas, 2023-09-20:
Sent!

zhp0406, 2023-09-21:
@matislovas 
thanks,Has the message been sent successfully on your end? I haven't received it here. Could you please tell me your email address?

zhp0406, 2023-09-21:
@matislovas 
Due to some filtering policies, the email didn't get sent. Could you please try sending the file again **after packaging it**? Thank you.

zhp0406, 2023-09-21:
@matislovas 
received，I will check it ...

zhp0406, 2023-09-21:
@matislovas 
We have located the issue based on the information you provided. In order to ensure your continued use, I can offer you a workaround. It involves replacing the "components/bt/controller/lib_esp32/esp32/libbtdm_app.a" file with the one I will send you. 

[esp32c3.zip](https://github.com/espressif/esp-idf/files/12684181/esp32c3.zip)

However, please note that this is not a permanent solution.

Could you please provide some details about when the issue generally occurs（Does the issue generally occur when the signal is weak?）? Additionally, could you explain how I can reproduce this problem?

Furthermore, if we are unable to reproduce the issue on our end, we may need to add some logs for analysis purposes. We will require your assistance in testing to help us further investigate.

matislovas, 2023-09-21:
Generally this occurs in random intervals (have 1 device connected), 2-3 times a day, and mostly, when device is far from ESP (5-7 meters) and RSSI is less than ~95dBm (reported by ESP).

Yeah, I can add additional logs for the debugging purposes.

matislovas, 2023-09-22:
@zhp0406 

So with your provided lib, there's a linker error, any ideas?

```
[3/5] Linking CXX executable Olimex_BT_GW.elfFAILED: Olimex_BT_GW.elf 
: && /home/matas/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=/home/matas/Devel/softdent/esp32/build/Olimex_BT_GW.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld CMakeFiles/Olimex_BT_GW.elf.dir/project_elf_src_esp32.c.obj -o Olimex_BT_GW.elf -L/home/matas/esp/esp-idf/components/soc/esp32/ld   -L/home/matas/esp/esp-idf/components/esp_rom/esp32/ld   -L/home/matas/Devel/softdent/esp32/build/esp-idf/esp_system/ld   -L/home/matas/esp/esp-idf/components/esp_phy/lib/esp32   -L/home/matas/esp/esp-idf/components/esp_coex/lib/esp32   -L/home/matas/esp/esp-idf/components/esp_wifi/lib/esp32   -L/home/matas/esp/esp-idf/components/bt/controller/lib_esp32/esp32 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/bt/libbt.a  -lbtdm_app  -u ld_include_hli_vectors_bt  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/matas/esp/esp-idf/components/esp_c [... cut ...]
/home/matas/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: llc_hci.c:(.text+0x737): undefined reference to `btdm_gpio_set_level'
/home/matas/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/matas/esp/esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a(llc_hci.o): in function `hci_disconnect_cmd_handler':
llc_hci.c:(.text+0x83a): undefined reference to `btdm_gpio_set_level'
```

It works fine with stock `libbtdm_app.a`.

zhp0406, 2023-09-22:
@matislovas 
sorry for that,I seem to have made some mistakes,I will check ...

matislovas, 2023-09-23:
Yeah, now it builds! Will try for couple days and will get back with results.

matislovas, 2023-09-25:
After couple days of work, ESP didn't throw me that error. So I can carefully say  the workaround works. Should I expect any side effects from using yours provided `libbtdm_app.a`?

zhp0406, 2023-09-26:
@matislovas 
There are no any side effects , but because it's difficult to reproduce, we haven't yet identified the root cause of the issue.

mjaneczek, 2023-11-05:
```
ASSERT_PARAM(0 0), in lld_evt.c at line 1487
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090518: 00f01d00 00004136 f01d0000
Core  0 register dump:
PC      : 0x4009051f  PS      : 0x00060234  A0      : 0x80085d31  A1      : 0x3ffc2f94  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f41614e
A6      : 0x000005cf  A7      : 0xfffffffb  A8      : 0x8000814b  A9      : 0x3ffc2f04  
A10     : 0x00000000  A11     : 0x3ffc2f28  A12     : 0x3ffc2ed3  A13     : 0x00000037
A14     : 0x00000000  A15     : 0x3ffc2ed9  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40085c39  LEND    : 0x40085c41  LCOUNT  : 0x00000000

Backtrace: 0x4009051c:0x3ffc2f94 |<-CORRUPTED
```

I am getting the same error with this https://github.com/espressif/esp-idf/issues/12182 update, I doubt it helps but wanted to let you it happened only only for the issue creator.

zhp0406, 2023-11-06:
@mjaneczek 
Does the updated lib（[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/12706484/libbtdm_app.zip)） work for you?

mjaneczek, 2023-11-06:
@zhp0406 I am using your update now, today I got:

```
ASSERT_PARAM(0 0), in lld_pdu.c at line 883
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400905c0: 00f01d00 00004136 f01d0000
Core  0 register dump:
PC      : 0x400905c7  PS      : 0x00060234  A0      : 0x80085c02  A1      : 0x3ffc2f70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f4163a2
A6      : 0x00000373  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2ee0  
A10     : 0x00000000  A11     : 0x3ffc2f03  A12     : 0x3ffc2eaf  A13     : 0x00000033
A14     : 0x00000000  A15     : 0x3ffc2eb4  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400920a5  LEND    : 0x400920b5  LCOUNT  : 0xfffffffe

Backtrace: 0x400905c4:0x3ffc2f70 0x40085bff:0x3ffc2f90 0x4008b0be:0x3ffc2fb0 0x4008b299:0x3ffc2ff0 0x40089801:0x3ffc3030 0x400861c2:0x3ffc3070 0x4008c0f6:0x3ffc3090 0x4008cd57:0x3ffc30b0 0x40085582:0x3ffc30d0 0x400858a1:0x3ffc30f0 0x40085883:0x00000000 |<-CORRUPTED
```

The line is different: `ASSERT_PARAM(0 0), in lld_pdu.c at line 883` might be more related to https://github.com/espressif/esp-idf/issues/11635

matislovas, 2023-12-05:
Hey @zhp0406 , any progress on this? We want to launch our product in the beginning of next year based on `esp32` and this is a blocker for us. This bug appears, when there's a big and long data transfers taking place between esp and ble server device.

zhp0406, 2023-12-05:
@matislovas 
Hi，
The MR has been submitted and will be merged into all branches. Currently, it is in the testing process, and this process will likely take **around 10 days**. After the MR is completed, we will inform you of the commit.

mjaneczek, 2023-12-14:
```
ASSERT_PARAM(0 0), in lld_pdu.c at line 883
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400905c0: 00f01d00 00004136 f01d0000
Core  0 register dump:
PC      : 0x400905c7  PS      : 0x00060e34  A0      : 0x80085c02  A1      : 0x3ffc2f70
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f4163a2  
A6      : 0x00000373  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2ee0
A10     : 0x00000000  A11     : 0x3ffc2f03  A12     : 0x3ffc2eaf  A13     : 0x00000033  
A14     : 0x00000000  A15     : 0x3ffc2eb4  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400920a5  LEND    : 0x400920b5  LCOUNT  : 0xfffffffe  

Backtrace: 0x400905c4:0x3ffc2f70 0x40085bff:0x3ffc2f90 0x4008b0be:0x3ffc2fb0 0x4008b299:0x3ffc2ff0 0x40089801:0x3ffc3030 0x400861c2:0x3ffc3070 0x4008c0f6:0x3ffc3090 0x4008cd57:0x3ffc30b0 0x40085582:0x3ffc30d0 0x400858a1:0x3ffc30f0 0x40085883:0x00000000 |<-CORRUPTED
```

I have downloaded a lib from your commit and it happened after a few hours of working. @zhp0406

zhp0406, 2023-12-18:
@mjaneczek 
Your library probably didn't update successfully; the new library shouldn't assert at line 883 anymore.

Moreover, in the latest release v5.1, our modifications have already taken effect, and you can also make use of the latest release/v5.1.

commit 5f2c8edd27e60d5c98f16a2f107133de86fd3ec1
Author: chenjianhua <chenjianhua@espressif.com>
Date:   Fri Nov 24 20:16:03 2023 +0800

    fix(bt): Update bt lib for ESP32(fa43201)

    - Fixed assert for instant passed workaround
    - Fixed scan evt timeout
    - Fixed random address setting when scanning

matislovas, 2023-12-18:
@zhp0406  I tried new release of 5.1 (through the provided Docker [image](https://hub.docker.com/layers/espressif/idf/release-v5.1/images/sha256-e3fb8c2a6d7d79d4feca93e1926ee06498e01abdf9e05635eda566bc41b9535b?context=explore)), but it still fails with those errors:

```
ASSERT_PARAM(10 0), in lld_pdu.c at line 672
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090234: f01d020c 00004136 f01d0000
0x40090234: btdm_sleep_check_duration at /opt/esp/idf/components/bt/controller/esp32/bt.c:911

Core  1 register dump:
PC      : 0x4009023b  PS      : 0x00060334  A0      : 0x80083be6  A1      : 0x3ffc1c30  
0x4009023b: r_assert at /opt/esp/idf/components/bt/controller/esp32/bt.c:1848

A2      : 0x00000000  A3      : 0x0000000a  A4      : 0x00000000  A5      : 0x3f41113a  
A6      : 0x000002a0  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc1ba0  
A10     : 0x00000000  A11     : 0x3ffc1bc3  A12     : 0x3ffc1b6f  A13     : 0x00000032  
A14     : 0x00000000  A15     : 0x3ffc1b74  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffe  
0x400012c5: strcmp in ROM

0x400012d5: strcmp in ROM

Backtrace: 0x40090238:0x3ffc1c30 0x40083be3:0x3ffc1c50 0x40088b5d:0x3ffc1c70 0x40087281:0x3ffc1cb0 0x40086ece:0x3ffc1cd0 0x40087b55:0x3ffc1cf0 0x400895e0:0x3ffc1d10 0x40089d99:0x3ffc1d30 0x4008ae0f:0x3ffc1d50 0x40083475:0x3ffc1d70 0x400833e1:0x3ffc1d90 0x400833c3:0x00000000 |<-CORRUPTED
0x40090238: r_assert at /opt/esp/idf/components/bt/controller/esp32/bt.c:1847

0x40083be3: r_assert_param at ??:?

0x40088b5d: r_lld_pdu_rx_handler at ??:?

0x40087281: r_lld_evt_end at ??:?

0x40086ece: r_lld_evt_end_isr at ??:?

0x40087b55: lld_adjust_after_reset at ??:?

0x400895e0: ble_master_soft_rst at ??:?

0x40089d99: r_rwble_isr at ??:?

0x4008ae0f: r_rwbtdm_isr_wrapper at intc.c:?

0x40083475: hli_c_handler at /opt/esp/idf/components/bt/controller/esp32/hli_api.c:92

0x400833e1: _highint4_stack_switch at /opt/esp/idf/components/bt/controller/esp32/hli_vectors.S:177

0x400833c3: xt_highint4 at /opt/esp/idf/components/bt/controller/esp32/hli_vectors.S:161

```

zhp0406, 2023-12-19:
@matislovas 
I will check

mjaneczek, 2023-12-19:
Sorry for adding more mess to the situation, you are right I was still using the old one, I tried to get new lib from here:
https://github.com/espressif/esp32-bt-lib/blob/master/esp32/libbtdm_app.a

But getting: `skipping incompatible libbtdm_app.a when searching for -lbtdm_app` 

Double checked that I was able to compile with the lib u provided in this thread in zip, not sure what I am doing wrong, is this some dev built or something, if you could give me a tip on how to make this work I will be happy to test and help with the issue.

Once again, I am sorry about the wrong information.

matislovas, 2023-12-19:
@zhp0406  sent you via wetransfer

zhp0406, 2024-03-14:
@matislovas 
We have fixed the issue across all branches. If there are any additional problems after the fix, please create a new issue and provide a complete log. We will respond to you as soon as possible. Please test again using the patched release version. 
Thank you.
Below are the specific commits for the fix：
master 74f99d267939674b7e67813aa12199d14ed1aa28
v5.2 6139b362a061fc6ed39522aaa7b8b9d50652299e
v5.1 5f2c8edd27e60d5c98f16a2f107133de86fd3ec1
v5.0 51ba771176e1a16560d86fd87cf8d1506f4e54af
v4.4 4e518185c557048179a25bf8b527f71cfa910e9f
v4.3 79e80d913432026b809aa5e16e318dde080136dd

zhp0406, 2024-03-15:
@mjaneczek 
Which specific commit are you currently using? I have made a patched fix based on the latest release/v5.1 (BT controller compile version [664bcc7]). Please directly replace the file at esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a
[esp32.zip](https://github.com/espressif/esp-idf/files/14612695/esp32.zip)

zhp0406, 2024-03-15:
@mjaneczek 
If you need to create a lib based on a specific commit, please tell me the IDF commit you are using, or provide me with the complete boot log.

zhp0406, 2024-03-18:
base on commit [f5e484fc79e38e668768735b4b63dd976cb4266b](https://github.com/espressif/esp-idf/issues/12274)
[445f19e.zip](https://github.com/espressif/esp-idf/files/14632434/445f19e.zip)

AxelLin, 2024-06-06:
> @matislovas We have fixed the issue across all branches. 

Then this issue should be closed.

closed 2024-06-06 by weiyuhannnn