espressif/esp-idf #12275: Error while connecting to a website via IPv6 connection

my-dudhwala, 2023-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am getting a lot of errors while trying to work on any web project, the only and only problem is IPv6 router, I have tried a lot of codes from different articles, git repos and inbuilt examples with different configurations, most of are working with IPv4,

**I am updating each codes with its error on on a github repo**
https://github.com/my-dudhwala/ESP_IPv6/

Error in esp_http_client code,

E (13733) esp-tls: [sock=54] select() timeout
E (13734) TRANSPORT_BASE: Failed to open a new connection: 32774
E (13735) HTTP_CLIENT: Connection failed, sock < 0
E (13740) HTTP_CLIENT: HTTP GET request failed: ESP_ERR_HTTP_CONNECT

david-cermak, 2023-10-11:
Have you tried to enable `CONFIG_LWIP_IPV6_AUTOCONFIG` ?

Please check the IDF docs in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#ipv6-support

sgryphon, 2024-02-24:
I am not sure if it is relevant for your scenario, but I recently put up a pull request to update the protocol examples to work with all/any network configuration. The branch is here: https://github.com/espressif/esp-idf/pull/13250

The update examples work out of the box (no config or code changes) with all network types: IPv4-only, IPv6-only, and dual stack.

I have tested the http_request example with the changes and there is example output in the README of it working on different network types and different servers (dual stack, IPv6 only, IPv4 only).

If you have DNS64/NAT64 available, then it even works for an IPv6-only network to connect to an IPv4-only server.

Maybe you can use something out of the examples to get your scenario working.

david-cermak, 2024-03-11:
Hi @my-dudhwala 

I'm closing this, as it seems your issues are unrelated to IDF's IPv6 handling. If you get more information, please feel free to share it here and reopen this issue.

closed 2024-03-11 by david-cermak

sgryphon, 2024-03-11:
Just a note that the above example changes only ended up working for HTTP, by doing a manual DNS lookup (to support some scenarios connecting to a dual-stack host from an IPv6 only network).

However the approach did not work with HTTPS, which does it's own lookup. Most production use will require HTTPS.

I tracked down the issue to the way LWIP does DNS result ordering, and have a change request up with a fix, but it is not merged yet, so there are still some IPv6 scenarios that ESP-IDF does not yet handle correctly.