espressif/esp-idf #12277: Support LE Audio on ESP32-C6

xobs, 2023-09-20:
### Is your feature request related to a problem?

Bluetooth 5.2 added support for LE Audio. The ESP32-C6 supports BLE 5.3, so it should be able to support LE Audio.

### Describe the solution you'd like.

Ideally, esp-idf would support some form of LE Audio. This could take several forms:

* Directly streaming the LC3 frames to/frm somewhere. possibly to a host processor
* Decoding or Encoding LC3 and presenting frames on the device
* Streaming LE Audio to or from an external device such as via I2S or using the PCM / ADC modules

### Describe alternatives you've considered.

The easiest solution is to just use an ESP32 with classic Bluetooth. However, this does not provide newer wifi or Bluetooth standards, so it is not ideal.

weiyuhannnn, 2023-09-22:
Hi @xobs. Thank you for your feature request.  We don't support LE Audio on ESP32-C6 currently. If this feature is of great importance to you, please contact our [customer support team](https://www.espressif.com/en/contact-us/sales-questions), and they can provide further assistance. Thanks!

pavel-cernohorsky, 2023-09-22:
Hello, considering this is one of the big / main features of the new Bluetooth standard which this chip supports (at least on paper, haven't had real sample in my hands yet), I am not sure it is a good idea that if all the people who want to use it will write to the customer support. But OK, I will do so as well. Hope it speeds up work on this feature.

weiyuhannnn, 2023-09-25:
Hello @pavel-cernohorsky, 

Thank you for your feedback!  Just to clarify,  the ESP32-C6 does not support LE Audio.  However, we are actively planning to integrate this feature into our upcoming chips. 

Our customer support team will be more than happy to assist you in collecting specific requirements related to this feature for your product. This will help us tailor our future offerings to better meet your product's needs. Your input is highly appreciated.  For information about our new chip roadmap, please also consult our customer support team.

pavel-cernohorsky, 2023-09-25:
Hello @weiyuhannnn,

Thank you very much for your response. This information indeed seems very strange to me, because as [BLE Specification](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=544799) says:

> This specification is compatible with the Bluetooth Core Specification, Version 5.2 [1] or later.

And as [your web pages](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf) say:

> Bluetooth LE: Bluetooth 5.3 certified

From that I would assume, that while claiming compatibility to 5.3, ESP32-C6 has to have everything which is needed for BLE Audio, considering the BLE Audio specification says that it is compatible with 5.2 or later. But I am not expert in this area.

Anyway, thank you very much for your answer and clarification. As searching the Internet, I wasn't the only person puzzled (as this issue, not started by me, also shows).

Looking forward to some next product line which will support BLE Audio.

Kind regards,
Pavel

mringwal, 2023-10-02:
The Bluetooth Core Specification only defines how things should work if they are implemented, and pretty much all features are optional.

LE Audio requires a significant extension of the Link Layer. As the Link Layer is implemented in software, it's probable that it could be added in a software update, as the hardware requirement is mainly 2M-PHY, which is supported by the newer Controller.

vwwwvwww, 2024-02-26:
i want LE Audio too

closed 2024-03-01 by Alvin1Zhang

jasmohan-narula, 2024-07-24:
> Hello @pavel-cernohorsky,
>
> Thank you for your feedback! Just to clarify, the ESP32-C6 does not support LE Audio. **However, we are actively planning to integrate this feature into our upcoming [...]

@weiyuhannnn Does ESP32-S3 support Bluetooth LE Audio? Also, can I get a timeline of when we will get your new chips supporting LE Audio?

weiyuhannnn, 2024-07-25:
@jasmohan-narula  @vwwwvwww Thanks for your interest in our Espressif product! 

- ESP32-S3 does not support LE Audio.
- Our new-released chip: [ESP32-H4](https://www.espressif.com/en/news/ESP32-H4) has supported LE Audio! For inquiries about the ESP32-H4 SoC, please contact Espressif’s [customer support team](https://www.espressif.com/en/contact-us/sales-questions).

jasmohan-narula, 2024-07-25:
> @jasmohan-narula  @vwwwvwww Thanks for your interest in our Espressif product!
>
> - ESP32-S3 does not support LE Audio.
> - Our new-released chip: [ESP32-H4](https://www.espressif.com/en/news/ESP32-H4) [...]

@weiyuhannnn ESP32-H4 does not have WiFi.

Is there any chip which supports both WiFi and Bluetooth LE Audio?

expresspotato, 2024-09-30:
We want LE audio!

expresspotato, 2024-11-09:
> The Bluetooth Core Specification only defines how things should work if they are implemented, and pretty much all features are optional.
>
> LE Audio requires a significant extension of the Link Layer. [...]

Ouch that this gets closed with "Won't do" considering how popular ESP32's are for Bluetooth audio... With the xtensa cores being phased out, and no Bluetooth Classic support at all on the newer chips, where does that leave us?

weiyuhannnn, 2024-11-11:
Hello @expresspotato，

This issue is about the C6 support for LE Audio, so it was closed with ”Won't do“.

We have already acknowledged the growing demand for Bluetooth audio, and are actively taking user needs into consideration. [H4 ](https://www.espressif.com/en/news/ESP32-H4) is our first chip to support LE Audio. Please stay tuned to our website for the latest updates on chip developments, or feel free to contact [customer support](https://www.espressif.com/en/contact-us/technical-inquiries) for more information.

We appreciate your feedback!

nathmo, 2024-12-10:
Is there an ETA before the ESP32-H4 hit the market ? An expected release date ? The information are really scarce, and I never got a reply from customer support :(

mintisan, 2025-08-26:
Yes, WiFi 6 and Bluetooth LE Audio are what we want.

fkjh14, 2025-09-24:
Pls BLE Audio on C6 we need it

jasmohan-narula, 2025-09-24:
> We implemented LE Audio on a Waveshare ESP32-S3 box by adding an nRF54L15 module :)
> https://bluekitchen-gmbh.com/le-audio-on-esp32/

It would be great if we get LE Audio support on the ESP32 module itself instead of relying on an external module.