espressif/esp-idf #12278: CMake was unable to find a build program corresponding to "Ninja".

TritiumXs, 2023-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

The firmware was built and flashed into ESP32.

### What is the actual behavior?

Some errors appeared on the information screen.

### Steps to reproduce.

1. Build this firmware
2. Flash the firmware into ESP32

### Build or installation Logs.

```plain
====================[ 构建 | flash | ESP32 ]======================================
"F:\Program Files\JetBrains\CLion 2021.3.4\bin\cmake\win\bin\cmake.exe" --build F:\espressif__led_strip\examples\led_strip_spi_ws2812\cmake-build-esp32 --target flash
[1/6] cmd.exe /C "cd /D F:\espressif__led_strip\examples\led_strip_spi_ws2812\cmake-build-esp32\esp-idf\esptool_py && C:\Users\Tritium\.espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe F:/Espressif/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app F:/espressif__led_strip/examples/led_strip_spi_ws2812/cmake-build-esp32/partition_table/partition-table.bin F:/espressif__led_strip/examples/led_strip_spi_ws2812/cmake-build-esp32/led_strip_spi_ws2812.bin"
led_strip_spi_ws2812.bin binary size 0x327b0 bytes. Smallest app partition is 0x100000 bytes. 0xcd850 bytes (80%) free.
[2/6] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure F:/espressif__led_strip/examples/led_strip_spi_ws2812/cmake-build-esp32/bootloader-prefix/src/bootloader-stamp/bootloader-configure 
cmd.exe /C "cd /D F:\espressif__led_strip\examples\led_strip_spi_ws2812\cmake-build-esp32\bootloader && "F:\Program Files\JetBrains\CLion 2021.3.4\bin\cmake\win\bin\cmake.exe" -DSDKCONFIG=F:/espressif__led_strip/examples/led_strip_spi_ws2812/sdkconfig -DIDF_PATH=F:/Espressif/esp-idf -DIDF_TARGET=esp32 -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:/Users/Tritium/.espressif/python_env/idf5.1_py3.10_env/Scripts/python.exe -DEXTRA_COMPONENT_DIRS=F:/Espressif/esp-idf/components/bootloader -DPROJECT_SOURCE_DIR=F:/espressif__led_strip/examples/led_strip_spi_ws2812 -GNinja F:/Espressif/esp-idf/components/bootloader/subproject && "F:\Program Files\JetBrains\CLion 2021.3.4\bin\cmake\win\bin\cmake.exe" -E touch F:/espressif__led_strip/examples/led_strip_spi_ws2812/cmake-build-esp32/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
See also "F:/espressif__led_strip/examples/led_strip_spi_ws2812/cmake-build-esp32/bootloader/CMakeFiles/CMakeOutput.log".
ninja: build stopped: subcommand failed.
```

### More Information.

I don't know if IDF_PATH is added to variables correctly. And when I said the argument of "CMAKE_MAKE_PROGRAM", it didn't work too.

dobairoland, 2023-09-22:
Have you followed the steps in the getting started guide? If you use the Windows Tool installer (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html) then it will create you a working environment. Does that environment work for you?

After that you can proceed to set it up ESP-IDF with Clion. It is not officially supported but you can find some guides.

dobairoland, 2023-09-22:
In that case this issue is not ESP-IDF specific. 

Oh, I just found out that we already have https://github.com/espressif/idf-clion-plugin. Have you tried it?

Here I found a setup guide: https://www.jetbrains.com/help/clion/esp-idf.html (edited this month so probably up-to-date)
This is an older one: https://blog.jetbrains.com/clion/2021/09/developing-for-esp32-with-clion-on-windows/

closed 2023-11-17 by TritiumXs