espressif/esp-idf #12280: WIFI-MESH: Loss of parent connection causes lockup, requires watchdog reboot

MisterScience5, 2023-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board, ESP32-DEVKITV1, ESP-WROVER-KIT

### Power Supply used.

USB

### What is the expected behavior?

Mesh network should gracefully handle loss of connectivity or packet loss.

Any errors on loss of connectivity should be non blocking.

### What is the actual behavior?

I am currently rewriting our firmware for v5.1 to hopefully fix some stability issues we've had in our previous firmware.

However, I am now facing a new problem. If there is poor mesh network connectivity, I will get the following error messages:
W (154519) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:81

Through reading the bug reports, it seems this message means no TX window was given from the parent/root.
To fix this, my units are free to disconnect from the mesh, find another parent or simply wait for the next sync window.

The problem arises because the TX window search seems to fully block my program, so I cannot disable the mesh or disconnect from the parent and search for another. I cannot gracefully restart either, the only thing I've found to be working is a task watchdog reboot.

This is my mesh enable function:

```
esp_err_t esp_mesh_enable()
{
    esp_mesh_sleep_allowed = false;
    if (!esp_mesh_active)
    {
        ESP_LOGI(MESH_TAG, "Starting mesh as %s", (basic_settings.is_root) ? "root" : "node");
        esp_err_t err = ESP_OK;
        if (!mesh_init_complete)
        {
            ESP_ERROR_CHECK(esp_netif_create_default_wifi_mesh_netifs(&netif_sta, NULL));
            wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
            ESP_ERROR_CHECK(esp_wifi_init(&config));
            ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
            ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
            ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_AP, mesh_settings.wifi_mode));
            ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_STA, mesh_settings.wifi_mode));
            ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
        }
        ESP_ERROR_CHECK(esp_wifi_start());
[... 259 lines cut ...]
W (165319) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:87
W (167119) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:88
W (168919) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:89
W (170719) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:90
W (172519) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:91
W (174319) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:92
W (176119) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:93
W (177919) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:94
W (179719) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:95
W (181519) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:96
W (183319) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:97
E (183451) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (183451) task_wdt:  - status_task (CPU 0)
E (183451) task_wdt: Tasks currently running:
E (183451) task_wdt: CPU 0: IDLE
E (183451) task_wdt: CPU 1: IDLE
E (183451) task_wdt: Aborting.
E (183451) task_wdt: Print CPU 0 (current core) backtrace
```

### More Information.

This problem did not seem to exist on our previos software which was running on v4.3, however as I've done a complete rewrite of the program there may be other factors affecting this.

I've also tried a lot of different things to handle this, however as far as I can tell, the moment I start getting RX window errors the only two things I've discovered that correct the problem is introducing another parent with a better signal for it to switch to automatically, or manually reboot the device.

MisterScience5, 2023-09-26:
I've been experimenting a lot with this, it seems to either be gone or less frequent after i did the following two changes:
- Disabled wifi/bluetooth SW controlled coexistence
- Changed the compiler optimization to performance instead of size

Sadly, I can't confirm which of these made a difference as I did them close in time and didn't register any errors meanwhile. I'll be able to confirm whether the error is fully gone or only less frequent in a few days as I have a large network of mesh nodes running and logging data.

I also made a better disable function to help with these sorts of deadlocks.

```
void esp_mesh_disable_task(void *arg)
{
    esp_err_t err = ESP_OK;
    esp_task_wdt_add(NULL);
    esp_mesh_disabling = true;
    err = esp_mesh_set_self_organized(false, false);
    ESP_LOGI(MESH_TAG, "Disabling self organized returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(20);
    err = esp_mesh_disconnect();
    ESP_LOGI(MESH_TAG, "Disconnecting from parent returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(20);
    err = esp_wifi_scan_stop();
    ESP_LOGI(MESH_TAG, "Stopping scan returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(20);
    err = esp_mesh_flush_scan_result();
    ESP_LOGI(MESH_TAG, "Flushing scan results returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(20);
    err = esp_mesh_flush_upstream_packets();
    ESP_LOGI(MESH_TAG, "Flushing upstream packets returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(20);
    esp_log_level_set("mesh", ESP_LOG_ERROR);
    err = esp_mesh_stop();
    ESP_LOGI(MESH_TAG, "Stopping mesh returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(100);
    err = esp_mesh_deinit();
    ESP_LOGI(MESH_TAG, "Deiniting mesh returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(100);
    esp_log_level_set("mesh", ESP_LOG_WARN);
    ESP_LOGW(MESH_TAG, "ESP mesh disabled!");
    err = esp_wifi_stop();
    ESP_LOGI(MESH_TAG, "Stopping wifi returned 0x%x (%s)", err, esp_err_to_name(err));
    sleep(100);
    esp_mesh_disabling = false;
    esp_task_wdt_delete(NULL);
    vTaskDelete(NULL);
}

esp_err_t esp_mesh_disable()
{
    uint32_t timeout = 0;
    mesh_tx_pending_t tx_pending = { 0, };
    esp_err_t err = ESP_OK;

    if(esp_mesh_active)
    {
        xTaskCreate(&esp_mesh_disable_task, "mesh_dis", 4096, NULL, 17, NULL);
        while(esp_mesh_disabling && timeout < 20000)
        {
            sleep(100);
        }
        if(!esp_mesh_active)
        {
            return ESP_OK;
        }
        return ESP_FAIL;
    } else {
        ESP_LOGE(MESH_TAG, "Mesh network already disabled!");
        return ESP_OK;
    }
    
    return err;
}
```

zhangyanjiaoesp, 2023-09-28:
@MisterScience5 
I think there are three things you can optimize in your enable function:
1. Call `esp_mesh_set_xon_qsize(128)` to enlarge the xon window .
2. Don't call `esp_mesh_set_self_organized(true, false)` since you have set the mesh network to `fixed-root`, the first parameter of the `esp_mesh_set_self_organized()` function means enable self-organized network, means the root node is automatically selected by voting.
3. The default mesh packet lifetime is BLOCKING(0xffffffff), you can call `esp_mesh_send_block_time()` to set the mesh packet lifetime, then the mesh node will not always show `W (32119) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:13`, it will return TX_TIMEOUT when the lifetime has reached.

MisterScience5, 2023-09-28:
Thank you for the tips, after reading other bugreports I did already implement the following in my esp_mesh_enable function:
esp_mesh_send_block_time(2000);
esp_mesh_set_xon_qsize(128);

As for the self organized, I take it you mean only for the root node? As I understand the docs, the nodes will need self organized to automatically connect to a parent, or have I misunderstood?

I have noticed the problem a lot less frequently, however I did manage to provoke it still by sending a lot of frames (~1300 frames of approx. 1024b) to the broadcast address, this caused the same lockup even with optimization 1 and 3 in place. The wifi seems to be stalling when too many packets build up, which causes esp_restart and network functions to become blocked indefinitely.

MisterScience5, 2023-10-05:
Another update, as the diagnostic states, I was using the Wifi LR protocol for these nodes. Having tested it with 802.11b/g/n, the problem seems to have gone away, so somehow this bug seems tied to using that protocol. It may be a timing issue?

To my greater surprise, I now seem to have a lot better connectivity, whereas when we started using the LR protocol a few years back, it gave us better results than b/g/n for devices with a low signal.

zhangyanjiaoesp, 2023-10-10:
The data rate LR protocol is 1/2 Mbps or 1/4 Mbps, it will give better results than b/g/n for devices with a low signal. 
Please refer to the doc on [when to use LR](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=long%20range#when-to-use-lr).

> I have noticed the problem a lot less frequently, however I did manage to provoke it still by sending a lot of frames (~1300 frames of approx. 1024B) to the broadcast address, this caused the same lockup

If you want to send a lot of frames, and your signal is not bad, suggest to set the 802.11 b/g/n protocol.

MisterScience5, 2023-10-12:
That is the expected result, but somehow we are now seeing better connections (more stable connections on a poor signal) with B/G/N than LR, however we prefer B/G/N, so this issue is not critical for me.

The main problem is that I do still see sometimes the lockup, which will trigger the task watchdog of my safety task, however the device will not reboot from the watchdog. Before, the lockup would happen during mesh use, but that has cleared up since moving to B/G/N. Now, the lockup so far has only happened when stopping the mesh.

On the device below, I have disabled bluetooth coex, but bluetooth is enabled in the SDK configuration. The device is set up to use a 32k crystal, however the device in question does not have one, and falls back to the internal 150kHz oscillator on boot.

Here is a log from the device when hit by the lockup, which tends to occur upon stopping the mesh:

```
I (11:17:33.866) esp_mesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (11:17:33.871) esp_mesh: <MESH_EVENT_ROOT_ADDRESS>root address:0c:dc:7e:c3:85:f5
I (11:17:33.870) esp_mesh: Got new time from root, node time was 1.7424 seconds compared to root
I (11:17:33.885) esp_mesh: <MESH_EVENT_STOPPED>
E (95759) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (95759) task_wdt:  - status_task (CPU 0/1)
E (95759) task_wdt: Tasks currently running:
E (95759) task_wdt: CPU 0: IDLE
E (95759) task_wdt: CPU 1: IDLE
E (95759) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400EFA7F:0x3FFBFD60 0x40082FB1:0x3FFBFD90 0x4008979B:0x3FFB2520 0x400EDBA5:0x3FFB2540 0x400D4A6B:0x3FFB2560 0x4008CAEC:0x3FFB2580
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400EFA7F: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x40082FB1: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008979B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400EDBA5: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400D4A6B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008CAEC: [WinError 2] Systemet finner ikke angitt fil

E (95759) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400863BF:0x3FFC0370 0x40082FB1:0x3FFC0390 0x4008979B:0x3FFB2C80 0x400EDBA5:0x3FFB2CA0 0x400D4A6B:0x3FFB2CC0 0x4008CAEC:0x3FFB2CE0
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400863BF: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x40082FB1: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008979B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400EDBA5: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400D4A6B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008CAEC: [WinError 2] Systemet finner ikke angitt fil

E (155809) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (155809) task_wdt:  - status_task (CPU 0/1)
E (155809) task_wdt: Tasks currently running:
E (155809) task_wdt: CPU 0: IDLE
E (155809) task_wdt: CPU 1: IDLE
E (155809) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400EFA7F:0x3FFBFD60 0x40082FB1:0x3FFBFD90 0x4008979B:0x3FFB2520 0x400EDBA5:0x3FFB2540 0x400D4A6B:0x3FFB2560 0x4008CAEC:0x3FFB2580
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400EFA7F: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x40082FB1: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008979B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400EDBA5: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400D4A6B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008CAEC: [WinError 2] Systemet finner ikke angitt fil

E (155809) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400863BF:0x3FFC0370 0x40082FB1:0x3FFC0390 0x4008979B:0x3FFB2C80 0x400EDBA5:0x3FFB2CA0 0x400D4A6B:0x3FFB2CC0 0x4008CAEC:0x3FFB2CE0
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400863BF: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x40082FB1: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008979B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400EDBA5: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400D4A6B: [WinError 2] Systemet finner ikke angitt fil
[... cut ...]
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x400D4A6B: [WinError 2] Systemet finner ikke angitt fil
xtensa-esp32-elf-addr2line -pfiaC -e c:\ESP\ilume\ilume\build\ilume.elf 0x4008CAEC: [WinError 2] Systemet finner ikke angitt fil

E (456079) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (456079) task_wdt:  - status_task (CPU 0/1)
E (456079) task_wdt: Tasks currently running:
E (456079) task_wdt: CPU 0: IDLE
E (456079) task_wdt: CPU 1: IDLE
E (456079) task_wdt: Print CPU 0 (current core) backtrace
```

MisterScience5, 2023-10-12:
Here is the sdkconfig for the software running on the crashed device:

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.1 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=20
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=9
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_IDCACHE_PER_CORE=y
CONFIG_SOC_CPU_CORES_NUM=2
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FPU=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
[... 1946 lines cut ...]
# CONFIG_PPP_MPPE_SUPPORT is not set
# CONFIG_PPP_DEBUG_ON is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

zhangyanjiaoesp, 2023-10-13:
@MisterScience5 
Can you provide the complete log and the `elf` file for the watchdog device?

MisterScience5, 2023-10-13:
Sure, is there a private address I could send the elf file and log too? The lof is fairly large, and I would like the not publish the elf file if possible.

Here is a log cutout with addr2line available, I had some path issues:

```
E (879809) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (879809) task_wdt:  - status_task (CPU 0/1)
E (879809) task_wdt: Tasks currently running:
E (879809) task_wdt: CPU 0: IDLE
E (879809) task_wdt: CPU 1: IDLE
E (879809) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400EFC3F:0x3FFBFD60 0x40082FB1:0x3FFBFD90 0x4008979B:0x3FFB2520 0x400EDD65:0x3FFB2540 0x400D4A73:0x3FFB2560 0x4008CAEC:0x3FFB2580
0x400efc3f: task_wdt_timeout_handling at C:/Users/prebe/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at C:/Users/prebe/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40082fb1: _xt_lowint1 at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1236

0x4008979b: xt_utils_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/cpu.c:111

0x400edd65: esp_pm_impl_waiti at C:/Users/prebe/esp/esp-idf/components/esp_pm/pm_impl.c:834

0x400d4a73: esp_vApplicationIdleHook at C:/Users/prebe/esp/esp-idf/components/esp_system/freertos_hooks.c:57

0x4008caec: prvIdleTask at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327

E (879809) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400863BF:0x3FFC0370 0x40082FB1:0x3FFC0390 0x4008979B:0x3FFB2C80 0x400EDD65:0x3FFB2CA0 0x400D4A73:0x3FFB2CC0 0x4008CAEC:0x3FFB2CE0
0x400863bf: esp_crosscore_isr at C:/Users/prebe/esp/esp-idf/components/esp_system/crosscore_int.c:96

0x40082fb1: _xt_lowint1 at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1236

0x4008979b: xt_utils_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/cpu.c:111

0x400edd65: esp_pm_impl_waiti at C:/Users/prebe/esp/esp-idf/components/esp_pm/pm_impl.c:834

0x400d4a73: esp_vApplicationIdleHook at C:/Users/prebe/esp/esp-idf/components/esp_system/freertos_hooks.c:57

0x4008caec: prvIdleTask at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327

E (939859) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (939859) task_wdt:  - status_task (CPU 0/1)
E (939859) task_wdt: Tasks currently running:
E (939859) task_wdt: CPU 0: IDLE
E (939859) task_wdt: CPU 1: IDLE
E (939859) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400EFC3F:0x3FFBFD60 0x40082FB1:0x3FFBFD90 0x4008979B:0x3FFB2520 0x400EDD65:0x3FFB2540 0x400D4A73:0x3FFB2560 0x4008CAEC:0x3FFB2580
0x400efc3f: task_wdt_timeout_handling at C:/Users/prebe/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at C:/Users/prebe/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40082fb1: _xt_lowint1 at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1236

0x4008979b: xt_utils_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/cpu.c:111

0x400edd65: esp_pm_impl_waiti at C:/Users/prebe/esp/esp-idf/components/esp_pm/pm_impl.c:834

0x400d4a73: esp_vApplicationIdleHook at C:/Users/prebe/esp/esp-idf/components/esp_system/freertos_hooks.c:57

0x4008caec: prvIdleTask at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327

E (939859) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400863BF:0x3FFC0370 0x40082FB1:0x3FFC0390 0x4008979B:0x3FFB2C80 0x400EDD65:0x3FFB2CA0 0x400D4A73:0x3FFB2CC0 0x4008CAEC:0x3FFB2CE0
0x400863bf: esp_crosscore_isr at C:/Users/prebe/esp/esp-idf/components/esp_system/crosscore_int.c:96

0x40082fb1: _xt_lowint1 at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1236

0x4008979b: xt_utils_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/cpu.c:111

0x400edd65: esp_pm_impl_waiti at C:/Users/prebe/esp/esp-idf/components/esp_pm/pm_impl.c:834

0x400d4a73: esp_vApplicationIdleHook at C:/Users/prebe/esp/esp-idf/components/esp_system/freertos_hooks.c:57

0x4008caec: prvIdleTask at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327
```

Xiehanxin, 2023-10-16:
hi @MisterScience5 ,could you provide the elf file and log by sending mail to  _sales@espressif.com_  , Please note that this is an issue from GitHub with the link. and please at me after you sending the email that we will know as soon as possible

zhangyanjiaoesp, 2023-10-19:
@MisterScience5  If you set the `RTC clock source` to `RTC_CLK_SRC_INT_RC`, is the task watchdog issue still there?

MisterScience5, 2023-10-19:
@Xiehanxin I have sent the ELF file now, sorry for the delay!

@zhangyanjiaoesp I am creating a build now with the crystal disabled to test, I did try to change the failing units that failed their crystal startup to the 8MD256 clock source with the code at the bottom, but this did not work.

I did also see a crash on an ESP-WROVER-KIT which has a 32k crystal, this is the first crystal enabled device I have seen the lockup on.

However during the last 5 days I have not yet seen any lockups on production devices. We are testing with devices in production that are close enough to service rapidly. We have 34 pieces in production that have been running the new firmware for 5 days, and we are flashing up two new batches for a total of 119 production units this morning, to get more data.

```
esp_err_t system_check_system_clock()
{
    soc_rtc_slow_clk_src_t rtc_slow_clk = rtc_clk_slow_src_get();

    if (rtc_slow_clk == SOC_RTC_SLOW_CLK_SRC_XTAL32K)
    {
        system_status.clock_error = false;
        return ESP_OK;
    }

    ESP_LOGE(CORE_TAG, "System clock error, could not start 32k crystal!");
    rtc_clk_8m_enable(true, true);
    rtc_clk_slow_src_set(SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256);
    if(rtc_clk_slow_src_get() == SOC_RTC_SLOW_CLK_SRC_RC_FAST_D256)
    {
        ESP_LOGW(CORE_TAG, "Successfully enabled 8MD256 clock source");
    }
    system_status.clock_error = true;
    return ESP_FAIL;
}
```

MisterScience5, 2023-10-22:
Sadly, it seems I am now seeing production units experiencing lockups. None dropped off for the first week, but now I am seeing a number of them disappearing, 8 out of 109 devices have disappeared longer than the 24h fixed restart period. Presumably the reason they are suddenly disappearing at an increased rate is the poor wifi communications in the chain mesh provoking task watchdogs that do not complete.

So far, the test devices without the 32k crystal enabled seem stable, although they require more testing time to be sure.

zhangyanjiaoesp, 2023-10-23:
@MisterScience5 Thanks for your update, we will check the elf file you sent.

MisterScience5, 2023-10-23:
It seems the ELF file was bounced by the mailserver as the zipped ELF was not allowed. I have sent a new mail with a hosted link to the ELF file.

Xiehanxin, 2023-10-23:
hi @MisterScience5  could you tell me your email address

MisterScience5, 2023-10-23:
My address is preben(at)dyrholm.no

zhangyanjiaoesp, 2023-10-24:
@MisterScience5 
Can you add this changes to your code, and test whether the lockup still exist?

[image]

MisterScience5, 2023-10-24:
Recompiled with the changes now, and disabled the RTC WDT (which so far seems to have worked, although I have yet to confirm it with a device I have serial data from). Hopefully I will get a lockup soon so I can confirm:)

MisterScience5, 2023-10-29:
Sadly the lockup persists, it took a while to hit a device I have serial logging on, but I have steadily lost devices from the mesh after implementing the changes.

```I (23:19:21.154) core: Disabling mesh
I (23:19:21.174) esp_mesh: <MESH_EVENT_PARENT_DISCONNECTED>reason:0 (ESP_OK)
I (23:19:21.180) esp_mesh: <MESH_EVENT_PARENT_DISCONNECTED>reason:8 (ERROR)
I (23:19:21.195) esp_mesh: <MESH_EVENT_PARENT_DISCONNECTED>reason:106 (ERROR)
I (23:19:21.234) esp_mesh: <MESH_EVENT_ROOT_ADDRESS>root address:0c:dc:7e:c3:85:f5
I (23:19:21.238) esp_mesh: <MESH_EVENT_PARENT_CONNECTED>layer:-1-->0, parent:30:c6:f7:2f:b8:8d, ID:62:74:67:6c:79:73, duty:0, rssi:-67
I (23:19:21.247) esp_mesh: Requesting TSF time from parent 30:c6:f7:2f:b8:8d
I (23:19:21.257) esp_mesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (23:19:21.261) esp_mesh: <MESH_EVENT_ROOT_ADDRESS>root address:0c:dc:7e:c3:85:f5
I (23:19:21.260) esp_mesh: Got new time from root, node time was 0.5249 seconds compared to root
I (23:19:21.276) esp_mesh: <MESH_EVENT_STOPPED>
E (12146295) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (12146295) task_wdt:  - status_task (CPU 0/1)
E (12146295) task_wdt: Tasks currently running:
E (12146295) task_wdt: CPU 0: IDLE
E (12146295) task_wdt: CPU 1: IDLE
E (12146295) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400F01A3:0x3FFBFD60 0x40082FA9:0x3FFBFD90 0x40089793:0x3FFB2520 0x400EE2C9:0x3FFB2540 0x400D4A4F:0x3FFB2560 0x4008CB00:0x3FFB2580
0x400f01a3: task_wdt_timeout_handling at C:/Users/prebe/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:442
 (inlined by) task_wdt_isr at C:/Users/prebe/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585

0x40082fa9: _xt_lowint1 at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240

0x40089793: xt_utils_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/cpu.c:111

0x400ee2c9: xt_utils_get_core_id at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:36
 (inlined by) esp_cpu_get_core_id at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/include/esp_cpu.h:127
 (inlined by) xPortGetCoreID at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:631
 (inlined by) esp_pm_impl_idle_hook at C:/Users/prebe/esp/esp-idf/components/esp_pm/pm_impl.c:777

0x400d4a4f: esp_vApplicationIdleHook at C:/Users/prebe/esp/esp-idf/components/esp_system/freertos_hooks.c:48 (discriminator 1)

0x4008cb00: prvIdleTask at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327

E (12146295) task_wdt: Print CPU 1 backtrace

Backtrace: 0x400863B7:0x3FFC0370 0x40082FA9:0x3FFC0390 0x40089793:0x3FFB2C80 0x400EE2C9:0x3FFB2CA0 0x400D4A4F:0x3FFB2CC0 0x4008CB00:0x3FFB2CE0
0x400863b7: esp_crosscore_isr at C:/Users/prebe/esp/esp-idf/components/esp_system/crosscore_int.c:96

0x40082fa9: _xt_lowint1 at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240

0x40089793: xt_utils_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/cpu.c:111

0x400ee2c9: xt_utils_get_core_id at C:/Users/prebe/esp/esp-idf/components/xtensa/include/xt_utils.h:36
 (inlined by) esp_cpu_get_core_id at C:/Users/prebe/esp/esp-idf/components/esp_hw_support/include/esp_cpu.h:127
 (inlined by) xPortGetCoreID at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:631
 (inlined by) esp_pm_impl_idle_hook at C:/Users/prebe/esp/esp-idf/components/esp_pm/pm_impl.c:777

0x400d4a4f: esp_vApplicationIdleHook at C:/Users/prebe/esp/esp-idf/components/esp_system/freertos_hooks.c:48 (discriminator 1)

0x4008cb00: prvIdleTask at C:/Users/prebe/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327```

zhangyanjiaoesp, 2023-10-30:
@MisterScience5 Where is `status_task` defined and what does it do?

MisterScience5, 2023-10-30:
It is defined in the same file, the status task manages periodically running a function to get a lighting setting from a configured calendar, as well as printing the current status, The program is run on battery powered street lamps.

Here is the function:

```
__attribute__((noreturn)) void main_status_loop(void *arg)
{
    bool got_updated_lamp_settings = false;
#if SHOW_TASK_LIST
    char task_list_buffer[2000] = "";
#endif

    // time_t loop_start_time = esp_timer_get_time();

    ESP_ERROR_CHECK(esp_task_wdt_add(NULL));
#if ENABLE_RTC_WDT
    rtc_wdt_protect_off();      //Disable RTC WDT write protection
    //Set stage 0 to trigger a system reset after 1000ms
    rtc_wdt_set_stage(RTC_WDT_STAGE0, RTC_WDT_STAGE_ACTION_RESET_SYSTEM);
    rtc_wdt_set_time(RTC_WDT_STAGE0, 60000);
    rtc_wdt_enable();           //Start the RTC WDT timer
    rtc_wdt_protect_on();       //Enable RTC WDT write protection
#endif
    ESP_LOGI(CORE_TAG, "Subscribed system monitor task to WDT, status is %s. RTC watchdog is %s.", esp_err_to_name(esp_task_wdt_status(NULL)), (rtc_wdt_is_on())? "on":"off");
    while(true)
    {
#if CONFIG_ENABLE_TRACING
        ESP_ERROR_CHECK(heap_trace_start(HEAP_TRACE_LEAKS));
#endif
        sleep(min(10000 - (system_get_current_epoch_ms() % 10000), 10000));
#if CONFIG_ENABLE_TRACING
        ESP_ERROR_CHECK(heap_trace_stop());
        heap_trace_dump();
#endif

        while(system_get_current_epoch_ms() % 10000 > 9800)
        {
            sleep(40);
        }
        // ESP_LOGD(CORE_TAG, "Status task reset watchdog, loop time was %lldms", (esp_timer_get_time() - loop_start_time) / 1000);
        esp_task_wdt_reset();
#if ENABLE_RTC_WDT
        rtc_wdt_feed();
#endif
        // loop_start_time = esp_timer_get_time();
        system_check_if_lamp_allowed(true);
        
        if(system_get_current_epoch() % 30 < 5)
        {
            ESP_LOGI(CORE_TAG, "Mesh: %3s, connected: %3s, lamp: %hhu%%%s, free mem/block: %3luk, min mem: %3luk, uptime: %lldm, next sync: %02lldm %02llds", (esp_mesh_active)? "on":"off", (esp_mesh_connected)? "yes":"no", ledc_get_main_lamp_duty_percent(), system_check_if_lamp_allowed(false)? "":" (blocked)", esp_get_free_heap_size() / 1024, esp_get_minimum_free_heap_size() / 1024, esp_timer_get_time() / 60000000, (system_get_next_sync_time() - time_now.tv_sec) / 60, (system_get_next_sync_time() - time_now.tv_sec) % 60);
            if(esp_mesh_active)
            {
                // esp_mesh_get_tx_pending(&tx_pending);
                // esp_mesh_get_rx_pending(&rx_pending);
                esp_mesh_get_parent_rssi();
                mesh_addr_t(parent_address);
                esp_mesh_get_parent_bssid(&parent_address);
                // ESP_LOGI(MESH_TAG, "Pending TX/RX: %d/%d - (Bc: %d, Mgmt: %d, Chd: %d, ChdP2P: %d, Pa: %d, ParP2P: %d / ToSf: %d, ToDS: %d)", tx_pending.broadcast + tx_pending.mgmt + tx_pending.to_child + tx_pending.to_child_p2p + tx_pending.to_parent + tx_pending.to_parent_p2p, rx_pending.toSelf + rx_pending.toDS, tx_pending.broadcast, tx_pending.mgmt, tx_pending.to_child, tx_pending.to_child_p2p, tx_pending.to_parent, tx_pending.to_parent_p2p, rx_pending.toSelf, rx_pending.toDS);
                if(esp_mesh_connected && !esp_mesh_is_root())
                {
                    ESP_LOGI(MESH_TAG, "Connected to: " MACSTR ", RSSI: %hhd", MAC2STR(parent_address.addr), esp_mesh_get_parent_rssi());
                }
            }
#if SHOW_TASK_LIST
        // if(!basic_settings.is_root)
        // {
            vTaskList(task_list_buffer);
            printf("%s", task_list_buffer);
            uart_wait_tx_idle_polling(UART_NUM_0);
            vTaskGetRunTimeStats(task_list_buffer);
            printf("%s", task_list_buffer);
            uart_wait_tx_idle_polling(UART_NUM_0);
        // }
#endif
        }

        if(!system_check_if_lamp_allowed(false))
        {
            ledc_set_main_lamp_with_fade(0, 500);
        }

        if(system_get_current_epoch() % 600 < 100)
        {
            if(!got_updated_lamp_settings) {
                system_activate_current_lamp_settings();
                got_updated_lamp_settings = true;
            }
        }
        else
        {
            got_updated_lamp_settings = false;
        }
    }
    vTaskDelete(NULL);
}
```

zhangyanjiaoesp, 2023-10-30:
> It is defined in the same file

The same file with who?

zhangyanjiaoesp, 2023-10-31:
Have you tried to increase the value of `CONFIG_ESP_TASK_WDT_TIMEOUT_S ` ?

MisterScience5, 2023-10-31:
Yes, it never triggers unless the wifi gets in a stuck state during the disable function.

zhangyanjiaoesp, 2023-10-31:
@MisterScience5 
Please check these line before `esp_task_wdt_reset()`:

> while(system_get_current_epoch_ms() % 10000 > 9800)
        {
            sleep(40);
        }

Suppose that 30 seconds of the TWDT period have expired, and you sleep for another 40. Then the watchdog will trigger 10 seconds before the task returns from sleep.

You can add some debug log near here to see whether your `status_task`   has fed the watchdog in time.

MisterScience5, 2023-10-31:
The watchdog triggering unexpectedly is not an issue in the code, it only triggers sometimes when an esp_mesh_disable() is called which causes the whole system to be so bogged down it starts triggering the watchdog. If I disable the watchdog, the system also stops functioning correctly indefinitely.

In fact the status task is the only task catching the malfunction, so the task_wdt is working as intended for my program except for the fact that it does not reboot the system to correct the problem leaving the system permanently unavailable until a hard reboot is performed.

zhangyanjiaoesp, 2023-10-31:
> If I disable the watchdog, the system also stops functioning correctly indefinitely.

So can you provide a complete log of the device that is not working properly with the task watchdog disabled?

MisterScience5, 2023-10-31:
Sure, I will reflash without the task watchdog, but it may take 1-3 days before I catch the crash

MisterScience5, 2023-11-02:
One unit froze now, however nothing much is visible in the logs, logging just stops as the device disables mesh.

This is the last log lines taken this morning at 07:47, last log line is from 20:13 previous night, last time the device was seen by the mesh root was 20:12.

```
I (19:47:00.028) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 137m, next sync: 07m 00s
I (19:47:30.027) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 138m, next sync: 06m 30s
I (19:48:00.026) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 138m, next sync: 06m 00s
I (19:48:30.025) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 139m, next sync: 05m 30s
I (19:49:00.024) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 139m, next sync: 05m 00s
I (19:49:30.034) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 140m, next sync: 04m 30s
I (19:50:00.034) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 140m, next sync: 04m 00s
I (19:50:00.036) core: Getting current lamp settings for Wednesday 19:50
I (19:50:30.033) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 141m, next sync: 03m 30s
I (19:51:00.033) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 141m, next sync: 03m 00s
I (19:51:30.033) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 142m, next sync: 02m 30s
I (19:52:00.032) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 142m, next sync: 02m 00s
I (19:52:30.032) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 143m, next sync: 01m 30s
I (19:53:00.031) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 143m, next sync: 01m 00s
I (19:53:30.031) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 144m, next sync: 00m 30s
I (19:54:00.030) core: Mesh: off, connected:  no, lamp: 0%, free mem/block: 167k, min mem: 109k, uptime: 144m, next sync: 18m 00s
I (19:54:00.582) core: Starting node synchronization
I (19:54:00.582) esp_mesh: Starting mesh as node
I (19:54:00.809) esp_mesh: <MESH_EVENT_MESH_STARTED>ID:62:74:67:6c:79:73
I (19:54:00.809) esp_mesh: ------------------------------------------------
I (19:54:00.815) esp_mesh: Mesh connection details
I (19:54:00.820) esp_mesh: ------------------------------------------------
I (19:54:00.828) esp_mesh: Mesh type:                 Node
I (19:54:00.835) esp_mesh: Mesh ID:                   62:74:67:6c:79:73
I (19:54:00.843) esp_mesh: Station MAC:               30:c6:f7:2f:d7:54
I (19:54:00.850) esp_mesh: Channel:                   0
I (19:54:00.857) esp_mesh: Country code:              NO 
I (19:54:00.864) esp_mesh: Max. connected stations:   6
I (19:54:00.870) esp_mesh: AP auth mode:              0
I (19:54:00.876) esp_mesh: AP assoc expire time:      10 s
I (19:54:00.883) esp_mesh: Mesh capacity:             200
I (19:54:00.890) esp_mesh: Crypt key:                 btg_ilume_62_74_67
I (19:54:00.898) esp_mesh: Crypt key length:          64
I (19:54:00.904) esp_mesh: AP protocol:               802.11b802.11g802.11n
I (19:54:00.912) esp_mesh: STA protocol:              802.11b802.11g802.11n
I (19:54:00.921) esp_mesh: Max. layer:                40
I (19:54:00.927) esp_mesh: Topology:                  Chain
I (19:54:00.934) esp_mesh: Root fixed:                Yes
I (19:54:00.941) esp_mesh: Power save enabled:        No
I (19:54:00.947) esp_mesh: Max. TX power:             19.5 dBm
I (19:54:00.954) esp_mesh: Free heap:                 117 kB
I (19:54:00.961) esp_mesh: Scan time:                 300 ms
I (19:54:00.968) esp_mesh: MF ID:      fc9cffdd-bf25-45b6-b51e-30bc18a759fd
I (19:54:00.976) esp_mesh: MF key:     de00d30e-aad9-403d-8d93-9da4e703d660
I (19:54:00.985) esp_mesh: MF chan:    6b6f88f6-77e5-4886-9e6d-96a3cf68776a
I (19:54:00.993) esp_mesh: ------------------------------------------------
I (19:54:04.720) esp_mesh: <MESH_EVENT_FIND_NETWORK>new channel:11, router BSSID:00:00:00:00:00:00
I (19:54:05.575) esp_mesh: <MESH_EVENT_ROOT_ADDRESS>root address:0c:dc:7e:c3:85:f5
I (19:54:05.579) esp_mesh: <MESH_EVENT_PARENT_CONNECTED>layer:-1-->3, parent:30:c6:f7:2f:b8:8d, ID:62:74:67:6c:79:73, duty:0, rssi:-74
I (19:54:05.588) esp_mesh: Requesting TSF time from parent 30:c6:f7:2f:b8:8d
I (19:54:05.604) esp_mesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (19:54:05.605) esp_mesh: <MESH_EVENT_ROOT_ADDRESS>root address:0c:dc:7e:c3:85:f5
I (19:54:06.234) esp_mesh: Got new time from root, node time was 1.8448 seconds compared to root
I (19:54:06.243) esp_mesh: Got status from node 41, 30:c6:f7:2f:b8:8c up 577m on v1030
I (19:54:19.422) esp_mesh: Got new time from root, node time was 0.8311 seconds compared to root
I (19:54:19.424) esp_mesh: Got status from node 41, 0c:dc:7e:c3:85:f4 up 608m on v1030
I (19:54:30.021) core: Mesh:  on, connected: yes, lamp: 0%, free mem/block: 117k, min mem: 109k, uptime: 145m, next sync: 17m 30s
I (19:54:30.024) esp_mesh: Connected to: 30:c6:f7:2f:b8:8d, RSSI: -75
I (19:54:33.465) esp_mesh: Received command to send basic settings from 0c:dc:7e:c3:85:f4
I (19:54:33.466) esp_mesh: Sending basic settings to root
[... cut ...]
I (20:13:21.766) esp_mesh: <MESH_EVENT_PARENT_CONNECTED>layer:-1-->0, parent:30:c6:f7:2f:be:d5, ID:62:74:67:6c:79:73, duty:0, rssi:-76
I (20:13:21.774) esp_mesh: Requesting TSF time from parent 30:c6:f7:2f:be:d5
I (20:13:21.784) esp_mesh: <MESH_EVENT_TODS_REACHABLE>state:0
I (20:13:21.788) esp_mesh: <MESH_EVENT_ROOT_ADDRESS>root address:0c:dc:7e:c3:85:f5
I (20:13:21.789) esp_mesh: Got new time from root, node time was 1.5798 seconds compared to root
I (20:13:21.805) esp_mesh: <MESH_EVENT_STOPPED>
```

MisterScience5, 2023-11-03:
I am rerunning the test with all logging set to debug in the hopes of catching a crash with more info.

MisterScience5, 2023-11-06:
So far no freezes - do you have any suggestions as to any other mechanisms I can use when the watchdogs don't work?

MisterScience5, 2023-11-08:
Thank you, I have added the rtc_wdt_disable() and moved the setup to early in app_main which functions as a setup function. Will run it for a few days to see how it performs!

MisterScience5, 2023-11-17:
So far I have not caught a crash/freeze on my test system, however they are still happening on units in the wild. Hopefully I will catch one soon, after numerous changes to the code and system configuration it would be good to confirm the disappearance issue is still caused by the same problem.

MisterScience5, 2023-11-27:
I am still working on this, having tried the following:

MisterScience5, 2023-11-28:
Luckily, we found a few units that deadlock fairly often and have been using these to test.

It seems the current deadlock happens somewhere in the vicinity of dimming up the main lamp using LEDC in response to a GPIO interrupt, however I tried to remove the use of LEDC and just run the lamp with the GPIO library with no dimming but it still crashed.

After some more testing, this seems to only happen with tickless idle / automatic light sleep enabled, so I am mounting the worst of the devices in my rig to test more.

MisterScience5, 2023-11-29:
I have finally identified the deadlock, it turns out that the unit was crashing due to motion on our motion sensor, the pertinent code was this:

```
esp_sleep_enable_ext0_wakeup(PIN_MOTION_SENS, 1);

io_conf.intr_type = GPIO_INTR_ANYEDGE;
io_conf.pin_bit_mask = (1ULL << PIN_MOTION_SENS);
io_conf.mode = GPIO_MODE_INPUT;
io_conf.pull_down_en = 1;
io_conf.pull_up_en = 0;
gpio_config(&io_conf);

gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
gpio_isr_handler_add(PIN_MOTION_SENS, gpio_isr_handler, (void *)PIN_MOTION_SENS);
```

Removing the esp_sleep_enable_ext0_wakeup causes the chip to no longer crash or deadlock in these instances.

It seems that attaching an interrupt to a pin that is tagged as a wakeup pin causes a deadlock. Is this a known/expected problem, or should I make a seperate bug report for this?

I have yet to confirm whether this was the only deadlock, so please keep this bug report open for a while longer while I check that no more problems exist. So far I have not seen the original problem for a long time, but I am unsure as to why this is.

MisterScience5, 2023-11-29:
Correction, I spoke too soon, once again the interrupt crashed.

closed 2023-12-11 by Alvin1Zhang